nyanmisaka/ffmpeg-rockchip #3: avcodec_send_frame(codec_ctx_, frame_) failed.

MapleAura, 2024-01-03:
This work is fantastic. But I'm having a problem when using this library.
[image]
[image]
[image]

Do you know how to fix it?

nyanmisaka, 2024-01-03:
1. First please test the encoder to make sure there is not runtime or `/dev/{dma_heap,dri}` permission issues.
https://github.com/nyanmisaka/ffmpeg-rockchip/wiki/Encoder#test-encoders

2. The `mpp_buf_get` func from MPP segfaults usually means the parameter can be invalid. Such as `mpp_buf_size`. Can you trace the `hwfc->width` and `hwfc->height` to make sure it has valid values?

MapleAura, 2024-01-03:
> 1. First please test the encoder to make sure there is no runtime or `/dev/{dma_heap,dri}` permission issues.
>    https://github.com/nyanmisaka/ffmpeg-rockchip/wiki/Encoder#test-encoders
> 2. The `mpp_buf_get` func from MPP segfaults usually means the parameter can be invalid. Such as `mpp_buf_size`. Can you trace the `hwfc->width` and `hwfc->height` to make sure it has valid values?
> 
> These values MUST be set by the user. https://ffmpeg.org/doxygen/trunk/structAVCodecContext.html#a0d8f46461754e8abea0847dcbc41b956
> 
> https://github.com/nyanmisaka/ffmpeg-rockchip/blob/9e7314093f635b2641c15e756ec45d2a9bc9f0b9/libavcodec/rkmppenc.c#L775-L777

I tested it using the first method.
gdb --args ffmpeg -f lavfi -i testsrc2=s=1920x1080,format=nv12 -c:v hevc_rkmpp -qp_init 26 -profile:v main -level 4.1 -g:v 100 -vframes 5000 -y /data/tmp.mp4

Stack:
#0  0x0000007ff50a5e94 in mpp_buffer_create () from /usr/lib/aarch64-linux-gnu/rockchip/librockchip_mpp.so.1
#1  0x0000007ff50a900c in mpp_buffer_get_with_tag () from /usr/lib/aarch64-linux-gnu/rockchip/librockchip_mpp.so.1
#2  0x0000007ff5bf6240 in rkmpp_drm_pool_alloc (opaque=0x55555c29d0, size=<optimized out>) at libavutil/hwcontext_rkmpp.c:216
#3  0x0000007ff5be7b30 in pool_alloc_buffer (pool=0x55555d7be0) at libavutil/buffer.c:362
#4  av_buffer_pool_get (pool=0x55555d7be0) at libavutil/buffer.c:401
#5  0x0000007ff5bf5d30 in rkmpp_get_buffer (hwfc=0x55555c29d0, frame=0x55555d98c0) at libavutil/hwcontext_rkmpp.c:328
#6  0x0000007ff5bf4ae0 in av_hwframe_get_buffer (hwframe_ref=<optimized out>, frame=0x55555d98c0, flags=flags@entry=0) at libavutil/hwcontext.c:563
#7  0x0000007ff70e5bbc in rkmpp_submit_frame (frame=0x55555b8fd0, avctx=0x55555bb1a0) at libavcodec/rkmppenc.c:497
#8  rkmpp_encode_frame (avctx=0x55555bb1a0, packet=0x7fe0000bd0, frame=0x55555b8fd0, got_packet=0x7fffffe544) at libavcodec/rkmppenc.c:714
#9  0x0000007ff6f0af1c in ff_encode_encode_cb (avctx=avctx@entry=0x55555bb1a0, avpkt=0x7fe0000bd0, frame=0x55555b8fd0, got_packet=0x7fffffe544) at libavcodec/encode.c:264
#10 0x0000007ff6f0b338 in encode_simple_internal (avpkt=0x7fe0000bd0, avctx=0x55555bb1a0) at libavcodec/encode.c:350
#11 encode_simple_receive_packet (avpkt=<optimized out>, avctx=<optimized out>) at libavcodec/encode.c:364
#12 encode_receive_packet_internal (avctx=avctx@entry=0x55555bb1a0, avpkt=0x7fe0000bd0) at libavcodec/encode.c:398
#13 0x0000007ff6f0b598 in avcodec_send_frame (avctx=avctx@entry=0x55555bb1a0, frame=frame@entry=0x55555d7c50) at libavcodec/encode.c:541
#14 0x00000055555651e0 in encode_frame (of=of@entry=0x55555b9ad0, ost=ost@entry=0x55555bae40, frame=frame@entry=0x55555d7c50) at fftools/ffmpeg_enc.c:644
#15 0x0000005555565964 in submit_encode_frame (of=of@entry=0x55555b9ad0, ost=ost@entry=0x55555bae40, frame=<optimized out>, frame@entry=0x55555b92a0) at fftools/ffmpeg_enc.c:750
#16 0x0000005555565b30 in do_video_out (in_picture=0x55555b92a0, ost=0x55555bae40, of=0x55555b9ad0) at fftools/ffmpeg_enc.c:846
#17 enc_frame (ost=ost@entry=0x55555bae40, frame=frame@entry=0x55555b92a0) at fftools/ffmpeg_enc.c:860
#18 0x0000005555567ed8 in fg_output_frame (ofp=ofp@entry=0x55555bbe00, frame=frame@entry=0x55555b9730) at fftools/ffmpeg_filter.c:2125
#19 0x00000055555697ac in fg_output_step (flush=0, ofp=0x55555bbe00) at fftools/ffmpeg_filter.c:2220
#20 reap_filters (flush=<optimized out>, fg=<optimized out>) at fftools/ffmpeg_filter.c:2239
#21 reap_filters (fg=0x55555bb7e0, flush=0) at fftools/ffmpeg_filter.c:2228
#22 0x000000555555da88 in transcode_step (ost=<optimized out>) at fftools/ffmpeg.c:1155
#23 transcode (err_rate_exceeded=<synthetic pointer>) at fftools/ffmpeg.c:1204
#24 main (argc=<optimized out>, argv=<optimized out>) at fftools/ffmpeg.c:1330

It seems that the error is the same as before. (Stack#2)
I've checked the width and height to make sure the values are valid.(1920*1080)

nyanmisaka, 2024-01-03:
Which Rockchip linux kernel are you using? (5.10 or 6.1-devel)

And which platform? (3588, 356x or older 33xx)

MapleAura, 2024-01-03:
> Which Rockchip linux kernel are you using? (5.10 or 6.1-devel)
> 
> And which platform? (3588, 356x or older 33xx)

 5.10   3588

nyanmisaka, 2024-01-03:
- Make sure these device files exist.
```
# DRM allocator
/dev/dri

# DMA_HEAP allocator
/dev/dma_heap

# RGA filters
/dev/rga

# MPP codecs
/dev/mpp_service
```

- Add the following lines into `/etc/udev/rules.d/99-rk-device-permissions.rules` and reboot.
```
KERNEL=="mpp_service", MODE="0660", GROUP="video"
KERNEL=="rga", MODE="0660", GROUP="video"
KERNEL=="system-dma32", MODE="0666", GROUP="video"
KERNEL=="system-uncached", MODE="0666", GROUP="video"
KERNEL=="system-uncached-dma32", MODE="0666", GROUP="video" RUN+="/usr/bin/chmod a+rw /dev/dma_heap"
```

- Re-build and install the latest MPP from https://github.com/rockchip-linux/mpp and try again.

MapleAura, 2024-01-04:
> * Make sure these device files exist.
>
> ```
> # DRM allocator
> /dev/dri
>
> # DMA_HEAP allocator
> /dev/dma_heap
>
> # RGA filters
> /dev/rga
>
> # MPP codecs
> /dev/mpp_service
> ```
>
> * Add the following lines into [...]

Thank you very much. Followed your last piece of advice and is working now.

closed 2024-01-04 by nyanmisaka

gjm19761, 2024-01-08:
ok following that, doing the mpp tests, 1080p h264 hardware encoding etc work, but hevc does not, get this error ; 

[hevc_rkmpp @ 0xaaaae3ae2aa0] Failed to init MPP context: -1
[vost#0:0/hevc_rkmpp @ 0xaaaae3ae2700] Error while opening encoder - maybe incorrect parameters such as bit_rate, rate, width or height.
Error while filtering: Generic error in an external library
[out#0/mp4 @ 0xaaaae3ae1330] Nothing was written into output file, because at least one of its streams received no packets.

outhud, 2024-08-21:
> * Make sure these device files exist.
>
> ```
> # DRM allocator
> /dev/dri
>
> # DMA_HEAP allocator
> /dev/dma_heap
>
> # RGA filters
> /dev/rga
>
> # MPP codecs
> /dev/mpp_service
> ```
>
> * Add the following lines into [...]

Thank you. I was getting the same error with Armbian 24.5 Bookworm with kernel 6.1.75-vendor-rk35xx on RK3566.

For anyone with the same problem, it was the either adding those udev rules and/or the reboot which fixed it for me.