nyanmisaka/ffmpeg-rockchip #5: Generic error in external library

gjm19761, 2024-01-08:
[hevc_rkmpp @ 0xaaaaf62be8a0] Failed to init MPP context: -1
[vist#0:0/hevc @ 0xaaaaf62bcbd0] Error while opening decoder: Generic error in an external library
[vost#0:0/h264_rkmpp @ 0xaaaaf62dfa90] Error initializing a simple filtergraph
Error opening output file /ssd/Aliens.Special.Edition.1080p_h264_6M.mkv.
Error opening output files: Generic error in an external library

I get this on my rock 3c which has Rockchip RK3566 SoC and im using 5.10 rk35xx kernel

nyanmisaka, 2024-01-08:
@gjm19761 Take a look at https://github.com/nyanmisaka/ffmpeg-rockchip/issues/3#issuecomment-1875297620

nyanmisaka, 2024-01-08:
@gjm19761 
> ok following that, doing the mpp tests, 1080p h264 hardware encoding etc work, but hevc does not, get this error ;

> [hevc_rkmpp @ 0xaaaae3ae2aa0] Failed to init MPP context: -1
[vost#0:0/hevc_rkmpp @ 0xaaaae3ae2700] Error while opening encoder - maybe incorrect parameters such as bit_rate, rate, width or height.

It seems something went wrong in your device tree or the source code of rockchip linux kernel (5.10.110, 5.10.160, 5.10.198). Generally speaking, the [`rkvenc` node](https://github.com/radxa/kernel/blob/bfb9351f30808403c9fdb0131499b1e77a583e1d/arch/arm64/boot/dts/rockchip/rk3566-rock-3c.dtsi#L709-L716) enables both h.264 and hevc encoder on rk356x chips, so this is unlikely to happen.

closed 2024-02-10 by nyanmisaka