nyanmisaka/ffmpeg-rockchip #7: addr = mpp_buffer_get_ptr(desc->buffers[i]); core dump

MapleAura, 2024-01-13:
大佬您好，又来请教一个问题。
我在进行网络流解码或者文件解码操作时hwcontext_rkmpp.c:397出现了core dump。
命令行如下：
`gdb --args ffmpeg -re -c:v h264_rkmpp  -hwaccel_output_format drm_prime -i /xxx/8.mkv -an -sn -vframes 5000 -f null -`

堆栈：
`#0  0x0000007ff507f6cc in mpp_buffer_get_ptr_with_caller () from /usr/local/lib/librockchip_mpp.so.1
#1  0x0000007ff5bf57c0 in rkmpp_map_frame (hwfc=hwfc@entry=0x7fec0013c0, dst=0x7fec001fa0, 
    src=src@entry=0x7fec0017a0, flags=flags@entry=1) at libavutil/hwcontext_rkmpp.c:397
#2  0x0000007ff5bf5afc in rkmpp_transfer_data_from (hwfc=0x7fec0013c0, dst=0x7fec000b60, src=0x7fec0017a0)
    at libavutil/hwcontext_rkmpp.c:489
#3  0x0000007ff5bf4380 in av_hwframe_transfer_data (dst=dst@entry=0x7fec000b60, src=0x7fec0017a0, 
    flags=flags@entry=0) at libavutil/hwcontext.c:488
#4  0x0000007ff70e4ab8 in rkmpp_get_frame (timeout=<optimized out>, frame=0x7fec000b60, avctx=0x55555e1dd0)
    at libavcodec/rkmppdec.c:712
#5  rkmpp_decode_receive_frame (avctx=0x55555e1dd0, frame=0x7fec000b60) at libavcodec/rkmppdec.c:829
#6  0x0000007ff6edcda4 in decode_receive_frame_internal (avctx=avctx@entry=0x55555e1dd0, 
    frame=frame@entry=0x7fec000b60) at libavcodec/decode.c:626
#7  0x0000007ff6edd434 in ff_decode_receive_frame (avctx=avctx@entry=0x55555e1dd0, frame=frame@entry=0x7fec000b60)
--Type <RET> for more, q to quit, c to continue without paging--
   at libavcodec/decode.c:807
#8  0x0000007ff6e54ff4 in avcodec_receive_frame (avctx=avctx@entry=0x55555e1dd0, frame=frame@entry=0x7fec000b60)
    at libavcodec/avcodec.c:718
#9  0x000000555555e720 in packet_decode (frame=0x7fec000b60, pkt=0x7fec000d50, ist=<optimized out>)
    at fftools/ffmpeg_dec.c:592
#10 decoder_thread (arg=0x55555bb570) at fftools/ffmpeg_dec.c:712
#11 0x0000007ff5ae7624 in start_thread (arg=0x555555e1d8 <decoder_thread>) at pthread_create.c:477
#12 0x0000007ff5a3e49c in thread_start () at ../sysdeps/unix/sysv/linux/aarch64/clone.S:78`

采用调用ffmpeg接口的方式出现了同样的问题，mpp已经更新到官网最新，使用的是最新的6.1分支的代码。

附件是我测试用的视频
[8.zip](https://github.com/nyanmisaka/ffmpeg-rockchip/files/13926859/8.zip)

nyanmisaka, 2024-01-13:
@MapleAura 现在流出的6.1内核不是正式版，有很多bug，里面的dma-heap分配器没有移植有问题用不了，而mpp运行库写死了用/dev/dma_heap这个路径是否可读来检查dma-heaps是否可用，然而实际里面的设备不可用，所以就segfault了。

建议暂时换回5.10，或者用此补丁打在mpp运行库上。
https://github.com/nyanmisaka/mpp/commit/49b4fc71d25255b3005aa05f761bfd92b8c85e78

nyanmisaka, 2024-01-13:
@MapleAura It was a regression. Please apply the latest commit 97e6a5e and try again.

MapleAura, 2024-01-13:
> t was a regression

好了 结果能拿到了 谢谢您

closed 2024-01-13 by nyanmisaka