nyanmisaka/ffmpeg-rockchip #10: Hello

lvjunping, 2024-01-16:


lvjunping, 2024-01-16:
libavfilter/rkrga_common.c: In function 'get_drm_afbc_format':
libavfilter/rkrga_common.c:154:38: error: 'DRM_FORMAT_INVALID' undeclared (first use in this function); did you mean 'DRM_FORMAT_MOD_INVALID'?
  154 |     default:                  return DRM_FORMAT_INVALID;
      |                                      ^~~~~~~~~~~~~~~~~~
      |                                      DRM_FORMAT_MOD_INVALID
libavfilter/rkrga_common.c:154:38: note: each undeclared identifier is reported only once for each function it appears in
CC	libavfilter/setpts.o
libavfilter/rkrga_common.c: In function 'query_frame':
libavfilter/rkrga_common.c:554:29: error: 'DRM_FORMAT_INVALID' undeclared (first use in this function); did you mean 'DRM_FORMAT_MOD_INVALID'?
  554 |         if (drm_afbc_fmt == DRM_FORMAT_INVALID) {
      |                             ^~~~~~~~~~~~~~~~~~
      |                             DRM_FORMAT_MOD_INVALID
libavfilter/rkrga_common.c:562:13: error: implicit declaration of function 'DRM_FORMAT_MOD_ARM_AFBC'; did you mean 'DRM_FORMAT_MOD_ARM_TYPE_AFBC'? [-Werror=implicit-function-declaration]
  562 |             DRM_FORMAT_MOD_ARM_AFBC(AFBC_FORMAT_MOD_SPARSE | AFBC_FORMAT_MOD_BLOCK_SIZE_16x16);
      |             ^~~~~~~~~~~~~~~~~~~~~~~
      |             DRM_FORMAT_MOD_ARM_TYPE_AFBC
libavfilter/rkrga_common.c:562:37: error: 'AFBC_FORMAT_MOD_SPARSE' undeclared (first use in this function)
  562 |             DRM_FORMAT_MOD_ARM_AFBC(AFBC_FORMAT_MOD_SPARSE | AFBC_FORMAT_MOD_BLOCK_SIZE_16x16);
      |                                     ^~~~~~~~~~~~~~~~~~~~~~
libavfilter/rkrga_common.c:562:62: error: 'AFBC_FORMAT_MOD_BLOCK_SIZE_16x16' undeclared (first use in this function)
  562 |             DRM_FORMAT_MOD_ARM_AFBC(AFBC_FORMAT_MOD_SPARSE | AFBC_FORMAT_MOD_BLOCK_SIZE_16x16);
      |                                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
libavfilter/rkrga_common.c: In function 'get_drm_afbc_format':
libavfilter/rkrga_common.c:156:1: error: control reaches end of non-void function [-Werror=return-type]
  156 | }

nyanmisaka, 2024-01-17:
@lvjunping 
These definitions are part of the header files in `libdrm-dev` package. If you are using a very old distro please download and compile latest header files from the drm repo.

https://gitlab.freedesktop.org/mesa/drm
https://github.com/nyanmisaka/ffmpeg-rockchip/wiki/Compilation

lvjunping, 2024-01-18:
Thank you very much! I have compilation was successful!

closed 2024-01-18 by nyanmisaka