nyanmisaka/ffmpeg-rockchip #11: The decoding time of a single frame is too long.

MapleAura, 2024-01-20:
When I calculated the time to decode, I found out the average time  is almost 84ms.

Open the log and find such a problem “Decoder buffer is full”. This happens regardless of whether I export to NV12 format or drm_prime.

[image]

This is the test code:
[image]

nyanmisaka, 2024-01-20:
Bear in mind that `memcpy` and `Resize` time should not be counted in HW decoding time.

In particular, `av_hwframe_transfer_data()` already includes `memcpy` for AV_PIX_FMT_DRM_PRIME => AV_PIX_FMT_NV12 conversion.

Therefore your code contains redundant `memcpy`.

daoan1412, 2024-01-20:
I have a similar problem, even without using memcpy and Resize.

MapleAura, 2024-01-21:
I've counted the cost on several interfaces. ‘av_read_frame’ cost almost 20ms per frame. I send data three times just receive decoded data once. The log message indicates that the queue is full. In other words, the result can only be returned once if the data is read three times and it takes almost 60 ms. According to the information I have observed from the log, it seems that the decoding time is larger than reading time. But from the experiments done by others, if I do not carry out the copy operation (just export drm_prime), the decoding time is only a few ms.

nyanmisaka, 2024-01-21:
@MapleAura

I did find an issue regarding the high latency. Please try [this patch](https://github.com/nyanmisaka/ffmpeg-rockchip/commit/ddf7cf1d77aa9bbadb30e62b0d824cddaf55cb07) and see if it helps.

MapleAura, 2024-01-22:
That's great. It works. :)

nyanmisaka, 2024-01-23:
Closed by e753665

closed 2024-01-23 by nyanmisaka