nyanmisaka/ffmpeg-rockchip #14: Rendering Docs

hbiyik, 2024-01-22:
I am giving the rendering wiki page update as a patch, i think gh does not support prs for wiki.

Basically documents how to get acceleration on mpv and kodi including afbc support as well.

[rendering-docs.patch.tar.gz](https://github.com/nyanmisaka/ffmpeg-rockchip/files/14009139/rendering-docs.patch.tar.gz)

depends on
https://github.com/nyanmisaka/ffmpeg-rockchip/pull/13
https://github.com/nyanmisaka/ffmpeg-rockchip/pull/4
https://github.com/nyanmisaka/ffmpeg-rockchip/commit/75eca72cc4be48912d3c935f68748d73d884dce0 (afbc stride issue)

Preview is here:
https://github.com/hbiyik/ffmpeg-rockchip/wiki/Rendering

hbiyik, 2024-01-27:
And here is the patch with latest PRs
[rendering-docs_v2.patch.tar.gz](https://github.com/nyanmisaka/ffmpeg-rockchip/files/14073371/rendering-docs_v2.patch.tar.gz)

nyanmisaka, 2024-01-28:
```patch
// 0002-Support-P210-LF.patch
@@ -624,6 +625,7 @@ pipe_format_to_chroma_format(enum pipe_format format)
       case PIPE_FORMAT_P010:
       case PIPE_FORMAT_P012:
       case PIPE_FORMAT_P016:
+      case PIPE_FORMAT_P210:
       case PIPE_FORMAT_P030:
       case PIPE_FORMAT_Y16_U16_V16_420_UNORM:
          return PIPE_VIDEO_CHROMA_FORMAT_420;
```

Shouldn't it be `PIPE_VIDEO_CHROMA_FORMAT_422`?

hbiyik, 2024-01-28:
yes, i have also looked at what i have done in P210 mesa patchset, there are alot of mistakes in there. But the biggest mistake is its existance :).

Mesa upstream or panfork both supports P210 through egl without any patchset, so those need to be removed.

hbiyik, 2024-01-28:
here is the v3 with latest changes

[rendering-docs_v3.patch.tar.gz](https://github.com/nyanmisaka/ffmpeg-rockchip/files/14076500/rendering-docs_v3.patch.tar.gz)

nyanmisaka, 2024-03-07:
Done in https://github.com/nyanmisaka/ffmpeg-rockchip/wiki/Rendering

closed 2024-03-07 by nyanmisaka