nyanmisaka/ffmpeg-rockchip #16: Hardware decode/encode not working

ShyneTurtle, 2024-01-22:
Hello,

after seeing your work here: https://github.com/jellyfin/jellyfin-ffmpeg/issues/34
i decided to try building your ffmpeg fork, following the wiki.

However, when i try to use it using the following command: `DISPLAY=:0 ffplay -codec:v hevc_rkmpp /etc/jellyfinlib/bbb_sunflower_2160p_60fps_normal.mp4`

The video doesn't even open. I also tried to use vlc with this custom ffmpeg and the output is unreadable. I also tried transcoding video and reading it on an x86 pc and the video is corrupted.

While trying to read the original video with the your ffmpeg fork, FFPLAY gives me this error:
```
[hevc_mp4toannexb @ 0x7f382b4b50] No parameter sets in the extradata
[hevc_rkmpp @ 0x7f382dc220] Failed to get packet (code = -1094995529)
    Last message repeated 437 times
[hevc_rkmpp @ 0x7f382dc220] Failed to get packet (code = -1094995529)
    Last message repeated 1705 times
[hevc_rkmpp @ 0x7f382dc220] Failed to get packet (code = -1094995529)
    Last message repeated 1944 times
```

Hardware: CM3588 from friendlyelec
Uname: `Linux CM3588 6.1.25 #123 SMP Thu Nov 16 11:25:37 CST 2023 aarch64 aarch64 aarch64 GNU/Linux`
FFMPEG version: `668585f built with gcc 11 (Ubuntu 11.4.0-1ubuntu1~22.04)`
Configuration `configuration: --prefix=/usr --enable-gpl --enable-version3 --enable-libdrm --enable-rkmpp --enable-rkrga`

If you need further details please let me know, i've been looking into this for hours now and i can't find anything telling me what's wrong.

Thank you for your time.

nyanmisaka, 2024-01-23:
@ShyneTurtle 
There is no such line of code `Failed to get packet (code = %d)` is this repository [`rkmppdec.c`](https://github.com/nyanmisaka/ffmpeg-rockchip/blob/master/libavcodec/rkmppdec.c). It comes from [JeffyCN/FFmpeg](https://github.com/JeffyCN/FFmpeg/blob/003204d8e110ab907c8d9f4a0f9d4cf7530b64c3/libavcodec/rkmppdec.c#L719).

So I concluded that there is more than one ffmpeg/ffplay binary in your environment and you are not using absolute path to execute it.

Moreover, when building `ffplay` it will check for the `libsdl2-dev` package, and if it is not detected, it will skip building `ffplay`.

ShyneTurtle, 2024-01-23:
Hello and thank you for your answer.

I thought that the previous builds would be overwritten, my bad.
I tried running ffmpeg again (from this repo) and the transcode worked flawlessly.

Only problem is that ffplay is not built, even though i added the `--enable-ffplay` flag and no errors showed up.
You can read the whole build logs here (i used the script provided in the wiki as a baseline): https://pastebin.com/nkyWMFQY

This is a minor problem and most certainly an error in my configure call.

This is definitely off topic but would you mind telling me what i should do to use this in jellyfin ?
Just so you know i used the .dev package from their official builds `jellyfin-ffmpeg5_5.1.4-2-jammy_arm64.deb`.
Problem being i can't provide a path to the custom ffmpeg since it uses its internal ffmpeg build and the `ffmpeg path` option is disabled in the dashboard.

nyanmisaka, 2024-01-23:
```
// line 751
Programs:
ffmpeg                  ffprobe
```

Try `--enable-sdl`.

You can download the prebuilt jellyfin-ffmpeg here.
[Checks -> Artifacts -> arm64.deb](https://github.com/jellyfin/jellyfin-ffmpeg/pull/318)

Use my [docker image](https://github.com/jellyfin/jellyfin-ffmpeg/issues/34) instead.

ShyneTurtle, 2024-01-23:
Thank you for your answer.

Enabling SDL didn't fix the ffplay issue.

Using the docker image after clearing out all the old jellyfin data (even in /home/pi/.local/jellyfin ???) the hardware transcoding works in jellyfin.

Thank you again for all the work you have done to bring this amazing chip into a usable state.

nyanmisaka, 2024-01-24:
oops. it should be `--enable-sdl2`

```
./configure --disable-doc --disable-shared --enable-static --enable-gpl --enable-version3 --enable-libdrm --enable-rkmpp --enable-rkrga --enable-sdl2
ERROR: sdl2 requested but not found
```

nyanmisaka, 2024-01-30:
It seems there are permission issues in your environment. See the links below to fix them.
https://github.com/nyanmisaka/ffmpeg-rockchip/issues/3#issuecomment-1875297620
https://github.com/nyanmisaka/ffmpeg-rockchip#important

nyanmisaka, 2024-03-25:
Closing as it can not be reproduced.

closed 2024-03-25 by nyanmisaka