nyanmisaka/ffmpeg-rockchip #22: HW decoding HEVC w/Jellyfin doesn't work when subtitles enabled

bbklopfer, 2024-01-28:
(First, thanks for this project --- it's awesome, and the performance on my Orange Pi 5+ is fantastic with HW transcoding!)

I recently did a fresh pull on the docker image `nyanmisaka/jellyfin:latest-rockchip` and noticed that, with HW HEVC decode enabled, some videos will not show the video with the subtitles enabled (the subtitles will show, however it's just subtitles on a black screen). This happens on browser and android client. At least one video successfully worked on android client when the subtitles were displayed by the client and not embedded into the video stream, however.

My recollection is that this did not happen prior to pulling the docker image today; I'm pretty new to using docker so I'm not sure how to revert to the previous image to test this out.

Happy to assist however I can in debugging, and apologies if this is the wrong place for this.

bbklopfer, 2024-01-28:
Update: I'm guessing the previous version of `latest-rockchip` was [nyanmisaka/jellyfin:240117-arm64](https://hub.docker.com/layers/nyanmisaka/jellyfin/240117-arm64/images/sha256-3f8afaed4408eb421448116beb1202db082534f5c61abc3cb665bf97fdcb9ab5?context=explore), so I pulled that and it works as expected: with HW decode (and encode) enabled, subtitles appear as expected.

nyanmisaka, 2024-01-28:
@bbklopfer 
That’s the expected behavior since Rockchip [changed](https://github.com/Joshua-Riek/linux-rockchip/commit/533b152d3217de742f97764a84e605c372e160d9) their librga kernel API last year. You have to update the BSP kernel to 5.10 **rkr6** to get the fully-functional subtitle burn-in output.

You can simply install https://github.com/Joshua-Riek/ubuntu-rockchip instead.

bbklopfer, 2024-01-29:
Thanks so much!

closed 2024-01-29 by bbklopfer

bbklopfer, 2024-02-02:
Follow up in case anyone ended up here with a similar question:

I installed the kernel from https://github.com/Joshua-Riek/ubuntu-rockchip , but unfortunately am experiencing higher power usage (~2x power consumption at idle), increased thermals, and a load average that won't go below 1.00 ( https://github.com/Joshua-Riek/ubuntu-rockchip/issues/606 ). I did verify that the transcoding works as expected, however.

nyanmisaka, 2024-02-03:
> Follow up in case anyone ended up here with a similar question:
> 
> I installed the kernel from https://github.com/Joshua-Riek/ubuntu-rockchip , but unfortunately am experiencing higher power usage (~2x power consumption at idle), increased thermals, and a load average that won't go below 1.00 ( [Joshua-Riek/ubuntu-rockchip#606](https://github.com/Joshua-Riek/ubuntu-rockchip/issues/606) ). I did verify that the transcoding works as expected, however.

This distro is mainly for desktop applications. Rockchip's default cpu/gpu governor can cause high latency or lag. So they set the governor to performance by default.

https://github.com/Joshua-Riek/ubuntu-rockchip/blob/main/overlay/usr/lib/systemd/system/cpu-governor-performance.service
https://github.com/Joshua-Riek/ubuntu-rockchip/blob/main/overlay/usr/lib/systemd/system/gpu-governor-performance.service

bbklopfer, 2024-02-05:
Thanks! Unfortunately performance governors don't appear to be the issue (or at least, not the whole issue) --- the same set of issues happens with the stock Orange Pi Debian userspace if I install the kernel from https://github.com/Joshua-Riek/ubuntu-rockchip (that userspace uses `ondemand` and `simple_ondemand` for CPU and GPU, respectively).

I wasn't super careful with looking at the thermals/power with this userspace though, so perhaps it is a little better; but it certainly runs hotter than the Orange Pi issued kernel, and the load average = 1 bug persists.

EvilOlaf, 2024-04-21:
Seems like hdmirx is the cause. Once disabled in device tree the load stays normal. So there must be some weirdness inside this driver.