nyanmisaka/ffmpeg-rockchip #23: Rockchip BSP/vendor kernel is necessary, 5.10 and 6.1 are two tested versions...?

derzahla, 2024-01-31:
Hi, thank you for your work on this.  What is this v6.1 BSP kernel you mention being tested?  Does it support rk3588? I have some OPi5pluses that I've been playing with and im only aware of a v5.10 BSP kernel.  Would be nice to have a 6.1 bsp kernel option. I've tinkered with building various 6.X mainline/collabora based kernels and random patches, but I don't believe theres been any working hardware video decoder support yet.   Thanks!

nyanmisaka, 2024-02-01:
https://github.com/nyanmisaka/linux-rockchip/commits/rk-6.1-dev2/

It’s based on https://github.com/JeffyCN/mirrors/tree/kernel-6.1-2024_01_02

derzahla, 2024-02-02:
Thanks!  So its just an unofficial dev build that a rockchip employee made available?  Any tips on compiling it?  gcc12 is failing for me. clang completes but upon boot I have the console flooded with rk_hdmirx messages and then kernel panic.

derzahla, 2024-02-02:
Nevermind, I used your build_def and it compiles and boots with GCC. Thanks!  Have you tried patching it up to the latest 6.1 release?

nyanmisaka, 2024-02-03:
No I don't have time to try. In addition, there is no point in doing this kind of thing on the BSP kernel tree, and Rockchip will do it regularly, usually once every few months.

https://github.com/ThomasKaiser/Knowledge/blob/master/articles/Quick_Preview_of_ROCK_5B.md#important-insights-and-suggested-optimisations

https://forum.radxa.com/t/the-radxa-bsp-kernel-patches-from-5-10-67-to-5-10-123/11647

closed 2024-02-10 by nyanmisaka