nyanmisaka/ffmpeg-rockchip #25: external library error? rkmpp decoder issue

great9, 2024-02-08:
```
2024-02-08 15:33:35.670125850  [2024-02-08 15:33:35] ffmpeg.camera2.detect ERROR   : Input #0, rtsp, from 'rtsp://*:*@0.0.0.1:554/cam/realmonitor?channel=1&subtype=2':
2024-02-08 15:33:35.670394175  [2024-02-08 15:33:35] ffmpeg.camera2.detect ERROR   :   Metadata:
2024-02-08 15:33:35.670695166  [2024-02-08 15:33:35] ffmpeg.camera2.detect ERROR   :     title           : Media Server
2024-02-08 15:33:35.670903993  [2024-02-08 15:33:35] ffmpeg.camera2.detect ERROR   :   Duration: N/A, start: 0.080000, bitrate: N/A
2024-02-08 15:33:35.671047780  [2024-02-08 15:33:35] ffmpeg.camera2.detect ERROR   :   Stream #0:0: Video: hevc (Main), yuvj420p(pc, bt470bg/bt470bg/smpte170m), 1920x1080, 25 fps, 25 tbr, 90k tbn
2024-02-08 15:33:35.671235607  [2024-02-08 15:33:35] ffmpeg.camera2.detect ERROR   : Stream mapping:
2024-02-08 15:33:35.671427518  [2024-02-08 15:33:35] ffmpeg.camera2.detect ERROR   :   Stream #0:0 -> #0:0 (hevc (hevc_rkmpp) -> rawvideo (native))
2024-02-08 15:33:35.671616512  [2024-02-08 15:33:35] ffmpeg.camera2.detect ERROR   : Press [q] to stop, [?] for help
2024-02-08 15:33:35.671804048  [2024-02-08 15:33:35] ffmpeg.camera2.detect ERROR   : Output #0, rawvideo, to 'pipe:':
2024-02-08 15:33:35.671993626  [2024-02-08 15:33:35] ffmpeg.camera2.detect ERROR   :   Metadata:
2024-02-08 15:33:35.672180870  [2024-02-08 15:33:35] ffmpeg.camera2.detect ERROR   :     title           : Media Server
2024-02-08 15:33:35.672371323  [2024-02-08 15:33:35] ffmpeg.camera2.detect ERROR   :     encoder         : Lavf60.16.100
2024-02-08 15:33:35.672562650  [2024-02-08 15:33:35] ffmpeg.camera2.detect ERROR   :   Stream #0:0: Video: rawvideo (I420 / 0x30323449), yuv420p(pc, bt470bg/bt470bg/smpte170m, progressive), 1920x1080, q=2-31, 622080 kb/s, 25 fps, 25 tbn
2024-02-08 15:33:35.672759519  [2024-02-08 15:33:35] ffmpeg.camera2.detect ERROR   :     Metadata:
2024-02-08 15:33:35.672947347  [2024-02-08 15:33:35] ffmpeg.camera2.detect ERROR   :       encoder         : Lavc60.31.102 rawvideo
2024-02-08 15:33:35.673150341  [2024-02-08 15:33:35] ffmpeg.camera2.detect ERROR   : frame=    0 fps=0.0 q=0.0 size=       0kB time=N/A bitrate=N/A dup=0 drop=1 speed=N/A    
2024-02-08 15:33:35.673332627  [2024-02-08 15:33:35] ffmpeg.camera2.detect ERROR   : frame=    0 fps=0.0 q=0.0 size=       0kB time=N/A bitrate=N/A dup=0 drop=1 speed=N/A    
2024-02-08 15:33:35.673523079  [2024-02-08 15:33:35] ffmpeg.camera2.detect ERROR   : frame=   38 fps=9.4 q=-0.0 size=  112388kB time=00:00:01.80 bitrate=511488.0kbits/s dup=0 drop=1 speed=0.445x    
2024-02-08 15:33:35.673711490  [2024-02-08 15:33:35] ffmpeg.camera2.detect ERROR   : frame=  103 fps= 23 q=-0.0 size=  309825kB time=00:00:04.40 bitrate=576837.8kbits/s dup=0 drop=1 speed=0.967x    
2024-02-08 15:33:35.673923234  [2024-02-08 15:33:35] ffmpeg.camera2.detect ERROR   : frame=  141 fps= 28 q=-0.0 size=  425250kB time=00:00:05.64 bitrate=617668.1kbits/s dup=0 drop=1 speed=1.11x    
2024-02-08 15:33:35.674118353  [2024-02-08 15:33:35] ffmpeg.camera2.detect ERROR   : frame=  155 fps= 28 q=-0.0 size=  467775kB time=00:00:06.16 bitrate=622080.0kbits/s dup=0 drop=1 speed=1.09x    
2024-02-08 15:33:35.674309680  [2024-02-08 15:33:35] ffmpeg.camera2.detect ERROR   : frame=  161 fps= 26 q=-0.0 size=  486000kB time=00:00:06.44 bitrate=618216.1kbits/s dup=0 drop=1 speed=1.04x    
2024-02-08 15:33:35.674503049  [2024-02-08 15:33:35] ffmpeg.camera2.detect ERROR   : frame=  167 fps= 23 q=-0.0 size=  504225kB time=00:00:06.64 bitrate=622080.0kbits/s dup=0 drop=1 speed=0.926x    
2024-02-08 15:33:35.674690293  [2024-02-08 15:33:35] ffmpeg.camera2.detect ERROR   : frame=  192 fps= 23 q=-0.0 size=  580162kB time=00:00:07.64 bitrate=622080.0kbits/s dup=0 drop=1 speed=0.925x    
2024-02-08 15:33:35.674876371  [2024-02-08 15:33:35] ffmpeg.camera2.detect ERROR   : frame=  217 fps= 24 q=-0.0 size=  656100kB time=00:00:08.64 bitrate=622080.0kbits/s dup=0 drop=1 speed=0.959x    
2024-02-08 15:33:35.675067115  [2024-02-08 15:33:35] ffmpeg.camera2.detect ERROR   : frame=  253 fps= 26 q=-0.0 size=  765450kB time=00:00:10.08 bitrate=622080.0kbits/s dup=0 drop=1 speed=1.05x    
2024-02-08 15:33:35.675254360  [2024-02-08 15:33:35] ffmpeg.camera2.detect ERROR   : frame=  266 fps= 26 q=-0.0 size=  804938kB time=00:00:10.64 bitrate=619741.4kbits/s dup=0 drop=1 speed=1.05x    
2024-02-08 15:33:35.675450354  [2024-02-08 15:33:35] ffmpeg.camera2.detect ERROR   : frame=  280 fps= 26 q=-0.0 size=  847462kB time=00:00:11.16 bitrate=622080.0kbits/s dup=0 drop=1 speed=1.05x    
2024-02-08 15:33:35.675640514  [2024-02-08 15:33:35] ffmpeg.camera2.detect ERROR   : frame=  289 fps= 26 q=-0.0 size=  874800kB time=00:00:11.56 bitrate=619927.5kbits/s dup=0 drop=1 speed=1.03x    
2024-02-08 15:33:35.675829800  [2024-02-08 15:33:35] ffmpeg.camera2.detect ERROR   : frame=  292 fps= 25 q=-0.0 size=  883912kB time=00:00:11.64 bitrate=622080.0kbits/s dup=0 drop=1 speed=0.986x    
2024-02-08 15:33:35.676017920  [2024-02-08 15:33:35] ffmpeg.camera2.detect ERROR   : frame=  295 fps= 23 q=-0.0 size=  893025kB time=00:00:12.08 bitrate=605601.1kbits/s dup=0 drop=1 speed=0.96x    
2024-02-08 15:33:35.676206914  [2024-02-08 15:33:35] ffmpeg.camera2.detect ERROR   : frame=  297 fps= 22 q=-0.0 size=  899100kB time=00:00:12.16 bitrate=605709.5kbits/s dup=0 drop=1 speed=0.902x    
2024-02-08 15:33:35.676396200  [2024-02-08 15:33:35] ffmpeg.camera2.detect ERROR   : frame=  305 fps= 22 q=-0.0 size=  923400kB time=00:00:12.48 bitrate=606129.2kbits/s dup=0 drop=1 speed=0.891x    
2024-02-08 15:33:35.676588110  [2024-02-08 15:33:35] ffmpeg.camera2.detect ERROR   : frame=  309 fps= 21 q=-0.0 size=  935550kB time=00:00:12.64 bitrate=606331.1kbits/s dup=0 drop=1 speed=0.87x    
2024-02-08 15:33:35.676783229  [2024-02-08 15:33:35] ffmpeg.camera2.detect ERROR   : frame=  312 fps= 21 q=-0.0 size=  944662kB time=00:00:12.76 bitrate=606479.2kbits/s dup=0 drop=1 speed=0.845x    
2024-02-08 15:33:35.676975140  [2024-02-08 15:33:35] ffmpeg.camera2.detect ERROR   : frame=  366 fps= 23 q=-0.0 size= 1108688kB time=00:00:14.92 bitrate=608737.8kbits/s dup=0 drop=1 speed=0.956x    
2024-02-08 15:33:35.677174926  [2024-02-08 15:33:35] ffmpeg.camera2.detect ERROR   : frame=  411 fps= 26 q=-0.0 size= 1245375kB time=00:00:16.52 bitrate=617561.3kbits/s dup=0 drop=1 speed=1.03x    
2024-02-08 15:33:35.677384919  [2024-02-08 15:33:35] ffmpeg.camera2.detect ERROR   : frame=  432 fps= 26 q=-0.0 size= 1309162kB time=00:00:17.24 bitrate=622080.0kbits/s dup=0 drop=1 speed=1.04x    
2024-02-08 15:33:35.677586747  [2024-02-08 15:33:35] ffmpeg.camera2.detect ERROR   : frame=  441 fps= 25 q=-0.0 size= 1336500kB time=00:00:17.64 bitrate=620669.4kbits/s dup=0 drop=1 speed=1.02x    
2024-02-08 15:33:35.677778949  [2024-02-08 15:33:35] ffmpeg.camera2.detect ERROR   : frame=  461 fps= 26 q=-0.0 size= 1397250kB time=00:00:18.40 bitrate=622080.0kbits/s dup=0 drop=1 speed=1.03x    
2024-02-08 15:33:35.677971151  [2024-02-08 15:33:35] ffmpeg.camera2.detect ERROR   : frame=  466 fps= 25 q=-0.0 size= 1412438kB time=00:00:18.64 bitrate=620745.1kbits/s dup=0 drop=1 speed=1.02x    
2024-02-08 15:33:35.678168604  [2024-02-08 15:33:35] ffmpeg.camera2.detect ERROR   : frame=  488 fps= 26 q=-0.0 size= 1479262kB time=00:00:19.48 bitrate=622080.0kbits/s dup=0 drop=1 speed=1.03x    
2024-02-08 15:33:35.678365764  [2024-02-08 15:33:35] ffmpeg.camera2.detect ERROR   : frame=  499 fps= 26 q=-0.0 size= 1512675kB time=00:00:19.96 bitrate=620833.3kbits/s dup=0 drop=1 speed=1.03x    
2024-02-08 15:33:35.678553592  [2024-02-08 15:33:35] ffmpeg.camera2.detect ERROR   : frame=  514 fps= 26 q=-0.0 size= 1558238kB time=00:00:20.56 bitrate=620869.7kbits/s dup=0 drop=1 speed=1.03x    
2024-02-08 15:33:35.678749003  [2024-02-08 15:33:35] ffmpeg.camera2.detect ERROR   : frame=  525 fps= 26 q=-0.0 size= 1591650kB time=00:00:21.00 bitrate=620895.1kbits/s dup=0 drop=1 speed=1.03x    
2024-02-08 15:33:35.678938580  [2024-02-08 15:33:35] ffmpeg.camera2.detect ERROR   : frame=  536 fps= 25 q=-0.0 size= 1625062kB time=00:00:21.40 bitrate=622080.0kbits/s dup=0 drop=1 speed=1.01x    
2024-02-08 15:33:35.679126408  [2024-02-08 15:33:35] ffmpeg.camera2.detect ERROR   : frame=  540 fps= 25 q=-0.0 size= 1637212kB time=00:00:21.64 bitrate=619780.3kbits/s dup=0 drop=1 speed=0.986x    
[... cut ...]
2024-02-08 15:33:35.683587313  [2024-02-08 15:33:35] ffmpeg.camera2.detect ERROR   : [out#0/rawvideo @ 0x55a1a47bd0] video:2454300kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: 0.000000%
2024-02-08 15:33:35.683766391  [2024-02-08 15:33:35] ffmpeg.camera2.detect ERROR   : frame=  808 fps= 24 q=-0.0 Lsize= 2454300kB time=00:00:32.28 bitrate=622850.9kbits/s dup=0 drop=1 speed=0.949x    
2024-02-08 15:33:35.683989801  [2024-02-08 15:33:35] ffmpeg.camera2.detect ERROR   : Conversion failed!
```
```

nyanmisaka, 2024-02-09:
Linux error -12 is -ENOMEM (cannot allocate memory).

This is more like RGA2 running out of 4GB dma32 memory than an ffmpeg error. You'd better check dmesg to see the more detailed error info of RGA driver.

great9, 2024-02-09:
```
[2739068.437367] warn_alloc: 82 callbacks suppressed
[2739068.437373] ffmpeg: page allocation failure: order:0, mode:0x140dc6(GFP_HIGHUSER|GFP_DMA32|__GFP_COMP|__GFP_ZERO), nodemask=(null),cpuset=15aae0b0f1d00ff317bd3caf7622267d962dbe6eec8db424c1791fa5434adddb,mems_allowed=0
[2739068.437385] CPU: 7 PID: 3962807 Comm: ffmpeg Tainted: G           O      5.10.110-37-rockchip-g74457be0716d #rockchip
[2739068.437386] Hardware name: Radxa ROCK 5B (DT)
[2739068.437388] Call trace:
[2739068.437394]  dump_backtrace+0x0/0x1a8
[2739068.437396]  show_stack+0x2c/0x38
[2739068.437400]  dump_stack_lvl+0xd0/0xfc
[2739068.437402]  dump_stack+0x14/0x2c
[2739068.437406]  warn_alloc+0xe0/0x174
[2739068.437409]  __alloc_pages_slowpath.constprop.150+0x3f4/0x728
[2739068.437411]  __alloc_pages_nodemask+0x194/0x258
[2739068.437414]  dmabuf_page_pool_alloc+0x78/0x7c
[2739068.437417]  system_heap_do_allocate.isra.11+0x2a4/0x54c
[2739068.437419]  system_heap_allocate+0x38/0x48
[2739068.437421]  dma_heap_buffer_alloc+0x5c/0x74
[2739068.437422]  dma_heap_bufferfd_alloc+0x3c/0x7c
[2739068.437424]  dma_heap_ioctl+0x200/0x284
[2739068.437427]  vfs_ioctl+0x58/0x68
[2739068.437429]  __arm64_sys_ioctl+0x80/0xb0
[2739068.437431]  el0_svc_common.constprop.5+0x134/0x1e8
[2739068.437433]  do_el0_svc+0x80/0x9c
[2739068.437435]  el0_svc+0x1c/0x28
[2739068.437437]  el0_sync_handler+0x64/0x12c
[2739068.437439]  el0_sync+0x178/0x180
[2739068.437441] Mem-Info:
[2739068.437446] active_anon:141204 inactive_anon:161343 isolated_anon:0
                  active_file:16015 inactive_file:22274 isolated_file:0
                  unevictable:0 dirty:193 writeback:0
                  slab_reclaimable:22466 slab_unreclaimable:31709
                  mapped:20568 shmem:26028 pagetables:8850 bounce:0
                  free:234484 free_pcp:746 free_cma:60461
[2739068.437451] Node 0 active_anon:564816kB inactive_anon:645372kB active_file:64060kB inactive_file:89096kB unevictable:0kB isolated(anon):0kB isolated(file):0kB mapped:82272kB dirty:772kB writeback:0kB shmem:104112kB writeback_tmp:0kB kernel_stack:16144kB all_unreclaimable? yes
[2739068.437455] DMA free:247188kB min:5452kB low:9276kB high:13100kB reserved_highatomic:0KB active_anon:44kB inactive_anon:0kB active_file:0kB inactive_file:0kB unevictable:0kB writepending:0kB present:3930112kB managed:3839940kB mlocked:0kB pagetables:948kB bounce:0kB free_pcp:252kB local_pcp:248kB free_cma:241844kB
[2739068.437456] lowmem_reserve[]: 0 0 3941 3941
[2739068.437463] Normal free:690748kB min:5756kB low:9792kB high:13828kB reserved_highatomic:4096KB active_anon:564772kB inactive_anon:645456kB active_file:64060kB inactive_file:89096kB unevictable:0kB writepending:772kB present:4194304kB managed:4036540kB mlocked:0kB pagetables:34452kB bounce:0kB free_pcp:2732kB local_pcp:300kB free_cma:0kB
[2739068.437464] lowmem_reserve[]: 0 0 0 0
[2739068.437468] DMA: 48*4kB (UEC) 28*8kB (UEC) 19*16kB (EC) 173*32kB (UEC) 12*64kB (C) 12*128kB (C) 9*256kB (C) 8*512kB (C) 7*1024kB (C) 6*2048kB (C) 52*4096kB (C) = 247408kB
[2739068.437485] Normal: 38713*4kB (UMEH) 33912*8kB (UMEH) 14447*16kB (UMEH) 1038*32kB (UMEH) 4*64kB (H) 0*128kB 0*256kB 0*512kB 0*1024kB 0*2048kB 0*4096kB = 690772kB
```

nyanmisaka, 2024-02-09:
```
ffmpeg: page allocation failure: order:0, mode:0x140dc6(GFP_HIGHUSER|GFP_DMA32|__GFP_COMP|__GFP_ZERO)
```

The DMA32 address requested by ffmpeg has been exhausted, so it automatically falls back to the DMA64 address. And RGA2 only supports DMA32, so it cannot use this memory.

Try reducing the number of concurrent video streams, and setting `:async_depth=0` for `scale_rkrga` may also help.

great9, 2024-02-09:
> ```
> ffmpeg: page allocation failure: order:0, mode:0x140dc6(GFP_HIGHUSER|GFP_DMA32|__GFP_COMP|__GFP_ZERO)
> ```
>
> The DMA32 address requested by ffmpeg has been exhausted, so it automatically falls [...]

RGA2 had this problem for a while as far as I know. Who's responsible for it? Rockchip?

nyanmisaka, 2024-02-09:
> RGA2 had this problem for a while as far as I know. Who's responsible for it? Rockchip?

This is a hardware limitation and is determined by the hardware designer. So yes rockchip.

RGA3 does not have this limitation but it cannot output YUV420P, you can switch to another format and use RGA3 instead.

closed 2024-02-10 by nyanmisaka

MarcA711, 2025-07-10:
Hey @nyanmisaka,

I used the patch from the developers (I hope... is there some way to check if I use the patched kernel? I don't have much experience with installing custom kernels). I still experience crashes. However, the error changed now:
```
[104914.037083] mali fb000000.gpu: OOM notifier: dev mali0  0 kB
[104914.037949] flexi_logger-fl invoked oom-killer: gfp_mask=0x140cca(GFP_HIGHUSER_MOVABLE|__GFP_COMP), order=0, oom_score_adj=0
```

<details>
  <summary>Full dmesg output</summary>

  
  ```
[104914.037083] mali fb000000.gpu: OOM notifier: dev mali0  0 kB
[104914.037949] flexi_logger-fl invoked oom-killer: gfp_mask=0x140cca(GFP_HIGHUSER_MOVABLE|__GFP_COMP), order=0, oom_score_adj=0
[104914.037969] CPU: 0 PID: 2648 Comm: flexi_logger-fl Not tainted 6.1.115-vendor-rk35xx #1
[104914.037977] Hardware name: Orange Pi 5 Plus (DT)
[104914.037982] Call trace:
[104914.037986]  dump_backtrace+0xe8/0x124
[104914.038000]  show_stack+0x20/0x30
[104914.038007]  dump_stack_lvl+0x6c/0x88
[104914.038017]  dump_stack+0x18/0x34
[104914.038024]  dump_header+0x60/0x210
[104914.038033]  oom_kill_process+0xac/0x170
[104914.038042]  out_of_memory+0x288/0x2b8
[104914.038050]  __alloc_pages+0x6e0/0x834
[104914.038058]  __folio_alloc+0x1c/0x3c
[104914.038064]  folio_alloc.constprop.0+0x1c/0x28
[104914.038071]  __filemap_get_folio+0xf4/0x204
[104914.038079]  filemap_fault+0x27c/0x440
[104914.038085]  __do_fault+0x3c/0x84
[104914.038094]  __handle_mm_fault+0x400/0x704
[104914.038101]  handle_mm_fault+0x144/0x1ac
[104914.038107]  do_page_fault+0x368/0x3a0
[104914.038115]  do_translation_fault+0x64/0x88
[104914.038123]  do_mem_abort+0x48/0x98
[104914.038131]  el0_ia+0x9c/0xc8
[104914.038138]  el0t_64_sync_handler+0xcc/0x13c
[104914.038147]  el0t_64_sync+0x19c/0x1a0
[104914.038154] Mem-Info:
[104914.038161] active_anon:1084951 inactive_anon:1145429 isolated_anon:4
                 active_file:217 inactive_file:93 isolated_file:0
                 unevictable:275105 dirty:0 writeback:0
                 slab_reclaimable:33774 slab_unreclaimable:42666
                 mapped:79331 shmem:333590 pagetables:20591
                 sec_pagetables:0 bounce:0
                 kernel_misc_reclaimable:0
                 free:17426 free_pcp:155 free_cma:4439
[104914.038176] Node 0 active_anon:4339804kB inactive_anon:4581716kB active_file:868kB inactive_file:372kB unevictable:1100420kB isolated(anon):16kB isolated(file):0kB mapped:317324kB dirty:0kB writeback:0kB shmem:1334360kB writeback_tmp:0kB kernel_stack:27120kB pagetables:82364kB sec_pagetables:0kB all_unreclaimable? no
[104914.038190] DMA free:52788kB boost:0kB min:3760kB low:7560kB high:11360kB reserved_highatomic:0KB active_anon:527140kB inactive_anon:757512kB active_file:4kB inactive_file:200kB unevictable:794728kB writepending:0kB present:3930112kB managed:3824956kB mlocked:0kB bounce:0kB free_pcp:496kB local_pcp:0kB free_cma:17756kB
[104914.038204] lowmem_reserve[]: 0 0 12222 12222
[104914.038219] Normal free:16916kB boost:12288kB min:24680kB low:37196kB high:49712kB reserved_highatomic:0KB active_anon:3812756kB inactive_anon:3824332kB active_file:1020kB inactive_file:888kB unevictable:305692kB writepending:0kB present:12838912kB managed:12516184kB mlocked:12kB bounce:0kB free_pcp:124kB local_pcp:0kB free_cma:0kB
[104914.038232] lowmem_reserve[]: 0 0 0 0
[104914.038246] DMA: 437*4kB (UMEC) 471*8kB (UMEC) 181*16kB (UMEC) 76*32kB (UMEC) 296*64kB (UMEC) 113*128kB (UEC) 28*256kB (UE) 2*512kB (E) 0*1024kB 0*2048kB 0*4096kB = 52444kB
[104914.038306] Normal: 2585*4kB (UME) 446*8kB (UME) 193*16kB (UME) 0*32kB 0*64kB 0*128kB 0*256kB 0*512kB 0*1024kB 0*2048kB 0*4096kB = 16996kB
[104914.038350] Node 0 hugepages_total=0 hugepages_free=0 hugepages_surp=0 hugepages_size=1048576kB
[104914.038356] Node 0 hugepages_total=0 hugepages_free=0 hugepages_surp=0 hugepages_size=32768kB
[104914.038362] Node 0 hugepages_total=0 hugepages_free=0 hugepages_surp=0 hugepages_size=2048kB
[104914.038367] Node 0 hugepages_total=0 hugepages_free=0 hugepages_surp=0 hugepages_size=64kB
[104914.038373] 334494 total pagecache pages
[104914.038376] 530 pages in swap cache
[104914.038380] Free swap  = 0kB
[104914.038383] Total swap = 8170568kB
[104914.038388] 4192256 pages RAM
[104914.038391] 0 pages HighMem/MovableOnly
[104914.038395] 106971 pages reserved
[104914.038398] 65536 pages cma reserved
[104914.038402] Tasks state (memory values in pages):
[104914.038406] [  pid  ]   uid  tgid total_vm      rss pgtables_bytes swapents oom_score_adj name
...
[... 3 lines cut ...]
[104914.040161] [ 448050]     0 448050   592640    73156  2510848   159327             0 frigate.detecto
[104914.040170] [ 448067]     0 448067   814044    38100  1384448    35416             0 frigate.output
[104914.040179] [ 448085]     0 448085    30992        0   172032      995             0 ffmpeg
[104914.040188] [ 448092]     0 448092    30992        0   163840      995             0 ffmpeg
[104914.040197] [ 448099]     0 448099    30992        0   163840      996             0 ffmpeg
[104914.040206] [ 448106]     0 448106    88298     2566   249856     1779             0 ffmpeg
[104914.040215] [ 448125]     0 448125   789028    47373  1499136    39123             0 frigate.process
[104914.040224] [ 448127]     0 448127   787646    39758  1486848    44962             0 frigate.process
[104914.040233] [ 448139]     0 448139   788690    39744  1499136    46279             0 frigate.process
[... cut ...] 87ba79b143ccf59b179990fdc73d9cd724609c495e9f086d9d6757559e.scope,task=frigate.capture,pid=448158,uid=0
[104914.040513] Out of memory: Killed process 448158 (frigate.capture) total-vm:20127420kB, anon-rss:7784844kB, file-rss:0kB, shmem-rss:63592kB, UID:0 pgtables:25904kB oom_score_adj:0
[104916.160207] systemd-journald[634]: Under memory pressure, flushing caches.
[104916.507740] oom_reaper: reaped process 448158 (frigate.capture), now anon-rss:0kB, file-rss:0kB, shmem-rss:0kB
  ```
  
</details>

nyanmisaka, 2025-07-10:
How much RAM do you have on your board? It looks like the process killed by the system due to out-of-ram is `frigate.capture`, not ffmpeg.

```
[104914.038406] [  pid  ]   uid  tgid total_vm      rss pgtables_bytes swapents oom_score_adj name
[104914.040242] [ 448158]     0 448158  5031855  1962109 26525696  1258035             0 frigate.capture
```

Total RAM used by `frigate.capture, pid=448158` is 1962109 * 4K / 1024 / 1024 ~= 7.48G

nyanmisaka, 2025-07-10:
There is no good way to tell if a kernel has been patched without calling `printk()`. Usually you can tell by the difference between the build time and the commit date in `dmesg` and `uname -a`.

MarcA711, 2025-07-10:
Thank you very much for your help!

> How much RAM do you have on your board? It looks like the process killed by the system due to out-of-ram is frigate.capture, not ffmpeg.
>
> [104914.038406] [  pid  ]   uid  tgid total_vm      rss [...]

My board has 16G. It was also my initial guess, that the RAM might be really full due to some reason. That's why I monitored my RAM usage in 10 min intervals:

[image]

As you can see, just before the error occurred my RAM usage was 5.5 GB. So maybe there was some issue that caused the allocation of all RAM in less than 10 minutes and caused the crash. However, the log message shows that the capture process that was killed used only around 1.9 GB:

```
[104914.040242] [ 448158]     0 448158  5031855  1962109 26525696  1258035             0 frigate.capture
```

However, it is also weird that one of the three capture processes uses so much more RAM than the other ones (the one that got killed uses 1.9GB, the other ones just 40MB). All cameras are the same with the same resolution.
```
[104914.040242] [ 448158]     0 448158  5031855  1962109 26525696  1258035             0 frigate.capture
[104914.040251] [ 448168]     0 448168   772285    40711  1347584    39282             0 frigate.capture
[104914.040260] [ 448172]     0 448172   790537    42086  1355776    39264             0 frigate.capture
```

Btw. the steady increase might also indicate a memory leak somewhere.

MarcA711, 2025-07-10:
> There is no good way to tell if a kernel has been patched without calling printk(). Usually you can tell by the difference between the build time and the commit date in dmesg and uname -a.

The date is correct. I also used this command that didn't work before the upgrade, now it works:
```
sudo ./ffmpeg -hwaccel rkmpp -hwaccel_output_format drm_prime -afbc rga -i input.mp4 -vf scale_rkrga=format=rgba,hwmap=mode=read,format=rgba -y -f null -
```

nyanmisaka, 2025-07-10:
`rss` needs to be multiplied by the `4kb` pagesize to get the memory size used.

https://stackoverflow.com/questions/9199731/understanding-the-linux-oom-killers-logs

I am not familiar with the memory allocation in frigate. If such a large amount of memory is used, it must be that some memory is not released correctly.