nyanmisaka/ffmpeg-rockchip #26: Scaling affects color output

jagheterfredrik, 2024-02-09:
Hello, thank you for your effort on this 👍 
I'm trying to do hardware scaling `-vf realtime,scale_rkrga=w=1280:h=720:format=yuv420p,hwmap=mode=read,format=yuv420p -r 5 -f rawvideo` but my image turns all weird. Without the w/h (`-vf realtime,scale_rkrga=format=yuv420p,hwmap=mode=read,format=yuv420p -r 5 -f rawvideo`) the video looks fine.

[image]

Full ffmpeg is
```
ffmpeg -threads 2 -hwaccel rkmpp -hwaccel_output_format drm_prime -avoid_negative_ts make_zero -fflags +genpts+discardcorrupt -rtsp_transport tcp -timeout 5000000 -use_wallclock_as_timestamps 1 -i rtsp://127.0.0.1:8554/front -f segment -segment_time 10 -segment_format mp4 -reset_timestamps 1 -strftime 1 -c:v copy -c:a aac /tmp/cache/framsida@%Y%m%d%H%M%S%z.mp4 -vf realtime,scale_rkrga=w=1280:h=720:format=yuv420p,hwmap=mode=read,format=yuv420p -r 5 -f rawvideo -y pipe:
```
Input format is
```
Input #0, rtsp, from 'rtsp://....':
  Stream #0:0: Video: h264 (High), yuvj420p(pc, bt709, progressive), 1920x1080, 15 fps, 30 tbr, 90k tbn
  Stream #0:1: Audio: pcm_alaw, 8000 Hz, mono, s16, 64 kb/s
Stream mapping:
  Stream #0:0 -> #0:0 (h264 (h264_rkmpp) -> rawvideo (native))
```
Any pointers are highly appreciated 🙏

nyanmisaka, 2024-02-10:
@jagheterfredrik 
I can't reproduce your issue. Please dump the rawvideo as file and check with the [YUView tool](https://github.com/IENT/YUView).

```console
ffmpeg -hwaccel rkmpp -hwaccel_output_format drm_prime -i INPUT -an -sn \
-vf scale_rkrga=format=yuv420p:w=1280:h=720,hwmap=mode=read,format=yuv420p \
-f rawvideo -vframes 500 -y /tmp/out.yuv
```

[image]

jagheterfredrik, 2024-02-10:
Thanks a bunch for the pointer to YUView! It's being displayed in the wrong resolution and therefore not showing correctly. At the right size it shows fine after scaling. Thank you again for your work on ffmpeg+rockchip 👍

closed 2024-02-10 by jagheterfredrik