nyanmisaka/ffmpeg-rockchip #28: Tone-mapping slow after upgrade kernel.

FirstMelody, 2024-02-22:
After upgrading the kernel from 5.1 to 6.1 (orangepi-linux latest kernel),tone-mapping got super slow even upgraded opencl driver.
Both latest and v1.9-1-3381696 is not functioning correctly.

clinfo output:
https://pastebin.com/eH0YEEfB

Jellyfin-ffmpeg log:
https://pastebin.com/fpUN4JMG

docker-compose.yml:
https://pastebin.com/jrbsDvrK

nyanmisaka, 2024-02-22:
OpenCL performance that is lower than expected can usually be caused by two reasons:

- Mali CSF firmware version and bifrost kernel driver version do not match. See also #24
- The power supply output power is insufficient, causing the GPU to throttle or reduce frequency.

FirstMelody, 2024-02-22:
> OpenCL performance that is lower than expected can usually be caused by two reasons:
> 
> * Mali CSF firmware version and bifrost kernel driver version do not match. See also [更换了Armbian_23.11.1_Orangepi5-plus_bookworm_legacy_5.10.160.img后好像hevc 10解码失效了😭 #24](https://github.com/nyanmisaka/ffmpeg-rockchip/issues/24)
> * The power supply output power is insufficient, causing the GPU to throttle or reduce frequency.

Thanks for your prompt reply!

I've checked the issue and tried three version of mali csf firmware(I am using valhall), latest and two versions you mentioned in the issue.
And power supply is also unlikely the cause of the issue for that it was performing normally before upgrading and it was came with orangepi5

nyanmisaka, 2024-02-22:
I have been using the 6.1 kernel for three months and have not encountered similar issues. Also, since the guy in the thread I linked to solved the problem on his OrangePi 5 Plus, I think it's not an issue with either the GPU driver or ffmpeg.

The video source file is large in size and has a high bit rate. Consider this a network connection speed or drive speed issue.

FirstMelody, 2024-02-22:
Qr

> I have been using the 6.1 kernel for three months and have not encountered similar issues. Also, since the guy in the thread I linked to solved the problem on his OrangePi 5 Plus, I think it's not [...]

The video I used for testing is a vidoe

FirstMelody, 2024-02-22:
I wrongly clicked the close issue button.
The video file I used is a movie I watched with transcoding before. And it was fine. Also, disabling tone-mapping would make the transcode speed back to normal (90fps ish). I am so confused right now. Maybe I need a clean install?

nyanmisaka, 2024-02-22:
I got ~55fps on 4k to 4k with tone-mapping and PGS burn-in enabled.
[image]
 
I use my own kernel configuration. Maybe orangepi has wrong kernel defconfig enabled for 6.1. You can compare both. https://github.com/nyanmisaka/linux-rockchip/commits/rk-6.1-dev2

FirstMelody, 2024-03-11:
Sorry for the delay. I have been so busy recently. And I've re-flashed the orangepi back to 5.10 kernel and everything is back to normal. And I didn't see any updates of the kernel from orangepi team. And armbian is also not releasing their kernel. So I guess this issue should be closed. Anyway, thanks for your contribution!

closed 2024-03-11 by FirstMelody