nyanmisaka/ffmpeg-rockchip #33: Use `h264_rkmpp` play rtsp stream, avcodec_receive_frame allways return -11.

Coloryr, 2024-03-05:
When debugging with breakpoints, it was found that `avcodec_receive_frame` allways return -11.
But using `h264` is normal, it can display.
```
try
{
    ffmpeg.av_frame_unref(_videoFrame);
    ffmpeg.av_frame_unref(_audioFrame);
    ffmpeg.av_frame_unref(_receivedFrame);
    int old = 0;
    byte* data = null;
    AVRational time_base = _pFormatContext->streams[_streamIndex]->time_base;
    long start_time = 0;
    bool first_frame = true;
    do
    {
        ffmpeg.av_packet_unref(_pPacket);
        error = ffmpeg.av_read_frame(_pFormatContext, _pPacket);

        if (error == ffmpeg.AVERROR_EOF || cancellationToken.IsCancellationRequested)
        {
            return;
        }

        error.ThrowExceptionIfError();

        if (_pPacket->stream_index == _audioStreamIndex)
        {
            ffmpeg.avcodec_send_packet(_audioCodecContext, _pPacket)
                .ThrowExceptionIfError();

            error = ffmpeg.avcodec_receive_frame(_audioCodecContext, _audioFrame);
            if (error == 0 && IPHostApi.EnableAudio)
            {
                BufferFormat format;
                var frame = *_audioFrame;
                if (frame.ch_layout.nb_channels == 1)
                {
                    format = frame.format == (int)AVSampleFormat.AV_SAMPLE_FMT_U8 ? BufferFormat.Mono8 : BufferFormat.Mono16;
                }
                else if (frame.ch_layout.nb_channels == 2)
                {
                    format = frame.format == (int)AVSampleFormat.AV_SAMPLE_FMT_U8 ? BufferFormat.Stereo8 : BufferFormat.Stereo16;
                }
                else
                {
                    continue;
                }

                int outputSampleCount = (int)ffmpeg.av_rescale_rnd(ffmpeg.swr_get_delay(swr_ctx, frame.sample_rate) + frame.nb_samples, 48000, frame.sample_rate, AVRounding.AV_ROUND_UP)
                    * 2 * 2;

                if (old != outputSampleCount)
                {
                    if (data != null)
                    {
                        Marshal.FreeHGlobal((nint)data);
                    }

                    data = (byte*)Marshal.AllocHGlobal(outputSampleCount);
                    old = outputSampleCount;
                }

                int size = ffmpeg.swr_convert(swr_ctx, &data, outputSampleCount, frame.extended_data, frame.nb_samples) * 4;

                if (size > 0 && IPHostApi.EnableAudio)
                {
                    OpenALPlayer.Write(format, data, size, 48000);
                }
            }
            else if (error != -11)
            {
                error.ThrowExceptionIfError();
            }
        }
        else if (_pPacket->stream_index == _streamIndex)
        {
            ffmpeg.avcodec_send_packet(_videoCodecContext, _pPacket)
                .ThrowExceptionIfError();
            error = ffmpeg.avcodec_receive_frame(_videoCodecContext, _videoFrame);
            if (error == 0)
            {
                AVFrame frame = *_videoFrame;

                if (_videoCodecContext->hw_device_ctx != null)
                {
                    ffmpeg.av_hwframe_transfer_data(_receivedFrame, _videoFrame, 0)
                        .ThrowExceptionIfError();
                    frame = *_receivedFrame;
                }

                var convertedFrame = vfc.Convert(frame);
                IPHostApi.SetVideoPtr(uuid, convertedFrame.width, convertedFrame.height, (IntPtr)convertedFrame.data[0]);

                if (first_frame)
                {
                    first_frame = false;
                    start_time =  ffmpeg.av_gettime() - ffmpeg.av_rescale_q(_videoFrame->pts, time_base, ffmpeg.av_make_q(1, ffmpeg.AV_TIME_BASE));
                }
            }
            else if (error != -11)
            {
                error.ThrowExceptionIfError();
            }
        }

        if (!first_frame)
        {
            long pts_time = ffmpeg.av_rescale_q(_videoFrame->pts, time_base, ffmpeg.av_make_q(1, ffmpeg.AV_TIME_BASE));
            long now = ffmpeg.av_gettime() - start_time;
            if (pts_time > now)
            {
                ffmpeg.av_usleep((uint)(pts_time - now));
            }
        }
    } while (!cancellationToken.IsCancellationRequested);
}
catch (Exception e)
{
    Console.WriteLine(e);
}
```

Coloryr, 2024-03-05:
Use ffplay test is

nyanmisaka, 2024-03-06:
-11 is `AVERROR(EAGAIN)`. This means you need to feed more packets to get a new frame.
You'd better learn how to handle `AVERROR(EAGAIN)` from the official examples.

https://github.com/FFmpeg/FFmpeg/blob/master/doc/examples/decode_video.c

Coloryr, 2024-03-06:
But enough data has already been thrown in, use `h264` will get the result soon.
And not this issus when playing videos with higher bitrates.

Coloryr, 2024-03-06:
[image]

Coloryr, 2024-03-06:
I tried to pull RTMP at a high bit rate, but there was no issus with it.
后面我试着拉rtmp高码率也没这个问题

nyanmisaka, 2024-03-06:
Can u test the video playback in ffplay locally without using rstp?

```
ffplay -vcodec h264_rkmpp -i movie.mp4
```

Coloryr, 2024-03-06:
> Can u test the video playback in ffplay locally without using rstp?
> 
> ```
> ffplay -vcodec h264_rkmpp -i movie.mp4
> ```

Is work now.

nyanmisaka, 2024-03-06:
Better to use `-v debug` and see what's going on.

Coloryr, 2024-03-06:
```
orangepi@orangepi3b:~$ ffplay -vcodec h264_rkmpp -i rtsp://192.168.12.9:10054/test -v debug
ffplay version d43f4f5 Copyright (c) 2003-2023 the FFmpeg developers
  built with gcc 11 (Ubuntu 11.4.0-1ubuntu1~22.04)
  configuration: --prefix=/usr --enable-gpl --enable-version3 --enable-libdrm --enable-rkmpp --enable-rkrga --enable-sdl2 --enable-ffmpeg --enable-ffplay --enable-ffprobe --enable-shared --disable-static
  libavutil      58. 29.100 / 58. 29.100
  libavcodec     60. 31.102 / 60. 31.102
  libavformat    60. 16.100 / 60. 16.100
  libavdevice    60.  3.100 / 60.  3.100
  libavfilter     9. 12.100 /  9. 12.100
  libswscale      7.  5.100 /  7.  5.100
  libswresample   4. 12.100 /  4. 12.100
  libpostproc    57.  3.100 / 57.  3.100
xcb_connection_has_error() returned true
Initialized opengl renderer.
[tcp @ 0x7f68003690] No default whitelist set 0KB sq=    0B f=0/0
[tcp @ 0x7f68003690] Original list of addresses:
[tcp @ 0x7f68003690] Address 192.168.12.9 port 10054
[tcp @ 0x7f68003690] Interleaved list of addresses:
[tcp @ 0x7f68003690] Address 192.168.12.9 port 10054
[tcp @ 0x7f68003690] Starting connection attempt to 192.168.12.9 port 10054
[tcp @ 0x7f68003690] Successfully connected to 192.168.12.9 port 10054
[rtsp @ 0x7f68000c20] SDP:
v=0
o=- 0 0 IN IP6 ::1
s=No Name
c=IN IP6 ::1
t=0 0
a=tool:libavformat 60.22.101
m=video 0 RTP/AVP 96
b=AS:80
a=rtpmap:96 H264/90000
a=fmtp:96 packetization-mode=1; sprop-parameter-sets=Z0LADZ4hgoP01BgEGQAAfa8AHR4ejxQq4A==,aM4yyA==; profile-level-id=42C00D
a=control:streamid=0
m=audio 0 RTP/AVP 97
b=AS:115
a=rtpmap:97 MPEG4-GENERIC/48000/2
a=fmtp:97 profile-level-id=1;mode=AAC-hbr;sizelength=13;indexlength=3;indexdeltalength=3; config=1190
a=control:streamid=1

[rtsp @ 0x7f68000c20] video codec set to: h264
[rtsp @ 0x7f68000c20] RTP Packetization Mode: 1
[rtsp @ 0x7f68000c20] Extradata set to 0x7f680054f0 (size: 37)
[rtsp @ 0x7f68000c20] RTP Profile IDC: 42 Profile IOP: c0 Level: d
[rtsp @ 0x7f68000c20] audio codec set to: aac
[rtsp @ 0x7f68000c20] audio samplerate set to: 48000
[rtsp @ 0x7f68000c20] audio channels set to: 2
[rtp @ 0x7f68009390] No default whitelist set
[udp @ 0x7f68009690] No default whitelist set
[udp @ 0x7f68009690] end receive buffer size reported is 425984
[udp @ 0x7f68019960] No default whitelist set
[udp @ 0x7f68019960] end receive buffer size reported is 425984
[rtsp @ 0x7f68000c20] setting jitter buffer size to 500
[rtp @ 0x7f68029ec0] No default whitelist set 0KB sq=    0B f=0/0
[udp @ 0x7f6802a160] No default whitelist set
[udp @ 0x7f6802a160] end receive buffer size reported is 425984
[udp @ 0x7f6803a430] No default whitelist set
[udp @ 0x7f6803a430] end receive buffer size reported is 425984
[rtsp @ 0x7f68000c20] setting jitter buffer size to 500
[rtsp @ 0x7f68000c20] hello state=0
[h264 @ 0x7f68005020] nal_unit_type: 7(SPS), nal_ref_idc: 3
[h264 @ 0x7f68005020] nal_unit_type: 8(PPS), nal_ref_idc: 3
[h264 @ 0x7f68005020] Decoding VUI 0KB vq=    0KB sq=    0B f=0/0
For transform of length 64, inverse, mdct_float, flags: [aligned, out_of_place], found 2 matches:
    1: mdct_inv_float_c - type: mdct_float, len: [2, ∞], factors[2]: [2, any], flags: [unaligned, out_of_place, inv_only], prio: 96
    2: mdct_naive_inv_float_c - type: mdct_float, len: [2, ∞], factors[2]: [2, any], flags: [unaligned, out_of_place, inv_only], prio: -130976
For transform of length 32, inverse, fft_float, flags: [aligned, inplace, preshuf], found 3 matches:
    1: fft32_ns_float_neon - type: fft_float, len: 32, factor: 2, flags: [aligned, inplace, out_of_place, preshuf], prio: 352
    2: fft_pfa_ns_float_c - type: fft_float, len: [6, ∞], factors[2]: [7, 5, 3, 2, any], flags: [unaligned, inplace, out_of_place, preshuf], prio: 112
    3: fft32_ns_float_c - type: fft_float, len: 32, factor: 2, flags: [unaligned, inplace, out_of_place, preshuf], prio: 96
[... 325 lines cut ...]
[h264_rkmpp @ 0x7f680f1590] Timeout getting decoded frame0B f=0/0
[h264_rkmpp @ 0x7f680f1590] Wrote 90 bytes to decoder
[h264_rkmpp @ 0x7f680f1590] Timeout getting decoded frame0B f=0/0
[h264_rkmpp @ 0x7f680f1590] Wrote 322 bytes to decoder
[h264_rkmpp @ 0x7f680f1590] Timeout getting decoded frame0B f=0/0
[h264_rkmpp @ 0x7f680f1590] Wrote 312 bytes to decoder
[h264_rkmpp @ 0x7f680f1590] Timeout getting decoded frame0B f=0/0
[h264_rkmpp @ 0x7f680f1590] Wrote 336 bytes to decoder
[h264_rkmpp @ 0x7f680f1590] Timeout getting decoded frame0B f=0/0
[h264_rkmpp @ 0x7f680f1590] Wrote 442 bytes to decoder
[h264_rkmpp @ 0x7f680f1590] Timeout getting decoded frame0B f=0/0
[h264_rkmpp @ 0x7f680f1590] Wrote 81 bytes to decoder
[h264_rkmpp @ 0x7f680f1590] Timeout getting decoded frame0B f=0/0
[h264_rkmpp @ 0x7f680f1590] Wrote 299 bytes to decoder
[h264_rkmpp @ 0x7f680f1590] Timeout getting decoded frame0B f=0/0
[h264_rkmpp @ 0x7f680f1590] Wrote 226 bytes to decoder
[h264_rkmpp @ 0x7f680f1590] Timeout getting decoded frame0B f=0/0
[h264_rkmpp @ 0x7f680f1590] Wrote 386 bytes to decoder
[h264_rkmpp @ 0x7f680f1590] Timeout getting decoded frame0B f=0/0
[h264_rkmpp @ 0x7f680f1590] Wrote 402 bytes to decoder
[h264_rkmpp @ 0x7f680f1590] Timeout getting decoded frame0B f=0/0
[h264_rkmpp @ 0x7f680f1590] Wrote 82 bytes to decoder
[h264_rkmpp @ 0x7f680f1590] Timeout getting decoded frame0B f=0/0
[h264_rkmpp @ 0x7f680f1590] Wrote 217 bytes to decoder
```

nyanmisaka, 2024-03-06:
Can u test https://github.com/JeffyCN/FFmpeg and see if it helps? It's a fork from rockchip maintainer.

Coloryr, 2024-03-06:
https://github.com/JeffyCN/FFmpeg Is work
```
orangepi@orangepi3b:~/FFmpeg$ ./ffplay -vcodec h264_rkmpp -i rtsp://192.168.12.9:10054/test -v debug
ffplay version git-2023-05-12-003204d Copyright (c) 2003-2022 the FFmpeg developers
  built with gcc 11 (Ubuntu 11.4.0-1ubuntu1~22.04)
  configuration: --prefix=/usr --enable-gpl --enable-version3 --enable-libdrm --enable-rkmpp --enable-sdl2 --enable-ffmpeg --enable-ffplay --enable-ffprobe
  libavutil      57. 33.101 / 57. 33.101
  libavcodec     59. 42.101 / 59. 42.101
  libavformat    59. 30.100 / 59. 30.100
  libavdevice    59.  8.101 / 59.  8.101
  libavfilter     8. 46.103 /  8. 46.103
  libswscale      6.  8.102 /  6.  8.102
  libswresample   4.  8.100 /  4.  8.100
  libpostproc    56.  7.100 / 56.  7.100
xcb_connection_has_error() returned true
Initialized opengl renderer.
[tcp @ 0x7f84003690] No default whitelist set 0KB sq=    0B f=0/0
[tcp @ 0x7f84003690] Original list of addresses:
[tcp @ 0x7f84003690] Address 192.168.12.9 port 10054
[tcp @ 0x7f84003690] Interleaved list of addresses:
[tcp @ 0x7f84003690] Address 192.168.12.9 port 10054
[tcp @ 0x7f84003690] Starting connection attempt to 192.168.12.9 port 10054
[tcp @ 0x7f84003690] Successfully connected to 192.168.12.9 port 10054
[rtsp @ 0x7f84000c20] SDP:
v=0
o=- 0 0 IN IP6 ::1
s=No Name
c=IN IP6 ::1
t=0 0
a=tool:libavformat 60.22.101
m=video 0 RTP/AVP 96
b=AS:80
a=rtpmap:96 H264/90000
a=fmtp:96 packetization-mode=1; sprop-parameter-sets=Z0LADZ4hgoP01BgEGQAAfa8AHR4ejxQq4A==,aM4yyA==; profile-level-id=42C00D
a=control:streamid=0
m=audio 0 RTP/AVP 97
b=AS:115
a=rtpmap:97 MPEG4-GENERIC/48000/2
a=fmtp:97 profile-level-id=1;mode=AAC-hbr;sizelength=13;indexlength=3;indexdeltalength=3; config=1190
a=control:streamid=1

[rtsp @ 0x7f84000c20] video codec set to: h264
[rtsp @ 0x7f84000c20] RTP Packetization Mode: 1
[rtsp @ 0x7f84000c20] Extradata set to 0x7f840054f0 (size: 37)
[rtsp @ 0x7f84000c20] RTP Profile IDC: 42 Profile IOP: c0 Level: d
[rtsp @ 0x7f84000c20] audio codec set to: aac
[rtsp @ 0x7f84000c20] audio samplerate set to: 48000
[rtsp @ 0x7f84000c20] audio channels set to: 2
[rtp @ 0x7f84009120] No default whitelist set
[udp @ 0x7f84009420] No default whitelist set
[udp @ 0x7f84009420] end receive buffer size reported is 425984
[udp @ 0x7f840196f0] No default whitelist set
[udp @ 0x7f840196f0] end receive buffer size reported is 425984
[rtsp @ 0x7f84000c20] setting jitter buffer size to 500
[rtp @ 0x7f84029c50] No default whitelist set
[udp @ 0x7f84029ef0] No default whitelist set
[udp @ 0x7f84029ef0] end receive buffer size reported is 425984
[udp @ 0x7f8403a1c0] No default whitelist set
[udp @ 0x7f8403a1c0] end receive buffer size reported is 425984
[rtsp @ 0x7f84000c20] setting jitter buffer size to 500
[rtsp @ 0x7f84000c20] hello state=0
[h264 @ 0x7f84005020] nal_unit_type: 7(SPS), nal_ref_idc: 3
[h264 @ 0x7f84005020] nal_unit_type: 8(PPS), nal_ref_idc: 3
[h264 @ 0x7f84005020] nal_unit_type: 7(SPS), nal_ref_idc: 3 f=0/0
[h264 @ 0x7f84005020] nal_unit_type: 8(PPS), nal_ref_idc: 3
[h264 @ 0x7f84005020] nal_unit_type: 5(IDR), nal_ref_idc: 3 f=0/0
[h264 @ 0x7f84005020] Format yuv420p chosen by get_format().
[h264 @ 0x7f84005020] Reinit context to 320x240, pix_fmt: yuv420p
[h264 @ 0x7f84005020] nal_unit_type: 1(Coded slice of a non-IDR picture), nal_ref_idc: 2
    Last message repeated 2 times
[... 299 lines cut ...]
[h264_rkmpp @ 0x7f840af2a0] Doing slow software conversion
[aac @ 0x7f840c4df0] skip whole frame, skip left: 0
[h264_rkmpp @ 0x7f840af2a0] Wrote 4656 bytes to decoder  0B f=0/0
[h264_rkmpp @ 0x7f840af2a0] Received a frame.
[h264_rkmpp @ 0x7f840af2a0] Doing slow software conversion
[h264_rkmpp @ 0x7f840af2a0] Received a frame. 3KB sq=    0B f=0/0
[h264_rkmpp @ 0x7f840af2a0] Doing slow software conversion
[h264_rkmpp @ 0x7f840af2a0] Received a frame. 4KB sq=    0B f=0/0
[h264_rkmpp @ 0x7f840af2a0] Doing slow software conversion
[h264_rkmpp @ 0x7f840af2a0] Received a frame. 4KB sq=    0B f=0/0
[h264_rkmpp @ 0x7f840af2a0] Doing slow software conversion
[h264_rkmpp @ 0x7f840af2a0] Wrote 366 bytes to decoder   0B f=0/0
[h264_rkmpp @ 0x7f840af2a0] Wrote 87 bytes to decoder
[h264_rkmpp @ 0x7f840af2a0] Wrote 180 bytes to decoder
[h264_rkmpp @ 0x7f840af2a0] Wrote 253 bytes to decoder
[h264_rkmpp @ 0x7f840af2a0] Buffer full
[h264_rkmpp @ 0x7f840af2a0] Received a frame.
[h264_rkmpp @ 0x7f840af2a0] Doing slow software conversion
[h264_rkmpp @ 0x7f840af2a0] Wrote 260 bytes to decoder   0B f=0/0
[h264_rkmpp @ 0x7f840af2a0] Received a frame.
[h264_rkmpp @ 0x7f840af2a0] Doing slow software conversion
[h264_rkmpp @ 0x7f840af2a0] Received a frame. 3KB sq=    0B f=0/0
[h264_rkmpp @ 0x7f840af2a0] Doing slow software conversion
^Corangepi@orangepi3b:~/FFmpeg$    9KB vq=    4KB sq=    0B f=0/0
```

nyanmisaka, 2024-03-06:
Can u try this patch d6a05bf?

Coloryr, 2024-03-06:
> Can u try this patch [d6a05bf](https://github.com/nyanmisaka/ffmpeg-rockchip/commit/d6a05bf90476549d626d13d58c66d87cd9f4301b)?

Nope, it's still `timeout`

nyanmisaka, 2024-03-06:
Pls try this branch https://github.com/nyanmisaka/ffmpeg-rockchip/commits/refactor-dec-logic/

Coloryr, 2024-03-06:
Is work now

nyanmisaka, 2024-03-06:
@Coloryr Can u try one more change and see if it still works? https://github.com/nyanmisaka/ffmpeg-rockchip/commit/9d14c5f878621669e522326b46355481d0c1dbcb

hbiyik, 2024-03-07:
@nyanmisaka may be it is usefull to see -v trace output, i think the crucial part is buffer full msg which is logged in trace level.

https://github.com/nyanmisaka/ffmpeg-rockchip/blob/d43f4f54e6c732cd47d5e1ab69b600afd8966897/libavcodec/rkmppdec.c#L840

If there are decoder is buffer full, then that dynamic  buffer estimation might be sus.

https://github.com/nyanmisaka/ffmpeg-rockchip/blob/d43f4f54e6c732cd47d5e1ab69b600afd8966897/libavcodec/rkmppdec.c#L881

Still speculating though.. @Coloryr  is it possible you dump the rtmp stream to a file, and upliad somewgere, so that we can reproduce?

Ps: sent from mobile.

Coloryr, 2024-03-07:
> @nyanmisaka may be it is usefull to see -v trace output, i think the crucial part is buffer full msg which is logged in trace level. [...]

I think it's rk_mpp decoder issus, the stream can play ues VLC or other player.

nyanmisaka, 2024-03-07:
> > @Coloryr Can u try one more change and see if it still works? [9d14c5f](https://github.com/nyanmisaka/ffmpeg-rockchip/commit/9d14c5f878621669e522326b46355481d0c1dbcb)
> 
> Yes, it work.

Thanks for the testing.

@hbiyik I’m thinking of switching back to the jeffyCN’s send-packet/get-frame logic. Because it finishes the task more elegantly and has lower decoding latency (even lower than what I handled in #8).

hbiyik, 2024-03-07:
i see, i remember jeffys loop had some implications in throughput and cpu load, and in some cases is spikes massively due to infinite loop, thats why i had switched to an algo which is high in throughput and high in latency and low in cpu usage. So there might be some trade offs there.

But my old fork was doing backlog management by its own unlike this fork, things might have changed, therefore it might be reasonable to have some benchmarks there.

nyanmisaka, 2024-03-07:
> i see, i remember jeffys loop had some implications in throughput and cpu load, and in some cases is spikes massively due to infinite loop, thats why i had switched to an algo which is high in throughput and high in latency and low in cpu usage. So there might be some trade offs there.
> 
> But my old fork was doing backlog management by its own unlike this fork, things might have changed, therefore it might be reasonable to have some benchmarks there.
> 
> Related: [JeffyCN/FFmpeg#18 (comment)](https://github.com/JeffyCN/FFmpeg/issues/18#issuecomment-1529167815) [JeffyCN/FFmpeg#18 (comment)](https://github.com/JeffyCN/FFmpeg/issues/18#issuecomment-1784135052)

Yes there is a trade-off here. I'd like to keep the rkmppdec latency close to RK gstreamer, otherwise there will be tons of user complaints.

Can you test the branch I mentioned above to see if it satisfies your requirements?

hbiyik, 2024-03-07:
yes, i will mos def, i got the hang of it.

hbiyik, 2024-03-15:
@nyanmisaka is there an ffmpeg way of measuring the decoder latency, i was in past using hand made print statements according to my taste, in the decoder and post processing them with excel, but it would be great if i could have some universal method of benchmarking the latency...

nyanmisaka, 2024-03-15:
IMHO, installing an RTSP server App on your Android phone and letting ffmpeg connect to the phone as a client is currently the most effective way to detect latency that I have found. You only need to compare ffplay on the computer side with the real-time recording on the phone screen.

hbiyik, 2024-03-15:
ok then ill setup an rtsp server and add some logging in the decoder to measure the decoder latency, it is really necessary to graph the latency by time to compare i guess, visual feedback can be deceiving.

Coloryr, 2024-03-26:
It coming again with rtmp
```
ffplay -vcodec h264_rkmpp rtmp://192.168.12.10:1935/live/video -v debug
```

```
orangepi@orangepi3b:~$ ffplay -vcodec h264_rkmpp rtmp://192.168.12.10:1935/live/video -v debug
ffplay version 9501bc0 Copyright (c) 2003-2023 the FFmpeg developers
  built with gcc 11 (Ubuntu 11.4.0-1ubuntu1~22.04)
  configuration: --prefix=/home/orangepi/ffmpeg --enable-gpl --enable-version3 --enable-libdrm --enable-rkmpp --enable-rkrga --enable-ffmpeg --enable-ffplay --enable-ffprobe --enable-avdevice --enable-indev=v4l2 --enable-indev=alsa --enable-outdev=alsa --enable-libmp3lame --enable-openal --enable-shared --disable-static
  libavutil      58. 29.100 / 58. 29.100
  libavcodec     60. 31.102 / 60. 31.102
  libavformat    60. 16.100 / 60. 16.100
  libavdevice    60.  3.100 / 60.  3.100
  libavfilter     9. 12.100 /  9. 12.100
  libswscale      7.  5.100 /  7.  5.100
  libswresample   4. 12.100 /  4. 12.100
  libpostproc    57.  3.100 / 57.  3.100
xcb_connection_has_error() returned true
Initialized opengl renderer.
[AVFormatContext @ 0x7f80000c20] Opening 'rtmp://192.168.12.10:1935/live/video' for reading
[rtmp @ 0x7f80001240] No default whitelist set
[tcp @ 0x7f800017d0] No default whitelist set
[tcp @ 0x7f800017d0] Original list of addresses:
[tcp @ 0x7f800017d0] Address 192.168.12.10 port 1935
[tcp @ 0x7f800017d0] Interleaved list of addresses:
[tcp @ 0x7f800017d0] Address 192.168.12.10 port 1935
[tcp @ 0x7f800017d0] Starting connection attempt to 192.168.12.10 port 1935
[tcp @ 0x7f800017d0] Successfully connected to 192.168.12.10 port 1935
[rtmp @ 0x7f80001240] Handshaking...
[rtmp @ 0x7f80001240] Type answer 3
[rtmp @ 0x7f80001240] Server version 13.14.10.13
[rtmp @ 0x7f80001240] Proto = rtmp, path = /live/video, app = live, fname = video
[rtmp @ 0x7f80001240] Window acknowledgement size = 5000000
[rtmp @ 0x7f80001240] Max sent, unacked = 5000000 sq=    0B f=0/0
[rtmp @ 0x7f80001240] New incoming chunk size = 1024
[rtmp @ 0x7f80001240] Creating stream...
[rtmp @ 0x7f80001240] Sending play command for 'video'   0B f=0/0
[flv @ 0x7f80000c20] Format flv probed with size=2048 and score=100
[flv @ 0x7f80000c20] Before avformat_find_stream_info() pos: 13 bytes read:2054 seeks:0 nb_streams:0
[flv @ 0x7f80010de0] Format yuv420p chosen by get_format(). f=0/0
[flv @ 0x7f80010de0] Format yuv420p chosen by get_format(). f=0/0
[flv @ 0x7f80000c20] All info found0KB vq=    0KB sq=    0B f=0/0
[flv @ 0x7f80000c20] rfps: 29.750000 0.014687
[flv @ 0x7f80000c20] rfps: 29.833333 0.006555
    Last message repeated 1 times
[flv @ 0x7f80000c20] rfps: 29.916667 0.001682
[flv @ 0x7f80000c20] rfps: 30.000000 0.000067
    Last message repeated 1 times
[flv @ 0x7f80000c20] rfps: 60.000000 0.000270
    Last message repeated 1 times
[flv @ 0x7f80000c20] rfps: 120.000000 0.001079
    Last message repeated 1 times
[flv @ 0x7f80000c20] rfps: 240.000000 0.004316
[flv @ 0x7f80000c20] rfps: 29.970030 0.000273
    Last message repeated 1 times
[flv @ 0x7f80000c20] rfps: 59.940060 0.001091
    Last message repeated 1 times
[flv @ 0x7f80000c20] After avformat_find_stream_info() pos: 75374 bytes read:75374 seeks:0 frames:122
Input #0, flv, from 'rtmp://192.168.12.10:1935/live/video':
  Metadata:
    |RtmpSampleAccess: true
    Server          : NGINX RTMP (github.com/arut/nginx-rtmp-module)
    displayWidth    : 640
    displayHeight   : 480
    fps             : 30
    profile         :
    level           :
  Duration: 00:00:00.00, start: 7.655000, bitrate: N/A
  Stream #0:0, 81, 1/1000: Audio: mp3, 48000 Hz, stereo, fltp, 128 kb/s
[... 196 lines cut ...]
[h264_rkmpp @ 0x7f8007f7e0] Wrote 299 bytes to decoder
[h264_rkmpp @ 0x7f8007f7e0] Wrote 191 bytes to decoder   0B f=0/0
[h264_rkmpp @ 0x7f8007f7e0] Wrote 229 bytes to decoder   0B f=0/0
[h264_rkmpp @ 0x7f8007f7e0] Wrote 209 bytes to decoder   0B f=0/0
[h264_rkmpp @ 0x7f8007f7e0] Wrote 228 bytes to decoder   0B f=0/0
[h264_rkmpp @ 0x7f8007f7e0] Wrote 233 bytes to decoder
[h264_rkmpp @ 0x7f8007f7e0] Wrote 342 bytes to decoder   0B f=0/0
[h264_rkmpp @ 0x7f8007f7e0] Wrote 254 bytes to decoder
[h264_rkmpp @ 0x7f8007f7e0] Wrote 256 bytes to decoder   0B f=0/0
[h264_rkmpp @ 0x7f8007f7e0] Wrote 289 bytes to decoder
[h264_rkmpp @ 0x7f8007f7e0] Wrote 8065 bytes to decoder  0B f=0/0
[h264_rkmpp @ 0x7f8007f7e0] Wrote 493 bytes to decoder   0B f=0/0
[h264_rkmpp @ 0x7f8007f7e0] Wrote 344 bytes to decoder   0B f=0/0
[h264_rkmpp @ 0x7f8007f7e0] Wrote 201 bytes to decoder   0B f=0/0
[h264_rkmpp @ 0x7f8007f7e0] Wrote 216 bytes to decoder   0B f=0/0
[h264_rkmpp @ 0x7f8007f7e0] Wrote 240 bytes to decoder   0B f=0/0
[h264_rkmpp @ 0x7f8007f7e0] Wrote 406 bytes to decoder   0B f=0/0
[h264_rkmpp @ 0x7f8007f7e0] Wrote 426 bytes to decoder
[h264_rkmpp @ 0x7f8007f7e0] Wrote 444 bytes to decoder   0B f=0/0
[h264_rkmpp @ 0x7f8007f7e0] Wrote 378 bytes to decoder   0B f=0/0
[h264_rkmpp @ 0x7f8007f7e0] Wrote 224 bytes to decoder   0B f=0/0
[h264_rkmpp @ 0x7f8007f7e0] Wrote 210 bytes to decoder   0B f=0/0
[h264_rkmpp @ 0x7f8007f7e0] Wrote 8016 bytes to decoder  0B f=0/0
^CSegmentation fault
```

hbiyik, 2024-04-09:
@nyanmisaka sorry i will not have much time soon to look in to this, if you think it is ok, may be you can just merge jeffy's decoder loop to main.

closed 2024-04-24 by nyanmisaka