nyanmisaka/ffmpeg-rockchip #34: lost the last frame in complex filter

mcerveny, 2024-03-08:
Hello.
I found that complex filter (double RGA filter) lost the last frame. `in.ts` has 100 frames.
- single RGA filter (overlay or scale)
  - `./ffmpeg -hwaccel rkmpp -hwaccel_output_format drm_prime -i in.ts -i ovr.png -filter_complex '[0:v]hwupload[main];[1:v]hwupload[overlay];[main][overlay]overlay_rkrga=eof_action=repeat:repeatlast=true:format=nv12' -c:v h264_rkmpp out.ts -y` --> 100 frames
  -  `./ffmpeg -hwaccel rkmpp -hwaccel_output_format drm_prime -i in.ts  -filter_complex 'hwupload,scale_rkrga=w=1280:h=720' -c:v h264_rkmpp  out.ts -y` --> 100 frames
- double RGA filter (overlay and scale)
  - `./ffmpeg -hwaccel rkmpp -hwaccel_output_format drm_prime -i in.ts -i ovr.png -filter_complex '[0:v]hwupload[main];[1:v]hwupload[overlay];[main][overlay]overlay_rkrga=eof_action=repeat:repeatlast=true:format=nv12,scale_rkrga=w=1280:h=720' -c:v h264_rkmpp  out.ts -y` --> **99 frames**

nyanmisaka, 2024-03-09:
Good catch. In my use case `overlay_rkrga` is always the last filter so I didn't notice this.

nyanmisaka, 2024-03-09:
@mcerveny 27136b05e0742450a019d6bab80d840ac5fd6421 should fix this issue.

mcerveny, 2024-03-09:
Yes ! resolved problem. Thank you.

closed 2024-03-09 by mcerveny