nyanmisaka/ffmpeg-rockchip #38: consider supporting RV1126？

geiliwanghaichao, 2024-03-15:
Integrating Rockchip's encoding and decoding SDK into ffmpeg is very convenient to use. not sure if author has considered supporting RV1126     Thanks

nyanmisaka, 2024-03-15:
Is there's any difference between RV1126 and the RK35xx in term of MPP support? The MPP interface should be platform agnostic.
- https://github.com/rockchip-linux/mpp/blob/ee946af015c350c926a0fca2f02da8e429f8b079/doc/Rockchip_Developer_Guide_MPP_EN.md?plain=1#L79
- https://github.com/rockchip-linux/mpp/blob/ee946af015c350c926a0fca2f02da8e429f8b079/osal/mpp_soc.cpp#L814-L823

Also, I have no plans to purchase the RV1126 board at the moment. If you make modifications to make this project work on RV1126, please open a PR.

geiliwanghaichao, 2024-03-15:
yes. MPP is platform independent. I just tried to compile on rv1126 and received an error message: libdrm not found using pkg config. I thought the source code did not support rv1126. Next, I need to solve the issue of compilation errors    
Thanks

nyanmisaka, 2024-03-25:
Closing as it's not an issue.

closed 2024-03-25 by nyanmisaka