nyanmisaka/ffmpeg-rockchip #39: Green screen with the attached video sample

kyak, 2024-03-15:
Originally reported at https://github.com/7Ji-PKGBUILDs/.meta/issues/29

I'm using `kodi-mpp-git` and `ffmpeg-rockchip-git`.

With the attached video sample, `mpv --profile=fast --hwdec=rkmpp path-to-file` displays green screen.

`kodi` under GBM (https://github.com/hbiyik/ffmpeg-rockchip/wiki/Rendering#kodi-under-gbm) seems to play it, although displays black screen after stopping the video.

After stopping the video, kodi shows black screen (it doesn't freeze, it really is just a blank or black screen).

I see these messages in journalctl:

```
kernel: [drm:rockchip_vop2_mod_supported] *ERROR* Unsupported format modifier 0x800000000000051
```

And these messages in kodi log:

```
2024-03-14 21:28:00.101 T:328297   error <general>: CDRMAtomic::DrmAtomicCommit - test commit failed: (Invalid argument) - falling back to last successful atomic request
2024-03-14 21:28:00.102 T:328297   error <general>:
                                                   Object: plane        ID: 71
                                                     Property: SRC_X    ID: 9   Value: 0
                                                     Property: SRC_Y    ID: 10  Value: 0
                                                     Property: SRC_W    ID: 11  Value: 251658240
                                                     Property: SRC_H    ID: 12  Value: 141557760
                                                     Property: CRTC_X   ID: 13  Value: 0
                                                     Property: CRTC_Y   ID: 14  Value: 0
                                                     Property: CRTC_W   ID: 15  Value: 3840
                                                     Property: CRTC_H   ID: 16  Value: 2160
                                                     Property: FB_ID    ID: 17  Value: 322
                                                     Property: CRTC_ID  ID: 20  Value: 85
2024-03-14 21:28:00.102 T:328297   error <general>: CDRMAtomic::DrmAtomicCommit - atomic commit failed: Invalid argument
```

Sorry for gzip'ed mkv, couldn't upload to github otherwise :(

[out.mkv.gz](https://github.com/7Ji-PKGBUILDs/.meta/files/14606172/out.mkv.gz)

hbiyik, 2024-03-15:
any your kernel is?

kyak, 2024-03-15:
> and your kernel is?

It's 5.10.160

nyanmisaka, 2024-03-15:
This clip itself is abnormal. When I play it on other x86 PCs, it also behaves very choppy. When I use ffmpeg-rockchip to transcode on RK3588, the problem after transcoding is still very choppy, but there is no green screen here.

hbiyik, 2024-03-15:
dont worry, i reproduced the issue, 

true that, the file is acting weird, but blue screen is because of the dimensions

w = 912, 
stride = 57*16 (this is way mpp is using align to 16) = 912
but mesa (EGL/mpv) expect 912 to be 64 dividable, 912/64 = 14.25, so wont work

however gbm should still work, seems like there is something weird there, i m trying find out if this is kernel related or xbmc related....

hbiyik, 2024-03-15:
to prevent kodi/gbm issues in the meanwhile you can use without afbc mdoe as below

`kodi --windowing=gbm --audio-backend=alsa`

it should work (as long the file is not corrupted, the file seems to be corrupted)

do not add env variable `FFMPEG_RKMPP_DEC_OPT="afbc=on" `, there seems to be an issue there.

nyanmisaka, 2024-03-15:
I think I know why. I've been using the mpp patch for Jellyfin to avoid 64-aligned issues when using RGA3. Maybe you should add these two to the repository too.

https://github.com/nyanmisaka/mpp/commit/6e76ecb50ab85b7c3de9368903e687b6aa21aad2
https://github.com/nyanmisaka/mpp/commit/6221ca89d9837935edbc56af46cf651bd3c5b6cf

hbiyik, 2024-03-15:
ok what if vp9 or hevc?

nyanmisaka, 2024-03-15:
iirc vp9 and hevc should both be 256-aligned. I forgot mpeg1/2/4, will add it later.

hbiyik, 2024-03-15:
i think not only, only when above certain size, at least for either vp9 or hevc afaik...

hbiyik, 2024-03-15:
linking the issue for future reference https://github.com/rockchip-linux/mpp/issues/507

hbiyik, 2024-03-15:
Found the root cause of GBM problem

https://github.com/xbmc/xbmc/pull/24431#issuecomment-2000680604

i  will push a fix, in the meanwhile you can disable afbc mode...

hbiyik, 2024-03-17:
@kyak 
i pushed a fix for xbmc here: https://github.com/hbiyik/xbmc/tree/gbm_drm_dynamic_afbc_video_planes_v2
can you test it?

if you do not know how to test it here is a quick recipe.

``` shell
# download the PKGBUILD of kodi-ext-git to clean directory
yay -G kodi-ext-git 
# go to dir
cd kodi-ext-git
# edit the PKGBUILD
nano PKGBUILD
# replace the source git repo to my test repo
# the line source=("git+https://github.com/xbmc/xbmc.git#branch=$_codename")
# should be source=("git+https://github.com/hbiyik/xbmc.git#branch=gbm_drm_dynamic_afbc_video_planes_v2")

# build the package
makepkg -s

# you dont have to install navigate to build directory
cd src/kodi-build

# and execute the kodi.bin locally
FFMPEG_RKMPP_DEC_OPT="afbc=on" ./kodi.bin --windowing=gbm --audio-backend=alsa
```

kyak, 2024-03-18:
@hbiyik although i build `kodi-ext-git` master branch on a regular basic, i couldn't build from your branch. Maybe it's because your branch is lagging from master branch? The error is the following:

```
[ 33%] Building CXX object build/music/tags/CMakeFiles/music_tags.dir/TagLibVFSStream.cpp.o
In file included from /home/user/build/kodi-ext-git/src/xbmc/xbmc/music/tags/TagLibVFSStream.cpp:8:
/home/user/build/kodi-ext-git/src/xbmc/xbmc/music/tags/TagLibVFSStream.h:40:34: error: «TagLib::ulong» was not declared
   40 |     TagLib::ByteVector readBlock(TagLib::ulong length) override;

....
```

hbiyik, 2024-03-18:
@kyak  yeah it is lagging but thats fine.

you can just add
```shell
-DENABLE_INTERNAL_TAGLIB=ON
```

somewhere around line 89~109, in to the `_args` section as a new line.

also please delete the

```shell
 [[ -d kodi-build ]] && rm -rf kodi-build
```

in line 70, under `prepare()` this will help to build faster next time, because the PKGBUILD will not delete the previous build and continue where it as left.

kyak, 2024-03-18:
@hbiyik thanks for the hints!

Unfortunately, it doesn't really work. Most of the videos don't play (although I found one that plays).

For example, `Big_Buck_Bunny_1080_10s_10MB.mp4` displays black screen.

kyak, 2024-03-18:
Without FFMPEG_RKMPP_DEC_OPT="afbc=on", it seems to work, albeit weird.

For example, it displays black screen behind OSD (i.e. OSD is drawn on top of black screen). When OSD is not active, the video displays normally.

hbiyik, 2024-03-18:
Weird tried it last night it was working, i will cgeck again.

Just to confirm, does ur kernel have below path?
0002-vop2_rbga2101010_capability_fix.patch

Which kernel package are you using?

kyak, 2024-03-18:
> Weird tried it last night it was working, i will check again.
> 
> Just to confirm, does ur kernel have below path? 0002-vop2_rbga2101010_capability_fix.patch
> 
> Which kernel package are you using?

I'm using `linux-aarch64-rockchip-rk3588-bsp5.10-orangepi-git` from 7Ji's repo.

> 
> Also you can you provide output of `drm_info` ?
> 
> Drm info pkg is on aur: https://aur.archlinux.org/packages/drm_info

[drm_info.txt](https://github.com/nyanmisaka/ffmpeg-rockchip/files/14641345/drm_info.txt)

hbiyik, 2024-03-18:
ok your kernel is not patched correctly

```
    │       ├───"IN_FORMATS" (immutable): blob = 58
    │       │   ├───ARM_AFBC(BLOCK_SIZE = 16x16) (0x800000000000001)
    │       │   │   ├───XRGB2101010 (0x30335258)
    │       │   │   ├───ARGB2101010 (0x30335241)
    │       │   │   ├───XBGR2101010 (0x30334258)
    │       │   │   ├───ABGR2101010 (0x30334241)
```

there should be no XRGB2101010 in drm formats... This is a bug confirmed by rockchip https://github.com/nyanmisaka/ffmpeg-rockchip/pull/4#issuecomment-1892934552

now we need a new ticket in 7ji-pkhbuilds. :) i will open it for you :)

kyak, 2024-03-19:
> @kyak [7Ji-PKGBUILDs/.meta#31](https://github.com/7Ji-PKGBUILDs/.meta/issues/31)

@hbiyik back to you here. As [mentioned](https://github.com/7Ji-PKGBUILDs/.meta/issues/31#issuecomment-2007938972), i'm now running the patched kernel:

```
    │       ├───"IN_FORMATS" (immutable): blob = 58
    │       │   ├───ARM_AFBC(BLOCK_SIZE = 16x16) (0x800000000000001)
    │       │   │   ├───XRGB8888 (0x34325258)
    │       │   │   ├───ARGB8888 (0x34325241)
    │       │   │   ├───XBGR8888 (0x34324258)
    │       │   │   ├───ABGR8888 (0x34324241)
    │       │   │   ├───RGB888 (0x34324752)
    │       │   │   ├───BGR888 (0x34324742)
    │       │   │   ├───RGB565 (0x36314752)
    │       │   │   ├───BGR565 (0x36314742)
    │       │   │   ├───YUV420_8BIT (0x38305559)
    │       │   │   ├───YUV420_10BIT (0x30315559)
    │       │   │   ├───YUYV (0x56595559)
    │       │   │   └───Y210 (0x30313259)
```

I need to mention that `kodi-mpp-git` from 7Ji's repo doesn't work with `FFMPEG_RKMPP_DEC_OPT="afbc=on"` nicely. It displays black screen instead of OSD and this is above the video. But I guess this is expected. Without the env var, it works fine.

As for `kodi-ext-git` built from your branch, it still doesn't work well. With the env var, depending on video, it displays OSD on top of black screen or a spinner (audio plays, video doesn't). Without the env var, it plays some videos fine (those with OSD on black screen with env variable) and shows spinner with audio on other videos.

hbiyik, 2024-03-20:
thanks for the feedback, i suggest the following

since we are on different hardwares we need to reduce the software variantions to minimum.

To even simplify the cases, lets focus 5.10 kernel, because 6.1 kernel has several differences in display driver of rockchip (vop2) and i dont want to create 6.1 related issues to impact, we we make sure 5.10 work, we can reiterate the test on 6.1

1)
so lets use https://github.com/7Ji-PKGBUILDs/linux-aarch64-rockchip-bsp5.10-joshua/blob/master/PKGBUILD.
first thing to do, patch this with
https://patch-diff.githubusercontent.com/raw/Joshua-Riek/linux-rockchip/pull/8.patch

2) 
lets focus on 1 single file and 1 single obvious problem, and that should not be [out.mkv.gz](https://github.com/7Ji-PKGBUILDs/.meta/files/14606172/out.mkv.gz), because this file is already broken. So you will see black screen with this file not because rendering is wrong but because the file is broken and ffmpeg cant decode properly.

If we can reproduce 1 single issue, i assume the root cause of it will fix the rest of the issues.

From my end, i have onyl tested this with 5.10 radxa kernel, so may be i will directly see the issue on joshua 5.10..

kyak, 2024-03-20:
> thanks for the feedback, i suggest the following
> 
> since we are on different hardwares we need to reduce the software variantions to minimum.

I'm running Orange Pi 5 Plus (have both 8 Gb and 16 Gb versions). What is your hardware?

> 
> To even simplify the cases, lets focus 5.10 kernel, because 6.1 kernel has several differences in display driver of rockchip (vop2) and i dont want possible 6.1 vop2 changes related issues to interfere, so lets make sure 5.10 works first. Later we can reiterate the test on 6.1..
> 
> 
> so lets use https://github.com/7Ji-PKGBUILDs/linux-aarch64-rockchip-bsp5.10-joshua/blob/master/PKGBUILD. first thing to do, patch this with https://patch-diff.githubusercontent.com/raw/Joshua-Riek/linux-rockchip/pull/8.patch
> 

This is clear, will do.

> lets focus on 1 single file and 1 single obvious problem, and that should not be [out.mkv.gz](https://github.com/7Ji-PKGBUILDs/.meta/files/14606172/out.mkv.gz), because this file is already broken. [...]

Also clear, I will find you the file that's properly broken :) Thing is, I've split this one from original video using ffmpeg and might've broken something in the video file along the way.

> 
> lets only test https://github.com/hbiyik/xbmc/tree/gbm_drm_dynamic_afbc_video_planes_v2 as xbmc, because i already noticed a problem in the first implementation (the one in the ffmpeg-mpp-git). And for you building this should be reletively easy may be in 2 or 3 mins, we can test a new change. We can forget ffmpeg-mpp-git for now until we clearify all the issues.
> 

It's clear I should test `kodi-ext-git` with your branch `gbm_drm_dynamic_afbc_video_planes_v2`. But what do you mean that we should forget about `ffmpeg-mpp-git`? Should I simply keep using the binary package of `ffmpeg-mpp-git` from 7Ji's repo? Should I still set the env var when testing? Do I need to point kodi to the system `ffmpeg-mpp-git` or it will use it automatically (since it's `-ext` package)?

> If we can reproduce 1 single issue, i assume the root cause of it will fix the rest of the issues.
> 
> From my end, i have onyl tested this with 5.10 radxa kernel, so may be i will directly see the issue on joshua 5.10.. we will see

hbiyik, 2024-03-20:
> What is your hardware?

i am on radxa 5b with 4g.

> But what do you mean that we should forget about `ffmpeg-mpp-git`

i mean we should forget about ffmpeg-mpp-git durinf testing gbm workflow, when we fix those issues in gbm_drm_dynamic_afbc_video_planes_v2, i will port them to ``ffmpeg-mpp-git`. You can keep it installed, and use it daily, as far as i know it works without the `afbc=on` so you can daily use it.

for testing and verifying the kodi-ext-git, point the source to gbm_drm_dynamic_afbc_video_planes_v2 and run `make pkg` but dont install the generated packages. Instead go to src/xbmc/kodi-build and run `./kodi.bin` for testing.

Whenever i push a fix, you will run makepkg again, and it will compile with the latest from my branch so that you can verify that it is fixed in opi5 as well. When we make sure it works fine, i can then port the changes to `kodi-mpp-git` and mainline `xbmc` repo with PR saying that it is tested.

hbiyik, 2024-03-20:
@kyak 
here is a patched joshua's 5.10 kernel https://file.io/y7zSI9ZVm1Bl
i have tested and everything seems to work fine with my rock5b

kyak, 2024-03-20:
> @kyak here is a patched joshua's 5.10 kernel https://file.io/y7zSI9ZVm1Bl i have tested and everything seems to work fine with my rock5b

Don't worry, it's almost built on my side. Will report back tomorrow.

kyak, 2024-03-21:
@hbiyik 

I'm now running `linux-aarch64-rockchip-bsp5.10-joshua` with the patch. I also built `kodi-ext-git` from your branch. I'm running kodi in GBM mode, with Direct to Plane rendering.

With `afbc=on`, i get black screen instead of video. OSD displays fine.

Without `afbc=on`, the video plays fine. OSD displays fine on top of the video.

Is this expected?

hbiyik, 2024-03-21:
no this is not expected, it should play in afbc mode aswell.

could you restart the device and provide the full output of `drm_info`. It is better for me to see the default drm state, therefore i need a clean start.

Then run kodi with afbc mode in gbm, make sure u enabled [debug mode ](https://kodi.wiki/view/Log_file/Easy#Steps). It is very important debug mode to be enabled otherwise i wont see anything.

Run the video which is resulting the black video output. Close kodi, and send the log file at `~/.kodi/temp.kodi.log`

Also, for me to reproduce can you send the file that produces black video?

kyak, 2024-03-21:
Here's the drm_info output after the reboot: 
[drm_info.txt](https://github.com/nyanmisaka/ffmpeg-rockchip/files/14705710/drm_info.txt)

Here's the verbose log: 
[kodi.log](https://github.com/nyanmisaka/ffmpeg-rockchip/files/14705713/kodi.log)

And here is the video file: https://file.io/QNZMTBNLMCxi

hbiyik, 2024-03-21:
ok something obviously is wrong according to your logs, let me investigate this a little bit.

`CDRMUtils::FindVideoPlane - Can not find a Video Plane plane for format 942691673, modifier 576460752303423553`

hbiyik, 2024-03-21:
@kyak is it possible to boot into https://github.com/7Ji-PKGBUILDs/linux-aarch64-rockchip-bsp6.1-joshua-git kernel, and provide the `drm_info` output.

I understood the problem, the soc rk3588s (used in opi5) and rk3588 (used in rock 5b) provides different drm layout, this is a little bit weird, to me looks like even a display driver (vop2) bug in kernel.

PS: no need to patch joshua6.1 all patches are already in there
I would like to compare my devices drm tree with your opi5, i think it should give some more hints.

kyak, 2024-03-22:
@hbiyik 

Here's the drm_info for 6.1: 
[drm_info.txt](https://github.com/nyanmisaka/ffmpeg-rockchip/files/14718369/drm_info.txt)

Please note that i'm running Orange Pi 5 Plus, which is rk3588, not rk3588s.

I also checked the situation with 6.1 kernel and it seems like it didn't change anything.

hbiyik, 2024-03-22:
@kyak 
could you replace your dtb with the dtb [rk3588-orangepi-5-plus.dtb.tar.gz](https://github.com/nyanmisaka/ffmpeg-rockchip/files/14727502/rk3588-orangepi-5-plus.dtb.tar.gz on joshua's 5.10 kernel, and run `drm_info` and paste the results?

the location should be:
`/boot/dtbs/linux-aarch64-rockchip-bsp5.10-joshua/rockchip/rk3588-orangepi-5-plus.dtb`

the actual change is [here](https://github.com/hbiyik/linux-rockchip/commit/83d0b9469244ce2d0a8f141ad973ced46f2240a9)

kyak, 2024-03-24:
@hbiyik thanks so much for your patience.

This is `drm_info` output from joshua's 5.10 kernel with your modified dtb: 
[drm_info.txt](https://github.com/nyanmisaka/ffmpeg-rockchip/files/14736006/drm_info.txt)

Using modified dtb still didn't change anything

kyak, 2024-03-27:
@hbiyik in case this was lost  in discussion, i want to point out that there is a configuration that works fine for me. I.e. all kinds of videos plays fine, OSD displays fine and this is with GBM and Direct to Plane mode of Kodi.

The configuration is:

* `linux-aarch64-rockchip-rk3588-bsp5.10-orangepi-git` with two patches:
[0001-rga3_uncompact_fix.patch](https://github.com/hbiyik/ffmpeg-rockchip/wiki/patches/rockchip-kernel/0001-rga3_uncompact_fix.patch)
[0002-vop2_rbga2101010_capability_fix.patch](https://github.com/hbiyik/ffmpeg-rockchip/wiki/patches/rockchip-kernel/0002-vop2_rgba2101010_capability_fix.patch)
* `ffmpeg-mpp-git` binary package
* `kodi-mpp-git` binary package, no environment variables

Maybe with can start from this configuration and figure out what the problem is?

hbiyik, 2024-03-27:
@kyak 
Even in this configuration kodi with gbm mode and afbc enabled with  FFMPEG_RKMPP_DEC_OPT="afbc=on should cause problems on opi5 variants.

I think i got the underlying problem, and i have something in my mind to solve it with xbmc.

Currently im on vacation may be next week i will prototype something, than i might need some testinv support.

hbiyik, 2024-08-28:
closing this to clean up historical discussions