nyanmisaka/ffmpeg-rockchip #41: SEI information support

MapleAura, 2024-03-18:
[image]

[image]

I set var of side_data for AVFrame with a value of “side_data”. I debugged the code and found that MPP didn't use this value when encoding. Is it  MPP doesn't support setting side_data?

barryzheng, 2025-04-14:
Has this issue been resolved，if anyone have good solutions? The demo supports SEI insertion with mpp_meta_set_ptr(meta, KEY_USER_DATA, &user_data);

nyanmisaka, 2025-04-15:
> Has this issue been resolved，if anyone have good solutions? The demo supports SEI insertion with mpp_meta_set_ptr(meta, KEY_USER_DATA, &user_data); Thandks

There is an experimental branch but the MPP runtime messes up the packet/SEI orders.

https://github.com/nyanmisaka/ffmpeg-rockchip/commits/udu-sei/

nyanmisaka, 2025-04-22:
Close in https://github.com/nyanmisaka/ffmpeg-rockchip/commit/5898b091ee6f7d833fa704cdde7a951df8e1b600

closed 2025-04-22 by nyanmisaka