nyanmisaka/ffmpeg-rockchip #44: Test filter error

BlackJulySnow, 2024-03-23:
chip:rk3566
board: [lubancat zero](https://doc.embedfire.com/linux/rk356x/quick_start/zh/latest/quick_start/lubancat/lubancat-zero.html#lubancat-zero-n)

Encoder and Decoder test normal

```
root@lubancat:~/dev/ffmpeg# cat /proc/version
Linux version 4.19.232 (jiawen@dev120.embedfire.local) (gcc version 6.3.1 20170404 (Linaro GCC 6.3-2017.05), GNU ld (Linaro_Binutils-2017.05) 2.27.0.20161019) #17 SMP Fri Mar 15 15:24:13 CST 2024

root@lubancat:~/dev/ffmpeg# ls /dev/rga -lh
crw-rw---- 1 root video 10, 58 Mar 21 17:00 /dev/rga

root@lubancat:~/dev/ffmpeg# ./ffmpeg -init_hw_device rkmpp=hw -filter_hw_device hw -f lavfi -i testsrc2=s=1920x1080,format=nv12 \
-vf hwupload,scale_rkrga=w=1280:h=720:format=nv12 -c:v h264_rkmpp -b:v 4M -maxrate 4M -vframes 1000 -y /tmp/tmp.mp4
ffmpeg version 1fa586a Copyright (c) 2000-2023 the FFmpeg developers
  built with gcc 11 (Ubuntu 11.4.0-1ubuntu1~22.04)
  configuration: --prefix=/usr --enable-gpl --enable-version3 --enable-libdrm --enable-rkmpp --enable-rkrga --enable-libass --enable-libfreetype --enable-filter=drawtext --enable-filter=subtitles --enable-filter=subtitles --enable-libharfbuzz
  libavutil      58. 29.100 / 58. 29.100
  libavcodec     60. 31.102 / 60. 31.102
  libavformat    60. 16.100 / 60. 16.100
  libavdevice    60.  3.100 / 60.  3.100
  libavfilter     9. 12.100 /  9. 12.100
  libswscale      7.  5.100 /  7.  5.100
  libswresample   4. 12.100 /  4. 12.100
  libpostproc    57.  3.100 / 57.  3.100
Input #0, lavfi, from 'testsrc2=s=1920x1080,format=nv12':
  Duration: N/A, start: 0.000000, bitrate: N/A
  Stream #0:0: Video: wrapped_avframe, nv12, 1920x1080 [SAR 1:1 DAR 16:9], 25 fps, 25 tbr, 25 tbn
Stream mapping:
  Stream #0:0 -> #0:0 (wrapped_avframe (native) -> h264 (h264_rkmpp))
Press [q] to stop, [?] for help
librga fail to get driver version! Compatibility mode will be enabled.

47 im2d_rga_impl rga_version_below_minimun_range_user_driver(304): The driver may be compatible, but it is best to update the driver to version 1.2.4. You can try to update the SDK or update the <SDK>/kernel/drivers/video/rockchip/rga3 directory individually. current version: librga 1.9.2, driver .
rga_api version 1.9.2_[1]
[Parsed_scale_rkrga_1 @ 0x55669235f0] RGA async blit returned invalid fence_fd: 0
[vf#0:0 @ 0x5566920eb0] Error while filtering: Generic error in an external library
Failed to inject frame into filter network: Generic error in an external library
Error while filtering: Generic error in an external library
[out#0/mp4 @ 0x556691f1f0] Nothing was written into output file, because at least one of its streams received no packets.
frame=    0 fps=0.0 q=0.0 Lsize=       0kB time=N/A bitrate=N/A speed=N/A    
Conversion failed!
```

nyanmisaka, 2024-03-23:
> librga fail to get driver version! Compatibility mode will be enabled.

> 47 im2d_rga_impl rga_version_below_minimun_range_user_driver(304): The driver may be compatible, but it is best to update the driver to version 1.2.4. You can try to update the SDK or update the <SDK>/kernel/drivers/video/rockchip/rga3 directory individually. current version: librga 1.9.2, driver .
rga_api version 1.9.2_[1]

The error messages are self-explanatory. You have to upgrade either the linux kernel to 5.10, or upgrade the RGA kernel driver.

BlackJulySnow, 2024-03-23:
Thanks! 
However, I did not find the system image of 5.10 kernel version or the related upgrade tutorial in the system image provided by the official website.
It would be appreciated if you could provide some information about the upgrade

closed 2024-04-04 by nyanmisaka