nyanmisaka/ffmpeg-rockchip #45: `Internal bug, should not have happened` 在同时编码与解码的时候

Coloryr, 2024-03-25:
```
System.ApplicationException: Internal bug, should not have happened
   at IPHost.Desktop.FFmpegHelper.ThrowExceptionIfError(Int32 error) in E:\code\IPHost\IPHost.Desktop\FFmpegHelper.cs:line 76
   at IPHost.Desktop.VideoDecoder.<.ctor>b__28_0() in E:\code\IPHost\IPHost.Desktop\VideoDecoder.cs:line 192
Aborted
```

```
if (packs.TryDequeue(out var packet))
{
    var data = (byte*)ffmpeg.av_malloc((ulong)packet.Size);
    Marshal.Copy(packet.Data, 0, (IntPtr)data, packet.Size);

    ffmpeg.av_packet_from_data(_pPacket, data, packet.Size)
        .ThrowExceptionIfError();

    ffmpeg.avcodec_send_packet(_videoCodecContext, _pPacket)
        .ThrowExceptionIfError();

    //ffmpeg.av_packet_unref(_pPacket);
    ffmpeg.av_free(data);

    error = ffmpeg.avcodec_receive_frame(_videoCodecContext, _videoFrame);
    if (error == 0)
    {
        AVFrame frame = *_videoFrame;

        if (_videoCodecContext->hw_device_ctx != null)
        {
            ffmpeg.av_hwframe_transfer_data(_receivedFrame, _videoFrame, 0)
                .ThrowExceptionIfError();
            frame = *_receivedFrame;
            if (first_frame)
            {
                first_frame = false;
            }
        }

        if (vfc == null)
        {
            var sourceSize = FrameSize;
            var sourcePixelFormat = HWDeviceType == AVHWDeviceType.AV_HWDEVICE_TYPE_NONE
            ? _videoCodecContext->pix_fmt
                     : DecodeHelper.GetHWPixelFormat(HWDeviceType);
            var destinationSize = sourceSize;
            var destinationPixelFormat = AVPixelFormat.AV_PIX_FMT_BGRA;
            vfc = new VideoFrameConverter(sourceSize, sourcePixelFormat, destinationSize, destinationPixelFormat);
        }

        var convertedFrame = vfc.Convert(frame);
        IPHostApi.SetVideoPtr(_uuid, convertedFrame.width, convertedFrame.height, (IntPtr)convertedFrame.data[0]);

        start_time = ffmpeg.av_gettime() - ffmpeg.av_rescale_q(_videoFrame->pts, TimeBase, ffmpeg.av_make_q(1, ffmpeg.AV_TIME_BASE));
    }
    else if (error != -11)
    {
        error.ThrowExceptionIfError();
    }
}
else
{
    Thread.Sleep(1);
}
```

Coloryr, 2024-03-25:
如果只是单一编码或者解码没这个问题，而且这个问题是概率发生

nyanmisaka, 2024-03-25:
Not reproducible via FFmpeg CLI transcoding. More information is needed to locate which line in FFmpeg is throwing this error `AVERROR_BUG`. There's [indeed one](https://github.com/nyanmisaka/ffmpeg-rockchip/blob/1fa586ac7fa224bde0100def5e5f3406f6c2e663/libavcodec/rkmppdec.c#L797C23-L797C34) in `rkmppdec.c` but it is used as an assertion.

nyanmisaka, 2024-04-24:
The problem cannot be reproduced through the CLI. My guess is that in the downstream application, the encoder takes up too many slots in the decoder buffer pool and the decoder is stuck.

closed 2024-04-24 by nyanmisaka