nyanmisaka/ffmpeg-rockchip #47: use scale_rkrga filter  error

gaop921766, 2024-03-26:
cmd: ./ffmpeg -timeout 10000000 -rtsp_transport tcp -loglevel debug -c:v h264_rkmpp -i rtsp://xxxx:xxxx@192.168.97.60/h264/ch2/main/av_stream -vf fps=4,scale_rkrga=-1:720 -c:v h264_rkmpp -global_quality 19  -g 16 -rw_timeout 5000000 -f flv -an rtmp://127.0.0.1/cams/live

log:
h264_rkmpp @ 0x55ba0f6360] Noticed an info change
[h264_rkmpp @ 0x55ba0f6360] Format nv12 chosen by get_format().
[h264_rkmpp @ 0x55ba0f6360] Configured with size: 1920x1080 | pix_fmt: nv12 | sw_pix_fmt: nv12
[h264_rkmpp @ 0x55ba0f6360] Received a frame
[AVFilterGraph @ 0x55b9f214c0] Setting 'fps' to value '4'
[AVFilterGraph @ 0x55b9f214c0] Setting 'w' to value '-1'
[AVFilterGraph @ 0x55b9f214c0] Setting 'h' to value '720'
detected 8 logical cores
[graph 0 input from stream 0:0 @ 0x55b9f52bb0] Setting 'video_size' to value '1920x1080'
[graph 0 input from stream 0:0 @ 0x55b9f52bb0] Setting 'pix_fmt' to value '23'
[graph 0 input from stream 0:0 @ 0x55b9f52bb0] Setting 'time_base' to value '1/90000'
[graph 0 input from stream 0:0 @ 0x55b9f52bb0] Setting 'pixel_aspect' to value '0/1'
[graph 0 input from stream 0:0 @ 0x55b9f52bb0] Setting 'frame_rate' to value '25/1'
[graph 0 input from stream 0:0 @ 0x55b9f52bb0] w:1920 h:1080 pixfmt:nv12 tb:1/90000 fr:25/1 sar:0/1
[format @ 0x55b9f52f70] Setting 'pix_fmts' to value 'gray|yuv420p|yuv422p|yuv444p|nv12|nv21|nv16|nv24|yuyv422|yvyu422|uyvy422|rgb24|bgr24|rgba|rgb0|bgra|bgr0|argb|0rgb|abgr|0bgr|drm_prime'
[auto_scale_0 @ 0x55b9f53b90] w:iw h:ih flags:'' interl:0
[Parsed_scale_rkrga_1 @ 0x55b9f52b00] auto-inserting filter 'auto_scale_0' between the filter 'Parsed_fps_0' and the filter 'Parsed_scale_rkrga_1'
Impossible to convert between the formats supported by the filter 'Parsed_fps_0' and the filter 'auto_scale_0'
[Parsed_fps_0 @ 0x55b9f733b0] 0 frames in, 0 frames out; 0 frames dropped, 0 frames duplicated.
Error reinitializing filters!
Failed to inject frame into filter network: Function not implemented
Error while processing the decoded data for stream #0:0
[AVIOContext @ 0x55ba0d2aa0] Statistics: 0 bytes written, 0 seeks, 0 writeouts
[rtmp @ 0x55b9f7c4f0] UnPublishing stream...
[rtmp @ 0x55b9f7c4f0] Deleting stream...
Terminating demuxer thread 0
Conversion failed!

nyanmisaka, 2024-03-26:
```
[Parsed_scale_rkrga_1 @ 0x55b9f52b00] auto-inserting filter 'auto_scale_0' between the filter 'Parsed_fps_0' and the filter 'Parsed_scale_rkrga_1'
Impossible to convert between the formats supported by the filter 'Parsed_fps_0' and the filter 'auto_scale_0'
```

This is **NOT** a software filter. You **MUST** insert a `hwupload` before it and a `hwdownload/hwmap` filter after it.

Read through the Wiki before opening issues.
- https://github.com/nyanmisaka/ffmpeg-rockchip/wiki/Video-Transcode
- https://github.com/nyanmisaka/ffmpeg-rockchip/wiki/Filter

closed 2024-03-26 by nyanmisaka

gaop921766, 2024-03-27:
thank you every much !!!!