nyanmisaka/ffmpeg-rockchip #48: Questions about cross-compilation

Kevin111369, 2024-03-26:
# Cross-compilation Error for ffmpeg-rockchip 
## libdrm and rockchip_mpp pkg-config Not Found

Build files for mpp, rga, and drm have been successfully generated through cross-compilation. Using the following commands, pkg-config files have been imported, and pkg-config information has been successfully queried, as shown below:
```bash
rk@rk-VirtualBox:~/linux/ffmpeg-rockchip-master$ pkg-config --modversion libdrm
2.4.107
rk@rk-VirtualBox:~/linux/ffmpeg-rockchip-master$ pkg-config --modversion rockchip_mpp
1.3.8
```

Then, attempting to compile ffmpeg with the following command:
```bash
./configure \
--enable-shared \
--enable-static \
--prefix=$PWD/_install \
--enable-cross-compile \
--cross-prefix=/home/rk/linux/rk3588_sdk/prebuilts/gcc/linux-x86/aarch64/gcc-arm-10.3-2021.07-x86_64-aarch64-none-linux-gnu/bin/aarch64-none-linux-gnu- \
--arch=aarch64 \
--target-os=linux \
--extra-cflags="-I/home/rk/linux/airockchip/librga/installed/include;-I/home/rk/linux/airockchip/mpp/installed/include;-I/home/rk/linux/libdrm-2.4.107/install/include" \
--extra-ldflags="-L/home/k/linux/airockchip/librga/installed/lib;-L/home/rk/linux/airockchip/mpp/installed/lib;-L/home/rk/linux/libdrm-2.4.107/install/lib" \
--enable-gpl \
--enable-version3 \
--enable-libdrm \
--enable-rkmpp \
--enable-rkrga \
--disable-stripping
```

Encountering the following errors:
```bash
ERROR: libdrm not found using pkg-config
ERROR: rockchip_mpp not found using pkg-config
```

nyanmisaka, 2024-03-26:
It seems to be an environment variable or configure setting issue. Please refer to the [official documentation](https://trac.ffmpeg.org/wiki/CompilationGuide/RaspberryPi), as well as [BtbN/FFmpeg-Builds](https://github.com/BtbN/FFmpeg-Builds) to learn how to cross-compile FFmpeg correctly.

jian263994241, 2024-03-27:
sudo apt install pkg-config

jacobbtbailey, 2024-04-03:
@Kevin111369 You can try something like this. I tested on WSL2 (Ubuntu 22.04).
```bash
PKG_CONFIG_SYSROOT_DIR=/root/mtea-tech/rk3568l/buildroot/output/rockchip_rk3568/host/aarch64-buildroot-linux-gnu/sysroot \
    PKG_CONFIG_PATH=/root/mtea-tech/rk3568l/buildroot/output/rockchip_rk3568/host/aarch64-buildroot-linux-gnu/sysroot/usr/lib/pkgconfig \
    ./configure --prefix=prefix --enable-gpl --enable-version3 --enable-libdrm --enable-rkmpp --enable-rkrga \
    --cross-prefix=/root/mtea-tech/rk3588l/buildroot/output/rockchip_rk3588/host/bin/aarch64-linux- \
    --arch=aarch64 --target-os=linux --enable-cross-compile --target-os=linux --disable-shared --pkg-config=pkg-config
```

Kevin111369, 2024-04-03:
@jacobbtbailey I added the paths to libdrm and mpp when configuring configure, and then successfully compiled ffmpeg-rockchip. The compilation script is as follows:
```bash
#!/bin/bash

# 设置路径变量
LIBDRM_PATH="/home/rk/linux/3rdparty/output/libdrm"
MPP_PATH="/home/rk/linux/3rdparty/output/mpp"
RGA_PATH="/home/rk/linux/3rdparty/output/librga"
TOOLCHAIN_PATH="/home/rk/linux/rk3588_sdk/prebuilts/gcc/linux-x86/aarch64/gcc-arm-10.3-2021.07-x86_64-aarch64-none-linux-gnu/bin/aarch64-none-linux-gnu-"
PREFIX_PATH="/home/rk/linux/3rdparty/output/ffmpeg"

# 设置 PKG_CONFIG_PATH
export PKG_CONFIG_PATH="$LIBDRM_PATH/lib/pkgconfig:$MPP_PATH/lib/pkgconfig:$PKG_CONFIG_PATH"

# 运行 configure 脚本
./configure \
--enable-shared \
--enable-static \
--prefix=$PREFIX_PATH \
--enable-cross-compile \
--cross-prefix=$TOOLCHAIN_PATH \
--arch=aarch64 \
--target-os=linux \
--enable-gpl \
--enable-version3 \
--enable-libdrm \
--enable-rkmpp \
--extra-cflags="-I$RGA_PATH/include" \
--extra-ldflags="-L$RGA_PATH/lib" \
--enable-rkrga \
--disable-stripping \
--pkg-config="pkg-config --static"
```
Thank you for your reply！

closed 2024-04-03 by Kevin111369

zjd1988, 2024-06-27:
@Kevin111369 will you post all the build script (contain rkmpp and libdrm cross compile steps)

Kevin111369, 2024-06-27:
> @Kevin111369 will you post all the build script (contain rkmpp and libdrm cross compile steps)

Sorry, I won't post these scripts, because I didn't use ffmpeg-rockchip in video development, but used ffmpeg+mpp

zjd1988, 2024-06-27:
@Kevin111369 Thanks for your reply. Did you cross compile rkmpp and libdrm, can you provide steps?