nyanmisaka/ffmpeg-rockchip #50: 使用av_read_frame进行rtsp拉流阻塞的问题

Pi-Boss, 2024-03-27:
当rtsp流设备以外断电后再恢复供电。av_read_frame会被永远阻塞。尽管我已经设置了stimeout / timeout 。这并没有效果
@nyanmisaka

nyanmisaka, 2024-03-27:
This sounds like an issue in the RTSP demuxer. The current RKMPP decoder logic is based on the Rockchip maintainer's reference implementation. Please try switching to a software decoder to see if it can still be reproduced.

Also don't submit an issue without any debug logs (-v debug), it won't help.

Pi-Boss, 2024-04-03:
It seems I've found the issue. In FFmpeg, obtaining RTSP streams and unwrapping them doesn't rely on hardware decoders. I managed to resolve the continuous blocking by adding callback functions. Thank you! @nyanmisaka

closed 2024-04-03 by nyanmisaka

stz-source, 2025-01-17:
Hello, I am also implementing something similar on the rk3588. Unfortunately, the RTSP stream cannot reconnect after being disconnected externally. Could you please share your work for reference?