nyanmisaka/ffmpeg-rockchip #51: crashes - various errors

great9, 2024-03-29:
### camera two
```
024-03-27 02:52:40.643412721  [2024-03-27 02:52:40] ffmpeg.camera_two.detect ERROR   :   Metadata:
2024-03-27 02:52:40.643416221  [2024-03-27 02:52:40] ffmpeg.camera_two.detect ERROR   :     title           : Media Presentation
2024-03-27 02:52:40.643420013  [2024-03-27 02:52:40] ffmpeg.camera_two.detect ERROR   :   Duration: N/A, start: 0.160000, bitrate: N/A
2024-03-27 02:52:40.643424971  [2024-03-27 02:52:40] ffmpeg.camera_two.detect ERROR   :   Stream #0:0: Video: hevc (Main), yuvj420p(pc), 640x360 [SAR 1:1 DAR 16:9], 10 fps, 12.50 tbr, 90k tbn
2024-03-27 02:52:40.643428179  [2024-03-27 02:52:40] ffmpeg.camera_two.detect ERROR   : Stream mapping:
2024-03-27 02:52:40.643431971  [2024-03-27 02:52:40] ffmpeg.camera_two.detect ERROR   :   Stream #0:0 -> #0:0 (hevc (hevc_rkmpp) -> rawvideo (native))
2024-03-27 02:52:40.643435470  [2024-03-27 02:52:40] ffmpeg.camera_two.detect ERROR   : Press [q] to stop, [?] for help
2024-03-27 02:52:40.643438970  [2024-03-27 02:52:40] ffmpeg.camera_two.detect ERROR   : Output #0, rawvideo, to 'pipe:':
2024-03-27 02:52:40.643442179  [2024-03-27 02:52:40] ffmpeg.camera_two.detect ERROR   :   Metadata:
2024-03-27 02:52:40.643445678  [2024-03-27 02:52:40] ffmpeg.camera_two.detect ERROR   :     title           : Media Presentation
2024-03-27 02:52:40.643449178  [2024-03-27 02:52:40] ffmpeg.camera_two.detect ERROR   :     encoder         : Lavf60.16.100
2024-03-27 02:52:40.643454428  [2024-03-27 02:52:40] ffmpeg.camera_two.detect ERROR   :   Stream #0:0: Video: rawvideo (I420 / 0x30323449), yuv420p(pc, progressive), 640x360 [SAR 1:1 DAR 16:9], q=2-31, 27648 kb/s, 10 fps, 10 tbn
2024-03-27 02:52:40.643457636  [2024-03-27 02:52:40] ffmpeg.camera_two.detect ERROR   :     Metadata:
2024-03-27 02:52:40.643461136  [2024-03-27 02:52:40] ffmpeg.camera_two.detect ERROR   :       encoder         : Lavc60.31.102 rawvideo

2024-03-27 02:52:20.682267528  [2024-03-27 02:52:20] ffmpeg.camera_two.detect ERROR   : frame=  268 fps=9.7 q=-0.0 size=   90112kB time=00:00:29.20 bitrate=25280.9kbits/s speed=1.05x    
2024-03-27 02:52:20.682451272  [2024-03-27 02:52:20] ffmpeg.camera_two.detect ERROR   : frame=  272 fps=9.6 q=-0.0 size=   91462kB time=00:00:29.70 bitrate=25227.6kbits/s speed=1.05x    
2024-03-27 02:52:20.682637350  [2024-03-27 02:52:20] ffmpeg.camera_two.detect ERROR   : frame=  279 fps=9.7 q=-0.0 size=   93825kB time=00:00:30.30 bitrate=25366.8kbits/s speed=1.05x    
2024-03-27 02:52:20.682821970  [2024-03-27 02:52:20] ffmpeg.camera_two.detect ERROR   : frame=  285 fps=9.6 q=-0.0 size=   95850kB time=00:00:30.90 bitrate=25411.1kbits/s speed=1.05x    
2024-03-27 02:52:20.683003089  [2024-03-27 02:52:20] ffmpeg.camera_two.detect ERROR   : frame=  291 fps=9.7 q=-0.0 size=   97875kB time=00:00:31.50 bitrate=25453.7kbits/s speed=1.05x    
2024-03-27 02:52:20.683192959  [2024-03-27 02:52:20] ffmpeg.camera_two.detect ERROR   : frame=  297 fps=9.7 q=-0.0 size=   99900kB time=00:00:32.10 bitrate=25494.7kbits/s speed=1.04x    
2024-03-27 02:52:20.683377286  [2024-03-27 02:52:20] ffmpeg.camera_two.detect ERROR   : frame=  303 fps=9.7 q=-0.0 size=  101925kB time=00:00:32.70 bitrate=25534.2kbits/s speed=1.04x    
2024-03-27 02:52:20.683634529  [2024-03-27 02:52:20] ffmpeg.camera_two.detect ERROR   : frame=  308 fps=9.6 q=-0.0 size=  103612kB time=00:00:33.30 bitrate=25489.3kbits/s speed=1.04x    
2024-03-27 02:52:20.683783857  [2024-03-27 02:52:20] ffmpeg.camera_two.detect ERROR   : frame=  314 fps=9.7 q=-0.0 size=  105638kB time=00:00:33.90 bitrate=25527.5kbits/s speed=1.04x    
2024-03-27 02:52:20.683985685  [2024-03-27 02:52:20] ffmpeg.camera_two.detect ERROR   : [Parsed_scale_rkrga_0 @ 0x5595640630] RGA blit failed: -12
2024-03-27 02:52:20.684170888  [2024-03-27 02:52:20] ffmpeg.camera_two.detect ERROR   : [vf#0:0 @ 0x559562ccf0] Error while filtering: Generic error in an external library
2024-03-27 02:52:20.684359299  [2024-03-27 02:52:20] ffmpeg.camera_two.detect ERROR   : Failed to inject frame into filter network: Generic error in an external library
2024-03-27 02:52:20.684550335  [2024-03-27 02:52:20] ffmpeg.camera_two.detect ERROR   : Error while filtering: Generic error in an external library
2024-03-27 02:52:20.684741370  [2024-03-27 02:52:20] ffmpeg.camera_two.detect ERROR   : [out#0/rawvideo @ 0x5595600090] video:108000kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: 0.000000%
2024-03-27 02:52:20.684930948  [2024-03-27 02:52:20] ffmpeg.camera_two.detect ERROR   : frame=  320 fps=9.6 q=-0.0 Lsize=  108000kB time=00:00:34.40 bitrate=25719.1kbits/s speed=1.03x    
2024-03-27 02:52:20.685118776  [2024-03-27 02:52:20] ffmpeg.camera_two.detect ERROR   : Conversion failed!
```

### camera four
```
2024-03-27 02:52:21.257455741  [2024-03-27 02:52:21] ffmpeg.camera_four.detect ERROR   : Input #0, rtsp, from 'rtsp://*:*@445.445.445.1:8787/cam/realmonitor?channel=1&subtype=2':
2024-03-27 02:52:21.257773065  [2024-03-27 02:52:21] ffmpeg.camera_four.detect ERROR   :   Metadata:
2024-03-27 02:52:21.258018349  [2024-03-27 02:52:21] ffmpeg.camera_four.detect ERROR   :     title           : Media Server
2024-03-27 02:52:21.258260134  [2024-03-27 02:52:21] ffmpeg.camera_four.detect ERROR   :   Duration: N/A, start: 0.000000, bitrate: N/A
2024-03-27 02:52:21.258504251  [2024-03-27 02:52:21] ffmpeg.camera_four.detect ERROR   :   Stream #0:0: Video: hevc (Main), yuvj420p(pc, bt470bg/bt470bg/smpte170m), 1920x1080, 25 fps, 25 tbr, 90k tbn
2024-03-27 02:52:21.258745452  [2024-03-27 02:52:21] ffmpeg.camera_four.detect ERROR   :   Stream #0:1: Audio: aac (LC), 64000 Hz, mono, fltp
2024-03-27 02:52:21.258984029  [2024-03-27 02:52:21] ffmpeg.camera_four.detect ERROR   : Stream mapping:
2024-03-27 02:52:21.259315352  [2024-03-27 02:52:21] ffmpeg.camera_four.detect ERROR   :   Stream #0:0 -> #0:0 (hevc (hevc_rkmpp) -> rawvideo (native))
2024-03-27 02:52:21.259559178  [2024-03-27 02:52:21] ffmpeg.camera_four.detect ERROR   : Press [q] to stop, [?] for help
2024-03-27 02:52:21.259798921  [2024-03-27 02:52:21] ffmpeg.camera_four.detect ERROR   : Output #0, rawvideo, to 'pipe:':
2024-03-27 02:52:21.260036622  [2024-03-27 02:52:21] ffmpeg.camera_four.detect ERROR   :   Metadata:
2024-03-27 02:52:21.260273448  [2024-03-27 02:52:21] ffmpeg.camera_four.detect ERROR   :     title           : Media Server
2024-03-27 02:52:21.260532149  [2024-03-27 02:52:21] ffmpeg.camera_four.detect ERROR   :     encoder         : Lavf60.16.100
2024-03-27 02:52:21.260773933  [2024-03-27 02:52:21] ffmpeg.camera_four.detect ERROR   :   Stream #0:0: Video: rawvideo (I420 / 0x30323449), yuv420p(pc, bt470bg/bt470bg/smpte170m, progressive), 1920x1080, q=2-31, 622080 kb/s, 25 fps, 25 tbn
2024-03-27 02:52:21.261009009  [2024-03-27 02:52:21] ffmpeg.camera_four.detect ERROR   :     Metadata:
2024-03-27 02:52:21.261250794  [2024-03-27 02:52:21] ffmpeg.camera_four.detect ERROR   :       encoder         : Lavc60.31.102 rawvideo
2024-03-27 02:52:21.261490537  [2024-03-27 02:52:21] ffmpeg.camera_four.detect ERROR   : frame=    0 fps=0.0 q=0.0 size=       0kB time=N/A bitrate=N/A dup=1 drop=0 speed=N/A    
2024-03-27 02:52:21.261729113  [2024-03-27 02:52:21] ffmpeg.camera_four.detect ERROR   : frame=   24 fps=0.0 q=-0.0 size=   69862kB time=00:00:00.96 bitrate=596160.0kbits/s dup=1 drop=0 speed=1.79x    
2024-03-27 02:52:21.261968856  [2024-03-27 02:52:21] ffmpeg.camera_four.detect ERROR   : frame=   38 fps= 36 q=-0.0 size=  112388kB time=00:00:01.52 bitrate=605709.5kbits/s dup=1 drop=0 speed=1.46x    
2024-03-27 02:52:21.262208599  [2024-03-27 02:52:21] ffmpeg.camera_four.detect ERROR   : frame=   51 fps= 32 q=-0.0 size=  151875kB time=00:00:02.04 bitrate=609882.4kbits/s dup=1 drop=0 speed=1.28x    
2024-03-27 02:52:21.262447758  [2024-03-27 02:52:21] ffmpeg.camera_four.detect ERROR   : frame=   66 fps= 30 q=-0.0 size=  197438kB time=00:00:02.60 bitrate=622080.0kbits/s dup=1 drop=0 speed=1.17x    
2024-03-27 02:52:21.262700625  [2024-03-27 02:52:21] ffmpeg.camera_four.detect ERROR   : frame=   81 fps= 28 q=-0.0 size=  243000kB time=00:00:03.24 bitrate=614400.0kbits/s dup=1 drop=0 speed=1.11x    
2024-03-27 02:52:21.262944160  [2024-03-27 02:52:21] ffmpeg.camera_four.detect ERROR   : frame=   98 fps= 29 q=-0.0 size=  294638kB time=00:00:03.92 bitrate=615732.2kbits/s dup=1 drop=0 speed=1.14x    
[... cut ...]
2024-03-27 02:52:21.264160373  [2024-03-27 02:52:21] ffmpeg.camera_four.detect ERROR   : [out#0/rawvideo @ 0x559dadbf70] video:328050kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: 0.000000%
2024-03-27 02:52:21.264400699  [2024-03-27 02:52:21] ffmpeg.camera_four.detect ERROR   : frame=  108 fps= 27 q=-0.0 Lsize=  328050kB time=00:00:04.28 bitrate=627893.8kbits/s dup=1 drop=0 speed=1.07x    
2024-03-27 02:52:21.264641901  [2024-03-27 02:52:21] ffmpeg.camera_four.detect ERROR   : Conversion failed!
```

nyanmisaka, 2024-03-29:
I seem to have answered your same question before. https://github.com/nyanmisaka/ffmpeg-rockchip/issues/25#issuecomment-1935235850

You can also search on google to see what does error (-)12 mean.
https://www.google.com/search?q=linux+error+12

Even on a 16G board, the actual available IOVA space is 4GB.
https://github.com/rockchip-linux/mpp/issues/552#issuecomment-2009124502

great9, 2024-03-29:
> Even on a 16G board, the actual available IOVA space is 4GB. [rockchip-linux/mpp#552 (comment)](https://github.com/rockchip-linux/mpp/issues/552#issuecomment-2009124502)

and this is a hardware design error that is not fixable, correct?

great9, 2024-03-29:
> Actually I'm not sure if it can be fixed in the kernel. There's nothing I
> can do about it, at least in userspace ffmpeg. You can ask more in the
> rockchip mpp repo I linked to.

Thank you. Will report there

closed 2024-03-30 by nyanmisaka