nyanmisaka/ffmpeg-rockchip #56: Segmentation fault: mpp_buffer_group_init

xaionaro, 2024-04-03:
I'm running command:
```
ffmpeg -f v4l2 -video_size 1280x720 -framerate 20 -input_format mjpeg -use_wallclock_as_timestamps 1 -i /dev/video0 -thread_queue_size 4096 -f alsa -use_wallclock_as_timestamps 1 -i default:CARD=DJIPocket3 -c:v h264_v4l2m2m -c:a aac -f rtsp -pkt_size 1100 -num_capture_buffers 12 rtsp://********:********@192.168.141.6:8554/********/********
```
And it segfaults:
```
ffmpeg version ba84e56c51 Copyright (c) 2000-2023 the FFmpeg developers
  built with gcc 10 (Debian 10.2.1-6)
  configuration: --prefix=/usr --enable-gpl --enable-version3 --enable-libdrm --enable-rkmpp --enable-rkrga
  libavutil      58. 29.100 / 58. 29.100
  libavcodec     60. 31.102 / 60. 31.102
  libavformat    60. 16.100 / 60. 16.100
  libavdevice    60.  3.100 / 60.  3.100
  libavfilter     9. 12.100 /  9. 12.100
  libswscale      7.  5.100 /  7.  5.100
  libswresample   4. 12.100 /  4. 12.100
  libpostproc    57.  3.100 / 57.  3.100
[video4linux2,v4l2 @ 0xaaab1fb67380] The driver changed the time per frame from 1/20 to 1/25
Input #0, video4linux2,v4l2, from '/dev/video0':
  Duration: N/A, start: 1712188503.338919, bitrate: N/A
  Stream #0:0: Video: mjpeg (Baseline), yuvj420p(pc, bt470bg/unknown/unknown), 1280x720 [SAR 1:1 DAR 16:9], 25 fps, 25 tbr, 1000k tbn
[aist#1:0/pcm_s16le @ 0xaaab1fb85870] Guessed Channel Layout: stereo
Input #1, alsa, from 'default:CARD=DJIPocket3':
  Duration: N/A, start: 1712188503.395639, bitrate: 1536 kb/s
  Stream #1:0: Audio: pcm_s16le, 48000 Hz, 2 channels, s16, 1536 kb/s
[out#0/rtsp @ 0xaaab1fbaf810] Codec AVOption num_capture_buffers (Number of buffers in the capture context) has not been used for any stream. The most likely reason is either wrong type (e.g. a video option with no video streams) or that it is a private option of some encoder which was not actually used for any stream.
Stream mapping:
  Stream #0:0 -> #0:0 (mjpeg (native) -> h264 (h264_rkmpp))
  Stream #1:0 -> #0:1 (pcm_s16le (native) -> aac (native))
Press [q] to stop, [?] for help
[swscaler @ 0xaaab1fbd9c80] deprecated pixel format used, make sure you did set range correctly
Segmentation fault (core dumped)s
```

`gdb` says:
```
root@radxa-zero3:/dev/shm# gdb ffmpeg core
GNU gdb (Debian 10.1-1.7) 10.1.90.20210103-git
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "aarch64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ffmpeg...
(No debugging symbols found in ffmpeg)

warning: Can't open file /SYSV0056b4d7 (deleted) during file-backed mapping note processing
[New LWP 32201]
[New LWP 32204]
[New LWP 32202]
[New LWP 32203]
[New LWP 32205]
[New LWP 32206]
[New LWP 32207]
[New LWP 32208]
[New LWP 32225]
[New LWP 32222]
[New LWP 32221]
[New LWP 32223]
[New LWP 32224]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/aarch64-linux-gnu/libthread_db.so.1".

warning: Loadable section ".note.gnu.build-id" outside of ELF segments
Core was generated by `ffmpeg -f v4l2 -video_size 1280x720 -framerate 20 -input_format mjpeg -use_wall'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x0000ffff838cdc0c in MppBufferService::get_group(char const*, char const*, MppBufferMode, MppBufferType, unsigned int) ()
   from /lib/aarch64-linux-gnu/librockchip_mpp.so.1
[Current thread is 1 (Thread 0xffff7fd47010 (LWP 32201))]
(gdb) bt
#0  0x0000ffff838cdc0c in MppBufferService::get_group(char const*, char const*, MppBufferMode, MppBufferType, unsigned int) ()
   from /lib/aarch64-linux-gnu/librockchip_mpp.so.1
#1  0x0000ffff838ce040 in mpp_buffer_group_init () from /lib/aarch64-linux-gnu/librockchip_mpp.so.1
#2  0x0000aaaae336b318 in ?? ()
#3  0x0000aaaae3369db8 in ?? ()
#4  0x0000aaaae2ed2d18 in ?? ()
#5  0x0000aaaae2bcd690 in ?? ()
#6  0x0000aaaae27eb438 in ?? ()
#7  0x0000aaaae27ecc50 in ?? ()
#8  0x0000aaaae27ee294 in ?? ()
#9  0x0000aaaae27f0bfc in ?? ()
#10 0x0000aaaae27e3350 in main ()
(gdb) 
```

nyanmisaka, 2024-04-04:
https://github.com/nyanmisaka/ffmpeg-rockchip#important

> User MUST be granted permission to access these device files.
```
# DRM allocator
/dev/dri

# DMA_HEAP allocator
/dev/dma_heap

# RGA filters
/dev/rga

# MPP codecs
/dev/mpp_service

# Optional, for compatibility with older kernels and socs
/dev/iep
/dev/mpp-service
/dev/vpu_service
/dev/vpu-service
/dev/hevc_service
/dev/hevc-service
/dev/rkvdec
/dev/rkvenc
/dev/vepu
/dev/h265e
```

closed 2024-04-04 by nyanmisaka

xaionaro, 2024-04-04:
ffmpeg was running under root and it had access to the devices.

nyanmisaka, 2024-04-04:
Do not use the outdated MPP libs from Radxa.

https://github.com/nyanmisaka/ffmpeg-rockchip/issues/3#issuecomment-1875297620

nyanmisaka, 2024-04-04:
Sadly this doesn't prevent people from still trying to use outdated libraries because Rockchip never bumps version numbers in [pkgconfig/rockchip_mpp.pc](https://github.com/rockchip-linux/mpp/blob/develop/pkgconfig/rockchip_mpp.pc.cmake#L9). The version checking during the `configure` stage of FFmpeg could have fundamentally avoided this problem, but Rockchip's approach makes this impossible.

Since I am not an employee of Rockchip, please report such issues to upstream and urge them to resolve them. Thanks.

xaionaro, 2024-04-22:
> Sadly this doesn't prevent people from still trying to use outdated libraries because Rockchip never bumps version numbers in [...]

Done:
https://github.com/rockchip-linux/mpp/commit/bcd3065dc9a8f915937efc17455f77b1e260ae3e

nyanmisaka, 2024-04-22:
> > Sadly this doesn't prevent people from still trying to use outdated libraries because Rockchip never bumps version numbers in [pkgconfig/rockchip_mpp.pc](https://github.com/rockchip-linux/mpp/blob/develop/pkgconfig/rockchip_mpp.pc.cmake#L9). The version checking during the `configure` stage of FFmpeg could have fundamentally avoided this problem, but Rockchip's approach makes this impossible.
> > Since I am not an employee of Rockchip, please report such issues to upstream and urge them to resolve them. Thanks.
> 
> Done: [rockchip-linux/mpp@bcd3065](https://github.com/rockchip-linux/mpp/commit/bcd3065dc9a8f915937efc17455f77b1e260ae3e)

I will use that in a future FFmpeg version. But for now it's too new for most distros and people will complain about this;)