nyanmisaka/ffmpeg-rockchip #58: 当前支持rk3528处理器吗？

sputnik2019, 2024-04-07:
在rk3528的armbian上安装ffmpeg后可以使用mpv硬解码视频 
sudo mpv --hwdec=rkmpp 1.mp4
[ffmpeg/demuxer] mov,mp4,m4a,3gp,3g2,mj2: stream 0, timescale not set
 (+) Video --vid=1 (*) (h264 1280x720 24.000fps)
     Video --vid=2 [P] (mjpeg 1.000fps)
 (+) Audio --aid=1 --alang=eng (*) (aac 2ch 48000Hz)
     Audio --aid=2 --alang=eng (eac3 6ch 48000Hz)
File tags:
 Date: 2024
 Description: Toranaga is forced to carve out a new deal with a long lost family member.
 Title: A Stick of Time
error: XDG_RUNTIME_DIR not set in the environment.
Using hardware decoding (rkmpp).
AO: [alsa] 48000Hz stereo 2ch float
VO: [gpu] 1280x720 drm_prime[nv12]
AV: 00:00:59 / 00:54:30 (2%) A-V:  0.000 Dropped: 8

Audio/Video desynchronisation detected! Possible reasons include too slow
hardware, temporary CPU spikes, broken drivers, and broken files. Audio
position will not match to the video (see A-V status field).

但是视频颜色不对，显示偏红色，RGA2没有中断产生
内核提示：
mpp_info: mpp version: unknown mpp version for missing VCS info

nyanmisaka, 2024-04-07:
I am not sure. Because I don't have a RK3528 board. This project is only tested on RK3588 and RK356x.

hbiyik, 2024-04-09:
> mpp_info: mpp version: unknown mpp version for missing VCS info

your mpp is very old, try it with latest git version.

closed 2024-04-24 by nyanmisaka