nyanmisaka/ffmpeg-rockchip #59: SIG11 when running ffmpeg

vvsvvsvvs, 2024-04-07:
**Stone:**
Linux orangepi5plus 5.10.160-rockchip-rk3588

**Input:**
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from 'uhd_vid.mp4':
  Metadata:
    major_brand     : isom
    minor_version   : 512
    compatible_brands: isomiso2avc1mp41
    encoder         : Lavf58.45.100
  Duration: 00:02:27.17, start: 0.000000, bitrate: 45285 kb/s
  Stream #0:0[0x1](eng): Video: h264 (High) (avc1 / 0x31637661), yuv420p(tv, bt709, progressive), 3840x2160 [SAR 1:1 DAR 16:9], 45061 kb/s, 29.97 fps, 29.97 tbr, 16k tbn (default)
    Metadata:
      handler_name    : VideoHandler
      vendor_id       : [0][0][0][0]
  Stream #0:1[0x2](eng): Audio: ac3 (ac-3 / 0x332D6361), 48000 Hz, stereo, fltp, 224 kb/s (default)
    Metadata:
      handler_name    : SoundHandler
      vendor_id       : [0][0][0][0]
    Side data:
      audio service type: main

**Trtanscoding:**
./ffmpeg_g -i ./uhd_vid.mp4 -c:v h264_rkmpp  -qp_init 26 -profile:v main -level 4.1 -y tmp.mp4

**SIG11:**
`warning: Loadable section "" outside of ELF segments
  in /lib/aarch64-linux-gnu/libmali.so.1
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/aarch64-linux-gnu/libthread_db.so.1".
Core was generated by `./ffmpeg_g -i ./uhd_vid.mp4 -c:v h264_rkmpp -qp_init 26 -profile:v main -level'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x0000007f990076ac in MppBufferService::get_group(char const*, char const*, MppBufferMode, MppBufferType, unsigned int) () from /lib/aarch64-linux-gnu/librockchip_mpp.so.1
[Current thread is 1 (Thread 0x7f99cc7020 (LWP 24452))]
(gdb) where
#0  0x0000007f990076ac in MppBufferService::get_group(char const*, char const*, MppBufferMode, MppBufferType, unsigned int) () from /lib/aarch64-linux-gnu/librockchip_mpp.so.1
#1  0x0000007f99007ae0 in mpp_buffer_group_init () from /lib/aarch64-linux-gnu/librockchip_mpp.so.1
#2  0x0000005583a98020 in rkmpp_frames_init (hwfc=0x558795d720) at libavutil/hwcontext_rkmpp.c:302
#3  0x0000005583a93f68 in av_hwframe_ctx_init (ref=0x558795cfd0) at libavutil/hwcontext.c:368
#4  0x00000055835eb1c4 in init_hwframes_ctx (avctx=0x55878ef4a0) at libavcodec/rkmppenc.c:916
#5  rkmpp_encode_init (avctx=0x55878ef4a0) at libavcodec/rkmppenc.c:1080
#6  0x00000055832dc734 in avcodec_open2 (avctx=avctx@entry=0x55878ef4a0, codec=codec@entry=0x5583fc7b08 <ff_h264_rkmpp_encoder>, options=options@entry=0x558793ee90) at libavcodec/avcodec.c:337
#7  0x0000005582ede9bc in enc_open (ost=ost@entry=0x558793ed80, frame=frame@entry=0x55878ef0c0) at fftools/ffmpeg_enc.c:311
#8  0x0000005582ee02a4 in enc_frame (ost=ost@entry=0x558793ed80, frame=frame@entry=0x55878ef0c0) at fftools/ffmpeg_enc.c:855
#9  0x0000005582ee199c in fg_output_frame (ofp=ofp@entry=0x55878f2100, frame=frame@entry=0x55878ef2b0) at fftools/ffmpeg_filter.c:2125
#10 0x0000005582ee449c in fg_output_step (flush=0, ofp=0x55878f2100) at fftools/ffmpeg_filter.c:2220
#11 reap_filters (flush=<optimized out>, fg=<optimized out>) at fftools/ffmpeg_filter.c:2239
#12 reap_filters (fg=0x558790e570, flush=flush@entry=0) at fftools/ffmpeg_filter.c:2228
#13 0x0000005582ed6554 in transcode_step (ost=<optimized out>) at fftools/ffmpeg.c:1155
#14 transcode (err_rate_exceeded=<synthetic pointer>) at fftools/ffmpeg.c:1204
#15 main (argc=<optimized out>, argv=<optimized out>) at fftools/ffmpeg.c:1330
(gdb) 
`

nyanmisaka, 2024-04-07:
@vvsvvsvvs See https://github.com/nyanmisaka/ffmpeg-rockchip/issues/56

[Built the latest MPP library](https://github.com/nyanmisaka/ffmpeg-rockchip/wiki/Compilation). DO NOT USE the out dated libs from the vendor image.

closed 2024-04-07 by nyanmisaka

vvsvvsvvs, 2024-04-07:
> @vvsvvsvvs See #56
> 
> [Build the latest MPP library](https://github.com/nyanmisaka/ffmpeg-rockchip/wiki/Compilation). DO NOT USE the outdated libs from the vendor image.
Hm... Which is latest?
Is https://github.com/rockchip-linux/mpp/tree/develop  OK?

nyanmisaka, 2024-04-07:
- https://github.com/nyanmisaka/mpp.git
- https://github.com/rockchip-linux/mpp.git

vvsvvsvvs, 2024-04-07:
Thanx a lot for pointing out the idea!!!
The newest mpp lib was shaded by the old system-wide installed within the distro of Ubuntu 22.04 on Orange Pi5.
The same situation with rga lib. Please, point these out in the Wiki (or guide).