nyanmisaka/ffmpeg-rockchip #61: h264_rkmpp 接收 rawvideo 灰度图数据失败。Failed to get key input frame from packet meta: -1

fuyousk, 2024-04-10:
ffmpeg 通过管道线接收灰度图 bytes 数据，在使用 libx264 编码时可以推流，但换用 h264_rkmpp 接收数据时报错。

ffmpeg 启动命令为：
ffmpeg -y -f rawvideo -pix_fmt gray -s 1280x100 -i - -c:v h264_rkmpp -f rtsp rtsp://127.0.0.1:8554/live/test

请问我需要如何处理？

调试信息如下：
ffmpeg version ba84e56 Copyright (c) 2000-2023 the FFmpeg developers
  built with gcc 9 (Ubuntu 9.4.0-1ubuntu1~20.04.3)
  configuration: --prefix=/usr/local/ffmpeg --toolchain=hardened --arch=arm64 --enable-gpl --disable-stripping --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opencl --enable-opengl --enable-sdl2 --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-frei0r --enable-libx264 --enable-shared --enable-version3 --enable-libdrm --enable-rkmpp --enable-rkrga
  libavutil      58. 29.100 / 58. 29.100
  libavcodec     60. 31.102 / 60. 31.102
  libavformat    60. 16.100 / 60. 16.100
  libavdevice    60.  3.100 / 60.  3.100
  libavfilter     9. 12.100 /  9. 12.100
  libswscale      7.  5.100 /  7.  5.100
  libswresample   4. 12.100 /  4. 12.100
  libpostproc    57.  3.100 / 57.  3.100
Splitting the commandline.
Reading option '-y' ... matched as option 'y' (overwrite output files) with argument '1'.
Reading option '-f' ... matched as option 'f' (force format) with argument 'rawvideo'.
Reading option '-pix_fmt' ... matched as option 'pix_fmt' (set pixel format) with argument 'gray'.
Reading option '-s' ... matched as option 's' (set frame size (WxH or abbreviation)) with argument '1280x100'.
Reading option '-i' ... matched as output url with argument '-'.
Reading option '-c:v' ... matched as option 'c' (codec name) with argument 'h264_rkmpp'.
Reading option '-loglevel' ... matched as option 'loglevel' (set logging level) with argument 'debug'.
Reading option '-f' ... matched as option 'f' (force format) with argument 'rtsp'.
Reading option 'rtsp://192.168.2.100:8554/live/24033403' ... matched as output url.
Finished splitting the commandline.
Parsing a group of options: global .
Applying option y (overwrite output files) with argument 1.
Applying option loglevel (set logging level) with argument debug.
Successfully parsed a group of options.
Parsing a group of options: input url -.
Applying option f (force format) with argument rawvideo.
Applying option pix_fmt (set pixel format) with argument gray.
Applying option s (set frame size (WxH or abbreviation)) with argument 1280x100.
Successfully parsed a group of options.
Opening an input file: -.
[rawvideo @ 0x558a20c6d0] Opening 'fd:' for reading
[fd @ 0x558a20cc30] Setting default whitelist 'crypto,data'
[rawvideo @ 0x558a20c6d0] Before avformat_find_stream_info() pos: 0 bytes read:65536 seeks:0 nb_streams:1
[rawvideo @ 0x558a20c6d0] All info found
[rawvideo @ 0x558a20c6d0] After avformat_find_stream_info() pos: 128000 bytes read:128000 seeks:0 frames:1
Input #0, rawvideo, from 'fd:':
  Duration: N/A, start: 0.000000, bitrate: 25600 kb/s
  Stream #0:0, 1, 1/25: Video: rawvideo, 1 reference frame (Y800 / 0x30303859), gray, 1280x100, 0/1, 25600 kb/s, 25 tbr, 25 tbn
Successfully opened the file.
Parsing a group of options: output url rtsp://192.168.2.100:8554/live/24033403.
Applying option c:v (codec name) with argument h264_rkmpp.
Applying option f (force format) with argument rtsp.
Successfully parsed a group of options.
Opening an output file: rtsp://192.168.2.100:8554/live/24033403.
[out#0/rtsp @ 0x558a23d200] No explicit maps, mapping streams automatically...
[vost#0:0/h264_rkmpp @ 0x558a240570] Created video stream from input stream 0:0
Successfully opened the file.
Stream mapping:
  Stream #0:0 -> #0:0 (rawvideo (native) -> h264 (h264_rkmpp))
[rawvideo @ 0x558a23d410] PACKET SIZE: 128000, STRIDE: 1280
detected 4 logical cores
[graph 0 input from stream 0:0 @ 0x558a2431e0] Setting 'video_size' to value '1280x100'
[graph 0 input from stream 0:0 @ 0x558a2431e0] Setting 'pix_fmt' to value '8'
[graph 0 input from stream 0:0 @ 0x558a2431e0] Setting 'time_base' to value '1/25'
[graph 0 input from stream 0:0 @ 0x558a2431e0] Setting 'pixel_aspect' to value '0/1'
[graph 0 input from stream 0:0 @ 0x558a2431e0] Setting 'frame_rate' to value '25/1'
[graph 0 input from stream 0:0 @ 0x558a2431e0] w:1280 h:100 pixfmt:gray tb:1/25 fr:25/1 sar:0/1
[format @ 0x558a243730] Setting 'pix_fmts' to value 'gray|yuv420p|yuv422p|yuv444p|nv12|nv21|nv16|nv24|yuyv422|yvyu422|uyvy422|rgb24|bgr24|rgba|rgb0|bgra|bgr0|argb|0rgb|abgr|0bgr|drm_prime'
[AVFilterGraph @ 0x558a23d0c0] query_formats: 4 queried, 3 merged, 0 already done, 0 delayed
[h264_rkmpp @ 0x558a2408b0] Rate Control mode is set to CBR
[... 44 lines cut ...]
[h264_rkmpp @ 0x558a2408b0] Wrote 204800 bytes to encoder
[h264_rkmpp @ 0x558a2408b0] Received a packet
[h264_rkmpp @ 0x558a2408b0] Failed to get key input frame from packet meta: -1
[vost#0:0/h264_rkmpp @ 0x558a240570] Error submitting video frame to the encoder
Error while filtering: Generic error in an external library
[vist#0:0/rawvideo @ 0x558a20cae0] Decoder thread received EOF packet
[vist#0:0/rawvideo @ 0x558a20cae0] Decoder returned EOF, finishing
[vist#0:0/rawvideo @ 0x558a20cae0] Terminating decoder thread
[h264_rkmpp @ 0x558a2408b0] End of stream
[h264_rkmpp @ 0x558a2408b0] Wrote 0 bytes to encoder
[h264_rkmpp @ 0x558a2408b0] Received an EOS packet
[out#0/rtsp @ 0x558a23d200] All streams finished
[out#0/rtsp @ 0x558a23d200] Terminating muxer thread
[AVIOContext @ 0x558a252d90] Statistics: 76 bytes written, 0 seeks, 2 writeouts
[out#0/rtsp @ 0x558a23d200] Output file #0 (rtsp://192.168.2.100:8554/live/24033403):
[out#0/rtsp @ 0x558a23d200]   Output stream #0:0 (video): 2 frames encoded; 1 packets muxed (39 bytes); 
[out#0/rtsp @ 0x558a23d200]   Total: 1 packets (39 bytes) muxed
[out#0/rtsp @ 0x558a23d200] video:0kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: unknown
frame=    1 fps=0.0 q=-0.0 Lsize=N/A time=00:00:00.00 bitrate=N/A speed=   0x    
[in#0/rawvideo @ 0x558a20c5d0] Terminating demuxer thread
[in#0/rawvideo @ 0x558a20c5d0] Input file #0 (fd:):
[in#0/rawvideo @ 0x558a20c5d0]   Input stream #0:0 (video): 4 packets read (512000 bytes); 2 frames decoded; 0 decode errors; 
[in#0/rawvideo @ 0x558a20c5d0]   Total: 4 packets (512000 bytes) demuxed
[AVIOContext @ 0x558a21cdb0] Statistics: 512000 bytes read, 0 seeks
Conversion failed!

nyanmisaka, 2024-04-10:
@fuyousk Cannot reproduce on RK3588.

1. Build and install the latest MPP library https://github.com/rockchip-linux/mpp
2. Upgrade to the latest linux 5.10 or 6.1 kernel
3. Make sure your SoC model support encoding `gray/monochroma`

```
./ffmpeg -f lavfi -i testsrc=s=1280x100,format=gray -c:v h264_rkmpp -vframes 2000 -v verbose -f null -
ffmpeg version ba84e56c51 Copyright (c) 2000-2023 the FFmpeg developers
  built with gcc 12 (Debian 12.2.0-14)
  configuration: --disable-ffplay --disable-sdl2 --disable-doc --disable-ptx-compression --disable-shared --enable-gpl --enable-version3 --enable-static --enable-gnutls --enable-opencl --enable-libdrm --enable-libdav1d --enable-libx264 --enable-libx265 --enable-rkmpp --enable-rkrga
  libavutil      58. 29.100 / 58. 29.100
  libavcodec     60. 31.102 / 60. 31.102
  libavformat    60. 16.100 / 60. 16.100
  libavdevice    60.  3.100 / 60.  3.100
  libavfilter     9. 12.100 /  9. 12.100
  libswscale      7.  5.100 /  7.  5.100
  libswresample   4. 12.100 /  4. 12.100
  libpostproc    57.  3.100 / 57.  3.100
[Parsed_testsrc_0 @ 0xaaaac71e53d0] size:1280x100 rate:25/1 duration:-1.000000 sar:1/1
[auto_scale_0 @ 0xaaaac71e6c40] w:iw h:ih flags:'' interl:0
[Parsed_format_1 @ 0xaaaac71e56c0] auto-inserting filter 'auto_scale_0' between the filter 'Parsed_testsrc_0' and the filter 'Parsed_format_1'
[auto_scale_0 @ 0xaaaac71e6c40] w:1280 h:100 fmt:rgb24 sar:1/1 -> w:1280 h:100 fmt:gray sar:1/1 flags:0x00000004
Input #0, lavfi, from 'testsrc=s=1280x100,format=gray':
  Duration: N/A, start: 0.000000, bitrate: N/A
  Stream #0:0: Video: wrapped_avframe, 1 reference frame, gray, 1280x100 [SAR 1:1 DAR 64:5], 25 fps, 25 tbr, 25 tbn
[out#0/null @ 0xaaaac7443050] No explicit maps, mapping streams automatically...
[vost#0:0/h264_rkmpp @ 0xaaaac74441d0] Created video stream from input stream 0:0
Stream mapping:
  Stream #0:0 -> #0:0 (wrapped_avframe (native) -> h264 (h264_rkmpp))
Press [q] to stop, [?] for help
[graph 0 input from stream 0:0 @ 0xaaaac74475c0] w:1280 h:100 pixfmt:gray tb:1/25 fr:25/1 sar:1/1
[h264_rkmpp @ 0xaaaac7444550] Rate Control mode is set to CBR
[h264_rkmpp @ 0xaaaac7444550] Bitrate Target/Min/Max is set to 2000000/1875000/2125000
[h264_rkmpp @ 0xaaaac7444550] QP Init/Max/Min/Max_I/Min_I is set to 26/48/0/48/0
[h264_rkmpp @ 0xaaaac7444550] Profile is set to HIGH
[h264_rkmpp @ 0xaaaac7444550] 8x8 Transform is enabled
[h264_rkmpp @ 0xaaaac7444550] Level is set to 0
[h264_rkmpp @ 0xaaaac7444550] Coder is set to CABAC
[h264_rkmpp @ 0xaaaac7444550] Created a RKMPP hardware device
Output #0, null, to 'pipe:':
  Metadata:
    encoder         : Lavf60.16.100
  Stream #0:0: Video: h264 (High), 1 reference frame, gray(pc, progressive), 1280x100 (0x0) [SAR 1:1 DAR 64:5], q=2-31, 2000 kb/s, 25 fps, 25 tbn
    Metadata:
      encoder         : Lavc60.31.102 h264_rkmpp
[h264_rkmpp @ 0xaaaac7444550] Configured with size: 1280x100 | pix_fmt: gray | sw_pix_fmt: gray
No more output streams to write to, finishing.1:00.44 bitrate=N/A speed=40.2x
[out#0/null @ 0xaaaac7443050] All streams finished
[out#0/null @ 0xaaaac7443050] Terminating muxer thread
[vist#0:0/wrapped_avframe @ 0xaaaac7443260] Decoder thread received EOF packet
[vist#0:0/wrapped_avframe @ 0xaaaac7443260] Decoder returned EOF, finishing
[vist#0:0/wrapped_avframe @ 0xaaaac7443260] Terminating decoder thread
[out#0/null @ 0xaaaac7443050] Output file #0 (pipe:):
[out#0/null @ 0xaaaac7443050]   Output stream #0:0 (video): 2000 frames encoded; 2000 packets muxed (1594939 bytes);
[out#0/null @ 0xaaaac7443050]   Total: 2000 packets (1594939 bytes) muxed
[out#0/null @ 0xaaaac7443050] video:1558kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: unknown
frame= 2000 fps=1021 q=-0.0 Lsize=N/A time=00:01:19.96 bitrate=N/A speed=40.8x
```

fuyousk, 2024-04-10:
好的，多谢。我更新下环境再试试

closed 2024-04-17 by nyanmisaka