nyanmisaka/ffmpeg-rockchip #62: RK3588测速问题

erquren, 2024-04-10:
测试视频为1080P

./ffmpeg -hwaccel rkmpp -i output.mp4 -f null -
FPS为440

./ffmpeg -hwaccel rkmpp -i outputh265.mp4 -f null -
FPS为440

ffmpeg -i output.mp4 -c:v hevc_nvenc -b:v 2000k outputh265.mp4
h265是特定H265么，没有达到比h264翻倍的性能，该h265是我在PC上转的

nyanmisaka, 2024-04-10:
> 没有达到比h264翻倍的性能
> It does not achieve double the performance of h264

@erquren This is impossible. Who told you that H265 decoder has twice the performance of H264 decoder?

You need to be clear about what you are testing.

Your command not only calls hardware decoding, but also uses the time it takes for the CPU to copy the decoded images from the VPU memory DMA-BUF to RAM.

The Wiki has made it clear how to test the decoder speed.
https://github.com/nyanmisaka/ffmpeg-rockchip/wiki/Decoder

erquren, 2024-04-11:
i try the test command,it`s run 1000FPS

closed 2024-04-11 by erquren