nyanmisaka/ffmpeg-rockchip #63: (Cannot open video device /dev/video0: No such device)

N0tiK44, 2024-04-15:
Had a try and came out with no luck unfortunately for HDMI input.

Installed all dependencies (MPP & RGA)tried multiple cmd's and got (Cannot open video device /dev/video0: No such device) which is better than (not a capture device) in the past I did v4l2-ctl --list-devices and my rk_hdmirx says hdmirx-controller as /dev/video20 if thats relevant, no matter what whenever /dev/video0 is ever called it will always mention /dev/video0: No such device

went to https://trac.ffmpeg.org/wiki/CompilationGuide/Ubuntu installed all libraries and dependencies also tried the example at the bottom of the page as you mentioned which showed the same result (no such device).When i try Orange Pi's test for hdmi input using gstreamer it works perfectly with test_hdmiin.sh, would be interesting to see how that works to begin with.

Is there a specific installation methed im not doing correctly? i must be overlooking obviously. seems to be a bit overkill for HDMI input to not have such terrible latency

this person here has the same issue
https://www.reddit.com/r/OrangePI/comments/1bn38u6/how_to_use_the_hdmi_in_on_orange_pi_5_plus_with/?sort=new

david3a, 2024-04-29:
have you tried this ?

./ffplay -f video4linux2 -i /dev/video0

after a build ?

worked for me, I missed off the ./ so got pre-installed ffplay which gives the warning

N0tiK44, 2024-04-30:
> have you tried this ?
> 
> ./ffplay -f video4linux2 -i /dev/video0
> 
> after a build ?
> 
> worked for me, I missed off the ./ so got pre-installed ffplay which gives the warning

So after a considerable amount of work, for some weird reason.. /dev/video0 isnt actually a usable device. infact its /dev/video20 which is linked to the hdmirx-controller

Using the following i was able to get a encoding view HDMI input

ffmpeg -re -f v4l2 /dev/video20 -c:v hevc_rkmpp -qp_init 10 /root/Desktop/test.mp4

as of now im still unsure how to just type in a command and let it encode & decode at the same time with 0 latency..

nyanmisaka, 2024-04-30:
The video input device outputs data as raw RGB or YUV. So no h264/h265 decoding is required.

nyanmisaka, 2024-04-30:
> at the same time with 0 latency..

MPP does not support 0 latency. It can only reduce latency by using an option `-flags +low_delay`.

https://github.com/nyanmisaka/ffmpeg-rockchip/wiki/Encoder#test-encoders

N0tiK44, 2024-04-30:
> > at the same time with 0 latency..
> 
> MPP does not support 0 latency. It can only reduce latency by using an option `-flags +low_delay`.
> 
> https://github.com/nyanmisaka/ffmpeg-rockchip/wiki/Encoder#test-encoders

What can support 0 latency or 1-2 frame latency? parsec, moonlight & sunshine is software 2 software which is pretty much instant.
but hardware to hardware is getting 1 second. 
is mpp and rga not the right tools for the specific task? didnt think it would be this difficult.

nyanmisaka, 2024-04-30:
> > > at the same time with 0 latency..
> > 
> > 
> > MPP does not support 0 latency. It can only reduce latency by using an option `-flags +low_delay`.
> > https://github.com/nyanmisaka/ffmpeg-rockchip/wiki/Encoder#test-encoders
> 
> What can support 0 latency or 1-2 frame latency? 16.7ms 60fps or 8.3ms 120fps. parsec, moonlight & sunshine is software 2 software which is pretty much instant. but hardware 2 hardware is getting 1 second. is mpp and rga not the right tools for the specific task? didnt think it would be this difficult.

I'd suggest opening a new issue on the upstream Rockchip MPP repo to ask the official developers about this type of question. They know hardware better than me. https://github.com/rockchip-linux/mpp/issues

hbiyik, 2024-04-30:
is the application here is streaming V4L2 input? Or encoding it?
If streaming, ffmpeg/mpp/rga is totally irrelevant, and ffplay is notorious in terms of latency.

To stream you can try mpv or something which is meant for that, which i believe ffplay is not meant to.

https://github.com/mpv-player/mpv/wiki/Video4Linux2-Input

Above gives some insights. And i had never tried this in past so i also dont know.

N0tiK44, 2024-04-30:
> is the application here is streaming V4L2 input? Or encoding it? If streaming, ffmpeg/mpp/rga is totally irrelevant, and ffplay is notorious in terms of latency.
>
> To stream you can try mpv or [...]

Its encoding video to the /root/Desktop/test.mp4 destination, which cannot be viewed at the same time from my testing
id use mpv but that is a GPL license.

nyanmisaka, 2024-04-30:
IMO if you just want to play the video captured by HDMI IN/RX on the screen, then there is no need to use MPP and RGA to transcode.

HDMI IN (YUV/RGB) -> HDMI/DP OUT (Display)

N0tiK44, 2024-04-30:
> IMO if you just want to play the video captured by HDMI IN/RX on the screen, then there is no need to use MPP and RGA to transcode.
> 
> HDMI IN (YUV/RGB) -> HDMI/DP OUT (Display)

Not as easy as you've put it. only way to read HDMI in video is by /dev/video20 via test_hdmiin.sh

nyanmisaka, 2024-04-30:
> > IMO if you just want to play the video captured by HDMI IN/RX on the screen, then there is no need to use MPP and RGA to transcode.
> > HDMI IN (YUV/RGB) -> HDMI/DP OUT (Display)
> 
> Not as easy as you've put it. only way to read HDMI in video is by /dev/video20 via test_hdmiin.sh or v4l2 test bench

Then this has nothing to do with this project. It's just that the video player you are using does not support reading videos from V4L2 input. MPP hardware decoders and encoders are not used in `test_hdmiin.sh`.

https://github.com/BredOS/sbc-pkgbuilds/blob/869a425c939aeef5903d8610cf7d175644a4159d/opi5plus-post-install/test_hdmiin.sh

closed 2024-04-30 by nyanmisaka