nyanmisaka/ffmpeg-rockchip #64: The encoder of rkmpp is  not working

mustlezoo, 2024-04-26:
I compiled this project and installed it .I can see the encoders of rkmpp with " ffmpeg -encoders | grep rkmpp". and show information below

[image]

when i try to convert video and I got the problem like this

[image]

Is the rkmpp encoder in ffmpeg not available for use in this project?

nyanmisaka, 2024-04-26:
> Important

> Make sure you have built and installed the latest MPP runtime and headers. Otherwise undefined behavior may occur, such as **segmentation fault**. Be careful if you use MPP deb packages from a vendor/BSP image, these packages are likely to be out of date.

https://github.com/nyanmisaka/ffmpeg-rockchip/wiki/Compilation#important

mustlezoo, 2024-04-26:
> n

thanks,the problem has been solved.The kernel which I download is too old.I changed a newer kernel and followed all step in wiki which you told me. The rkmpp convert avi to mp4 just 20fps in RK3566.
I just used this command.

ffmpeg -i "1.avi" -c:v h264_rkmpp -c:a aac -f mp4 output.mp4

It's just best performance of RK3566 or I make mistake in some steps?Anyway, thank you very much for your help

closed 2024-04-26 by nyanmisaka