nyanmisaka/ffmpeg-rockchip #66: 使用h264_rkmpp作为解码器时候，从avcodec_receive_frame返回的frme->pkt_size一直为-1

MUCHWAY, 2024-05-15:
1. 我正确安装了ffmpeg和rkmpp，并且使用wiki中的测试指令，可以正常找到h264_rkmpp解码器和编码器。
2. 使用命令行进行测试：
```bash
./ffmpeg -stream_loop -1 -hwaccel rkmpp -hwaccel_output_format drm_prime -i ./h264-video.mp4 -an -sn -vframes 5000 -f null -
./ffmpeg -stream_loop -1 -hwaccel rkmpp -i ./h264-video.mp4 -an -sn -vframes 5000 -f null -
./ffmpeg -stream_loop -1 -hwaccel rkmpp -hwaccel_output_format drm_prime -afbc 1 -i ./h264-video.mp4 -an -sn -vframes 5000 -f null -

./ffmpeg -i rtsp://192.168.101.37:554/live -c:v h264_rkmpp -c:a copy rtsp_output.h264
```
上面几个测试，h264_rkmpp工作都是正常的。

3. 我在程序中像下面这样调用h264_rkmpp作为解码器：
```c
    AVCodec *codec_ = avcodec_find_decoder_by_name("h264_rkmpp"); //只要我把h264_rkmpp换成h264，我的代码就是工作正常的。
    if (codec_ == NULL) {
        printf("Could not find video decoder!\n");
        return -1;
    }

    g_iDecCtx = avcodec_alloc_context3(codec_);
    if (g_iDecCtx == NULL) {
        printf("Could not allocate video codec context!\n");
        return -1;
    }
    
    avcodec_parameters_to_context(g_iDecCtx, g_iStream->codecpar);

    int ret = avcodec_open2(g_iDecCtx, codec_, NULL);
    if (ret < 0) {
        printf("Could not open codec: %s!\n", av_err2str(ret));
        return -1;
    }
```
我从解码器接收解码完成的frame：
```c
ret = avcodec_receive_frame(g_iDecCtx, g_Frame); //这里的ret返回值一直是0的
printf(" Packet Size: %d\n", g_Frame->pkt_size); //这里打印的g_Frame->pkt_size 一直是-1
```

4. 我将h264_rkmpp作为编码器：
```c
    AVCodec *codec = avcodec_find_encoder_by_name("h264_rkmpp");
    if (!codec) {
        printf("Codec libx264 not found\n");
        return -1;
    }
```
它作为编码器就是工作正常的。

我现在很疑惑，经过上面的验证，在命令行中使用h264_rkmpp作为编码器和解码器是ok的，在程序中作为编码器也是ok的，就是作为解码器时候就不行。

nyanmisaka, 2024-05-15:
According to the [FFmpeg documentation](https://ffmpeg.org/doxygen/trunk/structAVFrame.html#a8e4e67b353ab25ad13f00a9fda1d4904), the `AVFrame->pkt_size` is set to a negative value if unknown.

Hardware decoders are different from software decoders. You cannot trace packets based on decoded frames. This feature is not available here.

MUCHWAY, 2024-05-15:
Thank you very much for your answer, can you share the usage when using h264_rkmpp as the decoder.

nyanmisaka, 2024-05-16:
> Thank you very much for your answer, can you share the usage when using h264_rkmpp as the decoder.

You can refer to the example [qsv_decode.c](https://github.com/nyanmisaka/ffmpeg-rockchip/blob/master/doc/examples/qsv_decode.c). The usage of rkmpp and qsv decoders is very similar.

MUCHWAY, 2024-05-16:
hello，I refer to the example [qsv_decode.c](https://github.com/nyanmisaka/ffmpeg-rockchip/blob/master/doc/examples/qsv_decode.c) , but the size of frame returned by avcodec_receive_frame is still -1.
```c
/*
 * Copyright (c) 2015 Anton Khirnov
 *
 * Permission is hereby granted, free of charge, to any person obtaining a copy
 * of this software and associated documentation files (the "Software"), to deal
 * in the Software without restriction, including without limitation the rights
 * to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
 * copies of the Software, and to permit persons to whom the Software is
 * furnished to do so, subject to the following conditions:
 *
 * The above copyright notice and this permission notice shall be included in
 * all copies or substantial portions of the Software.
 *
 * THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
 * IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
 * FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL
 * THE AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
 * LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
 * OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN
 * THE SOFTWARE.
 */

/**
 * @file Intel QSV-accelerated H.264 decoding API usage example
 * @example qsv_decode.c
 *
 * Perform QSV-accelerated H.264 decoding with output frames in the
 * GPU video surfaces, write the decoded frames to an output file.
 */

#include <stdio.h>

#include "libavformat/avformat.h"
#include "libavformat/avio.h"

#include "libavcodec/avcodec.h"

#include "libavutil/buffer.h"
#include "libavutil/error.h"
#include "libavutil/hwcontext.h"
// #include "libavutil/hwcontext_qsv.h"
#include "libavutil/mem.h"

static int get_format(AVCodecContext *avctx, const enum AVPixelFormat *pix_fmts)
{
    while (*pix_fmts != AV_PIX_FMT_NONE) {
        if (*pix_fmts == AV_PIX_FMT_DRM_PRIME) {
            return AV_PIX_FMT_DRM_PRIME;
        }

        pix_fmts++;
    }

    fprintf(stderr, "The QSV pixel format not offered in get_format()\n");

    return AV_PIX_FMT_NONE;
}

static int decode_packet(AVCodecContext *decoder_ctx,
                         AVFrame *frame, AVFrame *sw_frame,
                         AVPacket *pkt, AVIOContext *output_ctx)
{
    int ret = 0;

    ret = avcodec_send_packet(decoder_ctx, pkt);
    if (ret < 0) {
        fprintf(stderr, "Error during decoding\n");
        return ret;
[... 150 lines cut ...]
    /* flush the decoder */
    ret = decode_packet(decoder_ctx, frame, sw_frame, NULL, output_ctx);

finish:
    if (ret < 0) {
        char buf[1024];
        av_strerror(ret, buf, sizeof(buf));
        fprintf(stderr, "%s\n", buf);
    }

    avformat_close_input(&input_ctx);

    av_frame_free(&frame);
    av_frame_free(&sw_frame);
    av_packet_free(&pkt);

    avcodec_free_context(&decoder_ctx);

    av_buffer_unref(&device_ref);

    avio_close(output_ctx);

    return ret;
}
````

nyanmisaka, 2024-05-16:
What do you mean size of frame?

The objects produced by the decoder is frame, not packet (pkt). So `pkt_size` is always the default value -1.

What you are looking for should be buffer size. (AVFrame->buf)

closed 2024-05-16 by nyanmisaka

wzw88486969, 2024-10-15:
`
int filter_stream(CHPlayer *p, AVPacket *pkt) {
    int ret = 0;
    if (ret = av_bsf_send_packet(p->bsf_ctx, pkt) < 0) {
        fprintf(stderr, "av_bsf_send_packet failed, ret=%d\n", ret);
        return ret;
    } 
    while ((ret = av_bsf_receive_packet(p->bsf_ctx, pkt) == 0)) {
        // ret = write_output(of, pkt); 
        // av_packet_unref(pkt);
        // if (ret < 0) return ret;
    }
    
    //if (ret == AVERROR(EAGAIN)) return 0;
    
    return ret;
}

`

wzw88486969, 2024-10-15:
AVCodec *video_codec = avcodec_find_decoder_by_name("h264_rkmpp"); 
AVBSFContext *bsf_ctx;
int is_annexb;
//-------------------------------------------------------------
for (int i = 0; i < ifmt_ctx->nb_streams; i++) {
	if (ifmt_ctx->streams[i]->codecpar->codec_type == AVMEDIA_TYPE_VIDEO) {
		is_annexb = strcmp(av_fourcc2str(ifmt_ctx->streams[i]->codecpar->codec_tag), "avc1") == 0 ? 0 : 1;
	}
}

fprintf(stdout, "is_annexb=%d\n", is_annexb);  
if (!is_annexb) {
    if (ret = open_bitstream_filter(ifmt_ctx->streams[video_stream_index], &bsf_ctx, "h264_mp4toannexb") < 0) {
       fprintf(stderr, "open_bitstream_filter failed, ret=%d\n", ret);  
       goto end;
    }
}
//-------------------------------------------------------------------
while (av_read_frame(ifmt_ctx, pkt) >= 0) {
    if (pkt->stream_index == video_stream_index) {
        if (is_annexb) {
			
        } else {
            ret = filter_stream(bsf_ctx, pkt);
        }
        
		AVCodecParameters *codecpar = p->frtCtx->streams[i]->codecpar;
		if((pkt->data[4] & 0x1f) == 5) && codecpar->extradata_size > 0){
			//store sps pps I frame 
			AVPacket tmppacket;
			char buffer[1024 * 1024];
			memcpy(buffer, codecpar->extradata, codecpar->extradata_size);
			memcpy(buffer + codecpar->extradata_size, pkt->data, pkt->size);

			tmppacket.data = (uint8_t *)buffer;
			tmppacket.size = codecpar->extradata_size + pkt->size;
			
			decode_simple(tmppacket);
		}else {
			decode_simple(pkt);
		}

    } else {
        fprintf(stdout, "read a packet, not a video frame\n");
    }
    av_packet_unref(pkt);
}

//---------------------------------------------------------

int open_bitstream_filter(AVStream *stream, const char *name) {
    int ret = 0;
    const AVBitStreamFilter *filter = av_bsf_get_by_name(name);
    if (!filter) {
        ret = -1;
        fprintf(stderr, "Unknow bitstream filter.\n");
    }
    if((ret = av_bsf_alloc(filter, &bsf_ctx) < 0)) {
        fprintf(stderr, "av_bsf_alloc failed\n");
        return ret;
    }
    if ((ret = avcodec_parameters_copy(bsf_ctx->par_in, stream->codecpar)) < 0) {
        fprintf(stderr, "avcodec_parameters_copy failed, ret=%d\n", ret);
        return ret;
    }
    if ((ret = av_bsf_init(bsf_ctx)) < 0) {
        fprintf(stderr, "av_bsf_init failed, ret=%d\n", ret);       
        return ret;
    }
    return ret;
}
int close_bitstream_filter(){
    if(is_annexb){
        if (bsf_ctx)
            av_bsf_free(&bsf_ctx);
    }
}

int filter_stream( AVPacket *pkt) {
    int ret = 0;
    if (ret = av_bsf_send_packet(bsf_ctx, pkt) < 0) {
        fprintf(stderr, "av_bsf_send_packet failed, ret=%d\n", ret);
        return ret;
    } 
    while ((ret = av_bsf_receive_packet(bsf_ctx, pkt) == 0)) {
        // ret = write_output(of, pkt); 
        // av_packet_unref(pkt);
        // if (ret < 0) return ret;
    }
    
    //if (ret == AVERROR(EAGAIN)) return 0;
    
    return ret;
}

wzw88486969, 2024-10-15:
`
`
AVCodec *video_codec = avcodec_find_decoder_by_name("h264_rkmpp"); 
AVBSFContext *bsf_ctx;
int is_annexb;
//-------------------------------------------------------------
for (int i = 0; i < ifmt_ctx->nb_streams; i++) {
	if (ifmt_ctx->streams[i]->codecpar->codec_type == AVMEDIA_TYPE_VIDEO) {
		is_annexb = strcmp(av_fourcc2str(ifmt_ctx->streams[i]->codecpar->codec_tag), "avc1") == 0 ? 0 : 1;
	}
}

fprintf(stdout, "is_annexb=%d\n", is_annexb);  
if (!is_annexb) {
    if (ret = open_bitstream_filter(ifmt_ctx->streams[video_stream_index], &bsf_ctx, "h264_mp4toannexb") < 0) {
       fprintf(stderr, "open_bitstream_filter failed, ret=%d\n", ret);  
       goto end;
    }
}
//-------------------------------------------------------------------
while (av_read_frame(ifmt_ctx, pkt) >= 0) {
    if (pkt->stream_index == video_stream_index) {
        if (is_annexb) {
			
        } else {
            ret = filter_stream(bsf_ctx, pkt);
        }
        
		AVCodecParameters *codecpar = p->frtCtx->streams[i]->codecpar;
		if((pkt->data[4] & 0x1f) == 5) && codecpar->extradata_size > 0){
			//store sps pps I frame 
			AVPacket tmppacket;
			char buffer[1024 * 1024];
			memcpy(buffer, codecpar->extradata, codecpar->extradata_size);
			memcpy(buffer + codecpar->extradata_size, pkt->data, pkt->size);

			tmppacket.data = (uint8_t *)buffer;
			tmppacket.size = codecpar->extradata_size + pkt->size;
			
			decode_simple(tmppacket);
		}else {
			decode_simple(pkt);
		}

    } else {
        fprintf(stdout, "read a packet, not a video frame\n");
    }
    av_packet_unref(pkt);
}

//---------------------------------------------------------

int open_bitstream_filter(AVStream *stream, const char *name) {
    int ret = 0;
    const AVBitStreamFilter *filter = av_bsf_get_by_name(name);
    if (!filter) {
        ret = -1;
        fprintf(stderr, "Unknow bitstream filter.\n");
    }
    if((ret = av_bsf_alloc(filter, &bsf_ctx) < 0)) {
        fprintf(stderr, "av_bsf_alloc failed\n");
        return ret;
    }
    if ((ret = avcodec_parameters_copy(bsf_ctx->par_in, stream->codecpar)) < 0) {
        fprintf(stderr, "avcodec_parameters_copy failed, ret=%d\n", ret);
        return ret;
    }
    if ((ret = av_bsf_init(bsf_ctx)) < 0) {
        fprintf(stderr, "av_bsf_init failed, ret=%d\n", ret);       
        return ret;
    }
    return ret;
}
int close_bitstream_filter(){
    if(is_annexb){
        if (bsf_ctx)
            av_bsf_free(&bsf_ctx);
    }
}

int filter_stream( AVPacket *pkt) {
    int ret = 0;
    if (ret = av_bsf_send_packet(bsf_ctx, pkt) < 0) {
        fprintf(stderr, "av_bsf_send_packet failed, ret=%d\n", ret);
        return ret;
    } 
    while ((ret = av_bsf_receive_packet(bsf_ctx, pkt) == 0)) {
        // ret = write_output(of, pkt); 
        // av_packet_unref(pkt);
        // if (ret < 0) return ret;
    }
    
    //if (ret == AVERROR(EAGAIN)) return 0;
    
    return ret;
}

`
`

wzw88486969, 2024-10-15:
```c
AVCodec *video_codec = avcodec_find_decoder_by_name("h264_rkmpp"); 
AVBSFContext *bsf_ctx;
int is_annexb;
//-------------------------------------------------------------
for (int i = 0; i < ifmt_ctx->nb_streams; i++) {
	if (ifmt_ctx->streams[i]->codecpar->codec_type == AVMEDIA_TYPE_VIDEO) {
		is_annexb = strcmp(av_fourcc2str(ifmt_ctx->streams[i]->codecpar->codec_tag), "avc1") == 0 ? 0 : 1;
	}
}

fprintf(stdout, "is_annexb=%d\n", is_annexb);  
if (!is_annexb) {
    if (ret = open_bitstream_filter(ifmt_ctx->streams[video_stream_index], &bsf_ctx, "h264_mp4toannexb") < 0) {
       fprintf(stderr, "open_bitstream_filter failed, ret=%d\n", ret);  
       goto end;
    }
}
//-------------------------------------------------------------------
while (av_read_frame(ifmt_ctx, pkt) >= 0) {
    if (pkt->stream_index == video_stream_index) {
        if (is_annexb) {
			
        } else {
            ret = filter_stream(bsf_ctx, pkt);
        }
        
		AVCodecParameters *codecpar = p->frtCtx->streams[i]->codecpar;
		if((pkt->data[4] & 0x1f) == 5) && codecpar->extradata_size > 0){
			//store sps pps I frame 
			AVPacket tmppacket;
			char buffer[1024 * 1024];
			memcpy(buffer, codecpar->extradata, codecpar->extradata_size);
			memcpy(buffer + codecpar->extradata_size, pkt->data, pkt->size);

			tmppacket.data = (uint8_t *)buffer;
			tmppacket.size = codecpar->extradata_size + pkt->size;
			
			decode_simple(tmppacket);
		}else {
			decode_simple(pkt);
		}

    } else {
        fprintf(stdout, "read a packet, not a video frame\n");
    }
    av_packet_unref(pkt);
}

//---------------------------------------------------------

int open_bitstream_filter(AVStream *stream, const char *name) {
    int ret = 0;
    const AVBitStreamFilter *filter = av_bsf_get_by_name(name);
    if (!filter) {
        ret = -1;
        fprintf(stderr, "Unknow bitstream filter.\n");
    }
    if((ret = av_bsf_alloc(filter, &bsf_ctx) < 0)) {
        fprintf(stderr, "av_bsf_alloc failed\n");
        return ret;
    }
    if ((ret = avcodec_parameters_copy(bsf_ctx->par_in, stream->codecpar)) < 0) {
        fprintf(stderr, "avcodec_parameters_copy failed, ret=%d\n", ret);
        return ret;
    }
    if ((ret = av_bsf_init(bsf_ctx)) < 0) {
        fprintf(stderr, "av_bsf_init failed, ret=%d\n", ret);       
        return ret;
    }
    return ret;
}
int close_bitstream_filter(){
    if(is_annexb){
        if (bsf_ctx)
            av_bsf_free(&bsf_ctx);
    }
}

int filter_stream( AVPacket *pkt) {
    int ret = 0;
    if (ret = av_bsf_send_packet(bsf_ctx, pkt) < 0) {
        fprintf(stderr, "av_bsf_send_packet failed, ret=%d\n", ret);
        return ret;
    } 
    while ((ret = av_bsf_receive_packet(bsf_ctx, pkt) == 0)) {
        // ret = write_output(of, pkt); 
        // av_packet_unref(pkt);
        // if (ret < 0) return ret;
    }
    
    //if (ret == AVERROR(EAGAIN)) return 0;
    
    return ret;
}

```