nyanmisaka/ffmpeg-rockchip #68: why do not support drawtext

yuyinw, 2024-05-23:
I  installed libxml2-v2.9.14、fribidi-1.0.12、freetype-2.10.0、fontconfig-2.13.95

[image]

nyanmisaka, 2024-05-23:
Building ffmpeg is non-trivial for beginners. You can use a pre-built binary instead.

https://github.com/MarcA711/Rockchip-FFmpeg-Builds/releases

yuyinw, 2024-05-24:
thank you very much !!!  use Rockchip-FFmpeg-Builds  can work !

i have a question:   the drawtext   defaults to using CPU to overlay watermarks,   will we consider using GPU to add watermarks in the future

nyanmisaka, 2024-05-24:
This is beyond what MPP/RGA runtime can support.

You can use the following command to achieve zero-copy, but the text will still be rendered by the CPU.

`-vf "scale_rkrga=format=nv12,hwmap,format=nv12,drawtext=...,hwmap"`

closed 2024-05-25 by nyanmisaka

xunfeng1980, 2025-09-24:
--enable-libharfbuzz ..............