nyanmisaka/ffmpeg-rockchip #71: Orangepi 5 Plus hdmi has stopped working - was working fine

david3a, 2024-05-26:
I have been using this repo to scale and encode HDMI inouts from 1080p25 sources. 

I am now  trying to get 4kp30/60 working but generally got green screens with the v4l test bench and errors like the following with ffmpeg

```
ffmpeg version a3239ac0c2 Copyright (c) 2000-2024 the FFmpeg developers
  built with gcc 11 (Ubuntu 11.4.0-1ubuntu1~22.04)
  configuration: --prefix=/usr --enable-gpl --enable-version3 --enable-libdrm --enable-rkmpp --enable-rkrga
  libavutil      59.  8.100 / 59.  8.100
  libavcodec     61.  3.100 / 61.  3.100
  libavformat    61.  1.100 / 61.  1.100
  libavdevice    61.  1.100 / 61.  1.100
  libavfilter    10.  1.100 / 10.  1.100
  libswscale      8.  1.100 /  8.  1.100
  libswresample   5.  1.100 /  5.  1.100
  libpostproc    58.  1.100 / 58.  1.100
[video4linux2,v4l2 @ 0x55b5b624c0] ioctl(VIDIOC_G_INPUT): Inappropriate ioctl for device
[video4linux2,v4l2 @ 0x55b5b624c0] ioctl(VIDIOC_G_PARM): Inappropriate ioctl for device
[video4linux2,v4l2 @ 0x55b5b624c0] Time per frame unknown
[video4linux2,v4l2 @ 0x55b5b624c0] ioctl(VIDIOC_REQBUFS): Cannot allocate memory
[in#0 @ 0x55b5b408e0] Error opening input: Cannot allocate memory
Error opening input file /dev/video0.
Error opening input files: Cannot allocate memory

```
have switched back to 1080p25 andf get same results, tried testing with the v4l2 utils, 

```
v4l2-ctl -d 0 --all
Driver Info:
	Driver name      : rk_hdmirx
	Card type        : rk_hdmirx
	Bus info         : fdee0000.hdmirx-controller
	Driver version   : 6.1.43
	Capabilities     : 0x84201000
		Video Capture Multiplanar
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps      : 0x04201000
		Video Capture Multiplanar
		Streaming
		Extended Pix Format
Priority: 2
DV timings:
	Active width: 1920
	Active height: 1080
	Total width: 2200
	Total height: 1125
	Frame format: progressive
	Polarities: -vsync -hsync
	Pixelclock: 148500000 Hz (60.00 frames per second)
	Horizontal frontporch: 88
	Horizontal sync: 44
	Horizontal backporch: 148
	Vertical frontporch: 4
	Vertical sync: 5
	Vertical backporch: 36
	Standards: 
	Flags: 
DV timings capabilities:
	Minimum Width: 640
	Maximum Width: 4096
	Minimum Height: 480
	Maximum Height: 2160
	Minimum PClock: 20000000
	Maximum PClock: 600000000
	Standards: CTA-861
	Capabilities: Interlaced, Progressive
Format Video Capture Multiplanar:
	Width/Height      : 1920/1080
	Pixel Format      : 'BGR3' (24-bit BGR 8-8-8)
	Field             : None
	Number of planes  : 1
	Flags             : premultiplied-alpha, 0x000000fe
	Colorspace        : sRGB
	Transfer Function : Unknown (0x000000b8)
	YCbCr/HSV Encoding: Unknown (0x000000ff)
	Quantization      : Limited Range
	Plane 0           :
	   Bytes per Line : 5760
	   Size Image     : 6220800

User Controls

            audio_sampling_rate 0x00981a80 (int)    : min=0 max=768000 step=1 default=0 value=44100 flags=read-only, volatile
                  audio_present 0x00981a81 (bool)   : default=0 value=0 flags=read-only, volatile

Digital Video Controls

                  power_present 0x00a00964 (bitmask): max=0x00000001 default=0x00000000 value=1 flags=read-only

```

and

```
v4l2-compliance
v4l2-compliance 1.22.1-8, 64 bits, 64-bit time_t
v4l2-compliance SHA: 9669a0273341 2023-02-16 08:09:27

Compliance test for rk_hdmirx device /dev/video0:

Driver Info:
	Driver name      : rk_hdmirx
	Card type        : rk_hdmirx
	Bus info         : fdee0000.hdmirx-controller
	Driver version   : 6.1.43
	Capabilities     : 0x84201000
		Video Capture Multiplanar
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps      : 0x04201000
		Video Capture Multiplanar
		Streaming
		Extended Pix Format

Required ioctls:
		fail: v4l2-compliance.cpp(639): missing bus_info prefix ('fdee0000.hdmirx-controller')
	test VIDIOC_QUERYCAP: FAIL
		fail: v4l2-compliance.cpp(783): doioctl(node, ioc, nullptr) != ENOTTY
	test invalid ioctls: FAIL

Allow for multiple opens:
	test second /dev/video0 open: OK
		fail: v4l2-compliance.cpp(639): missing bus_info prefix ('fdee0000.hdmirx-controller')
	test VIDIOC_QUERYCAP: FAIL
	test VIDIOC_G/S_PRIORITY: OK
	test for unlimited opens: OK

Debug ioctls:
	test VIDIOC_DBG_G/S_REGISTER: OK (Not Supported)
	test VIDIOC_LOG_STATUS: OK

Input ioctls:
	test VIDIOC_G/S_TUNER/ENUM_FREQ_BANDS: OK (Not Supported)
	test VIDIOC_G/S_FREQUENCY: OK (Not Supported)
	test VIDIOC_S_HW_FREQ_SEEK: OK (Not Supported)
	test VIDIOC_ENUMAUDIO: OK (Not Supported)
		fail: v4l2-test-input-output.cpp(462): G_INPUT not supported for a capture device
	test VIDIOC_G/S/ENUMINPUT: FAIL
	test VIDIOC_G/S_AUDIO: OK (Not Supported)
	Inputs: 0 Audio Inputs: 0 Tuners: 0

Output ioctls:
	test VIDIOC_G/S_MODULATOR: OK (Not Supported)
	test VIDIOC_G/S_FREQUENCY: OK (Not Supported)
	test VIDIOC_ENUMAUDOUT: OK (Not Supported)
	test VIDIOC_G/S/ENUMOUTPUT: OK (Not Supported)
	test VIDIOC_G/S_AUDOUT: OK (Not Supported)
	Outputs: 0 Audio Outputs: 0 Modulators: 0

Input/Output configuration ioctls:
	test VIDIOC_ENUM/G/S/QUERY_STD: OK (Not Supported)
	test VIDIOC_ENUM/G/S/QUERY_DV_TIMINGS: OK (Not Supported)
	test VIDIOC_DV_TIMINGS_CAP: OK (Not Supported)
	test VIDIOC_G/S_EDID: OK (Not Supported)

Control ioctls:
	test VIDIOC_QUERY_EXT_CTRL/QUERYMENU: OK
	test VIDIOC_QUERYCTRL: OK
	test VIDIOC_G/S_CTRL: OK
	test VIDIOC_G/S/TRY_EXT_CTRLS: OK
	test VIDIOC_(UN)SUBSCRIBE_EVENT/DQEVENT: OK
	test VIDIOC_G/S_JPEGCOMP: OK (Not Supported)
	Standard Controls: 3 Private Controls: 2

Format ioctls:
		fail: v4l2-test-formats.cpp(263): fmtdesc.description mismatch: was 'Y/UV 4:4:4', expected 'Y/CbCr 4:4:4'
	test VIDIOC_ENUM_FMT/FRAMESIZES/FRAMEINTERVALS: FAIL
	test VIDIOC_G/S_PARM: OK (Not Supported)
	test VIDIOC_G_FBUF: OK (Not Supported)
		fail: v4l2-test-formats.cpp(364): ycbcr_enc >= 0xff
		fail: v4l2-test-formats.cpp(481): testColorspace(!node->is_io_mc, pix_mp.pixelformat, pix_mp.colorspace, pix_mp.ycbcr_enc, pix_mp.quantization)
	test VIDIOC_G_FMT: FAIL
		fail: v4l2-test-formats.cpp(734): Video Capture Multiplanar: TRY_FMT(G_FMT) != G_FMT
		warn: v4l2-test-formats.cpp(784): TRY_FMT cannot handle an invalid pixelformat.
		warn: v4l2-test-formats.cpp(785): This may or may not be a problem. For more information see:
		warn: v4l2-test-formats.cpp(786): http://www.mail-archive.com/linux-media@vger.kernel.org/msg56550.html
	test VIDIOC_TRY_FMT: FAIL
		warn: v4l2-test-formats.cpp(1147): S_FMT cannot handle an invalid pixelformat.
		warn: v4l2-test-formats.cpp(1148): This may or may not be a problem. For more information see:
		warn: v4l2-test-formats.cpp(1149): http://www.mail-archive.com/linux-media@vger.kernel.org/msg56550.html
		fail: v4l2-test-formats.cpp(1175): Video Capture Multiplanar: S_FMT(G_FMT) != G_FMT
	test VIDIOC_S_FMT: FAIL
	test VIDIOC_G_SLICED_VBI_CAP: OK (Not Supported)
	test Cropping: OK (Not Supported)
	test Composing: OK (Not Supported)
	test Scaling: OK (Not Supported)

Codec ioctls:
	test VIDIOC_(TRY_)ENCODER_CMD: OK (Not Supported)
	test VIDIOC_G_ENC_INDEX: OK (Not Supported)
	test VIDIOC_(TRY_)DECODER_CMD: OK (Not Supported)

Buffer ioctls:
		fail: v4l2-test-buffers.cpp(607): q.reqbufs(node, 1)
	test VIDIOC_REQBUFS/CREATE_BUFS/QUERYBUF: FAIL
		fail: v4l2-test-buffers.cpp(783): VIDIOC_EXPBUF is supported, but the V4L2_MEMORY_MMAP support is missing or malfunctioning.
		fail: v4l2-test-buffers.cpp(784): VIDIOC_EXPBUF is supported, but the V4L2_MEMORY_MMAP support is missing, probably due to earlier failing format tests.
	test VIDIOC_EXPBUF: OK (Not Supported)
	test Requests: OK (Not Supported)

Total for rk_hdmirx device /dev/video0: 45, Succeeded: 36, Failed: 9, Warnings: 6

```

Seems the v4l2 is "broken" in some way, any ideas or suggestions? It was looking for really good till then!

David

nyanmisaka, 2024-05-26:
HDMI RX is disabled by default on OrangePi 5 Plus due to a kernel issue.

- https://github.com/armbian/linux-rockchip/pull/173
- https://github.com/Joshua-Riek/ubuntu-rockchip/issues/758

david3a, 2024-05-26:
hi, had loaded the hdmi overlay through the orangepi-config tool, have taken it out and will retest after re-adding it again

nyanmisaka, 2024-05-26:
> hi, had loaded the hdmi overlay through the orangepi-config tool, have taken it out and will retest after re-adding it again

I assume that you are using an vendor-provided imaged bc you mentioned the `orangepi-config` tool.

But the hdmi-rx controller is also disabled in their linux 6.1 kernel tree.

https://github.com/orangepi-xunlong/linux-orangepi/blob/bf29fd9652702c307fd4e636f457ac582cedb239/arch/arm64/boot/dts/rockchip/rk3588-orangepi-5-plus.dtsi#L394-L395

https://github.com/orangepi-xunlong/linux-orangepi/blob/bf29fd9652702c307fd4e636f457ac582cedb239/arch/arm64/boot/dts/rockchip/rk3588-orangepi-5-plus.dts#L323-L324

david3a, 2024-05-26:
after disabaling and re-enabling the overlay still the same, load average is 0.0

puzzled how its gone from working to not working, 

is it worth trying the older kernel? current is

```
uname -a
Linux orangepi5plus 6.1.43-rockchip-rk3588 #1.0.8 SMP Fri Feb  2 18:34:56 CST 2024 aarch64 aarch64 aarch64 GNU/Linux
```

david3a, 2024-05-26:
any suggestions on a different distro or version?

nyanmisaka, 2024-05-26:
It may work on legacy kernel 5.10.160, but I doubt you'll have difficulty downgrading.

HDMI-RX on OPi 5 Plus is already disabled by default on most famous distros, such as Armbian and Ubuntu-rockchip.

You can try https://github.com/Joshua-Riek/ubuntu-rockchip and use `overlays=rk3588-hdmirx` to enable the hdmi-rx controller.

How to use overlay: https://github.com/Joshua-Riek/ubuntu-rockchip/wiki/Ubuntu-24.04-LTS

david3a, 2024-05-26:
i'll try that, was just reading through that repo, thanks