nyanmisaka/ffmpeg-rockchip #72: rkmpp没有成功调用硬解码

sztukai, 2024-05-29:
使用命令：
ffmpeg -rtsp_transport tcp -hwaccel rkmpp -i "rtsp://admin:Admin707@192.168.1.64:554/h265/ch1/main/av_stream" -f rawvideo -pix_fmt bgr24 -max_muxing_queue_size 1024 /dev/null
输出：
frame=  224 fps= 34 q=-0.0 Lsize= 1360800kB time=00:00:08.92 bitrate=1249739.2kbits/s dup=48 drop=1 speed=1.35x
top显示ffmpeg的cpu占用为200%

环境：增加访问权限显示没有这些目录
无法访问 '/dev/iep': 没有那个文件或目录
无法访问 '/dev/vpu_service': 没有那个文件或目录
无法访问 '/dev/hevc_service': 没有那个文件或目录
无法访问 '/dev/rkvdec': 没有那个文件或目录
无法访问 '/dev/rkvenc': 没有那个文件或目录
无法访问 '/dev/vepu': 没有那个文件或目录
无法访问 '/dev/h265e': 没有那个文件或目录

sztukai, 2024-05-29:
Thank you for your response. After running the command you provided, I encountered the following error. Is my environment missing some kernels?

root@linux:/home/zysj-sz/Downloads# ffmpeg -rtsp_transport tcp -hwaccel rkmpp -hwaccel_output_format drm_prime -afbc rga -i "rtsp://admin:Admin707@192.168.1.64:554/h265/ch1/main/av_stream" -vf “scale_rkrga=format=bgr24,hwmap,format=bgr24” -f rawvideo -max_muxing_queue_size 1024 /dev/null
ffmpeg version 2093e2b Copyright (c) 2000-2023 the FFmpeg developers
  built with gcc 9 (Ubuntu 9.4.0-1ubuntu1~20.04.2)
  configuration: --prefix=/usr --enable-gpl --enable-version3 --enable-libdrm --enable-rkmpp --enable-rkrga
  libavutil      58. 29.100 / 58. 29.100
  libavcodec     60. 31.102 / 60. 31.102
  libavformat    60. 16.100 / 60. 16.100
  libavdevice    60.  3.100 / 60.  3.100
  libavfilter     9. 12.100 /  9. 12.100
  libswscale      7.  5.100 /  7.  5.100
  libswresample   4. 12.100 /  4. 12.100
  libpostproc    57.  3.100 / 57.  3.100
Input #0, rtsp, from 'rtsp://admin:Admin707@192.168.1.64:554/h265/ch1/main/av_stream':
  Metadata:
    title           : Media Presentation
  Duration: N/A, start: 0.000000, bitrate: N/A
  Stream #0:0: Video: h264 (Main), yuvj420p(pc, bt709, progressive), 1920x1080, 25 fps, 25 tbr, 90k tbn
  Stream #0:1: Audio: pcm_alaw, 8000 Hz, mono, s16, 64 kb/s
[AVFilterGraph @ 0x5599a4ca80] No such filter: '“scale_rkrga'
[vost#0:0/rawvideo @ 0x55998f6b80] Error initializing a simple filtergraph
Error opening output file /dev/null.
Error opening output files: Filter not found

nyanmisaka, 2024-05-29:
Try this again.
```
ffmpeg -rtsp_transport tcp -hwaccel rkmpp -hwaccel_output_format drm_prime -afbc rga -i "rtsp://admin:Admin707@192.168.1.64:554/h265/ch1/main/av_stream" -vf "scale_rkrga=format=bgr24,hwmap,format=bgr24" -f rawvideo -max_muxing_queue_size 1024 /dev/null
```

sztukai, 2024-05-30:
Thank you for your help. The problem has been resolved.

closed 2024-05-30 by nyanmisaka