nyanmisaka/ffmpeg-rockchip #73: What am I doing wrong with compiling? (ERROR: libdrm not found using pkg-config)

dison4linux, 2024-05-31:
I'm following the compilation procedure in the wiki pretty much word for word but am running into the error:
`ERROR: libdrm not found using pkg-config`

LibDRM is installed
```
$ sudo dpkg-query -l | grep libdrm
ii  libdrm-amdgpu1:arm64                    2.4.120-2build1                         arm64        Userspace interface to amdgpu-specific kernel DRM services -- runtime
ii  libdrm-common                           2.4.120-2build1                         all          Userspace interface to kernel DRM services -- common files
ii  libdrm-dev:arm64                        2.4.120-2build1                         arm64        Userspace interface to kernel DRM services -- development files
ii  libdrm-etnaviv1:arm64                   2.4.120-2build1                         arm64        Userspace interface to etnaviv-specific kernel DRM services -- runtime
ii  libdrm-freedreno1:arm64                 2.4.120-2build1                         arm64        Userspace interface to msm/kgsl kernel DRM services -- runtime
ii  libdrm-nouveau2:arm64                   2.4.120-2build1                         arm64        Userspace interface to nouveau-specific kernel DRM services -- runtime
ii  libdrm-radeon1:arm64                    2.4.120-2build1                         arm64        Userspace interface to radeon-specific kernel DRM services -- runtime
ii  libdrm-tegra0:arm64                     2.4.120-2build1                         arm64        Userspace interface to tegra-specific kernel DRM services -- runtime
ii  libdrm2:arm64                           2.4.120-2build1                         arm64        Userspace interface to kernel DRM services -- runtime
```
But on the configure step for ffmpeg it complains that libdrm is not found...
```
$ ./configure --prefix=/usr --enable-gpl --enable-version3 --enable-libdrm --enable-rkmpp --enable-rkrga
ERROR: libdrm not found using pkg-config

If you think configure made a mistake, make sure you are using the latest
version from Git.  If the latest version fails, report the problem to the
ffmpeg-user@ffmpeg.org mailing list or IRC #ffmpeg on irc.libera.chat.
Include the log file "ffbuild/config.log" produced by configure as this will help
solve the problem.
```
The contents of that file mentioned contain the same error
`ERROR: libdrm not found using pkg-config`

Tryanks, 2024-05-31:
Try checking if the file `libdrm.pc` exists on your device. On my device, it can be found at the following path:
> /usr/local/lib/pkgconfig/libdrm.pc

Next, set the `PKG_CONFIG_PATH` environment variable:
```
export PKG_CONFIG_PATH="$PKG_CONFIG_PATH:/usr/local/lib/pkgconfig"
```
Then try compiling again.

Regarding your compilation issue, this is not a problem solely related to this repository. Please consider using other community features instead of an issue.

nyanmisaka, 2024-05-31:
https://github.com/nyanmisaka/ffmpeg-rockchip/wiki/Compilation#examples

```
$ dpkg-query -l | grep -E "pkgconf|pkg-config"
ii  libpkgconf3:arm64                          1.8.1-1                              arm64        shared library for pkgconf
ii  pkg-config:arm64                           1.8.1-1                              arm64        manage compile and link flags for libraries (transitional package)
ii  pkgconf:arm64                              1.8.1-1                              arm64        manage compile and link flags for libraries
ii  pkgconf-bin                                1.8.1-1                              arm64        manage compile and link flags for libraries (binaries)

$ ls -lh /usr/lib/aarch64-linux-gnu/pkgconfig/ | grep libdrm.pc
-rw-r--r-- 1 root root  256 Mar 23  2023 libdrm.pc
```

dison4linux, 2024-05-31:
Thanks, it was installing pkgconf itself that was needed I missed that listed in the guide...
"
...such as git,meson,cmake,**pkg-config**,gcc,libdrm-dev,etc...
"

closed 2024-05-31 by dison4linux