nyanmisaka/ffmpeg-rockchip #75: difference between h264_rkmpp and hevc_rkmpp

ZH-1225, 2024-06-14:
./ffmpeg -i /home/pi/Videos/test_ffmpeg.mp4 -c:v hevc_rkmpp -c:a copy h264_rkmpp.mp4
./ffmpeg -i /home/pi/Videos/test_ffmpeg.mp4 -c:v hevc_rkmpp -c:a copy hevc_rkmpp.mp4
I used two commands to find that the generated MP4 files were about the same size

Tryanks, 2024-06-14:
./ffmpeg -i /home/pi/Videos/test_ffmpeg.mp4 -c:v **hevc_rkmpp** -c:a copy h264_rkmpp.mp4
./ffmpeg -i /home/pi/Videos/test_ffmpeg.mp4 -c:v **hevc_rkmpp** -c:a copy hevc_rkmpp.mp4

ZH-1225, 2024-06-14:
> 制 h264

./ffmpeg -i /home/pi/Videos/test_ffmpeg.mp4 -c:v h264_rkmpp -c:a copy h264_rkmpp.mp4
./ffmpeg -i /home/pi/Videos/test_ffmpeg.mp4 -c:v hevc_rkmpp -c:a copy hevc_rkmpp.mp4

ZH-1225, 2024-06-14:
> both using same encoder hevc_rkmpp (-c:v hevc_rkmpp )

./ffmpeg -i /home/pi/Videos/test_ffmpeg.mp4 -c:v h264_rkmpp -c:a copy h264_rkmpp.mp4
./ffmpeg -i /home/pi/Videos/test_ffmpeg.mp4 -c:v hevc_rkmpp -c:a copy hevc_rkmpp.mp4

thanks!

ZH-1225, 2024-06-14:
> ./ffmpeg -i /home/pi/Videos/test_ffmpeg.mp4 -c:v **hevc_rkmpp** -c:a copy h264_rkmpp.mp4 ./ffmpeg -i /home/pi/Videos/test_ffmpeg.mp4 -c:v **hevc_rkmpp** -c:a copy hevc_rkmpp.mp4
./ffmpeg -i /home/pi/Videos/test_ffmpeg.mp4 -c:v h264_rkmpp -c:a copy h264_rkmpp.mp4
./ffmpeg -i /home/pi/Videos/test_ffmpeg.mp4 -c:v hevc_rkmpp -c:a copy hevc_rkmpp.mp4

thanks!

nyanmisaka, 2024-06-14:
You have to set the bitrate or quality factor manually, otherwise the default bitrate of rkmpp encoders is 2000k.

https://github.com/nyanmisaka/ffmpeg-rockchip/blob/9efe5bcff082d6538702d01c9b7126f40da27361/libavcodec/rkmppenc.h#L245

closed 2024-06-14 by nyanmisaka