nyanmisaka/ffmpeg-rockchip #78: Upstreaming to FFmpeg

ramiropolla, 2024-07-01:
Hi @nyanmisaka,

How far along do you think this branch is? Fully functional, mostly functional...?

I want to start upstreaming it. Could you please rebase to master, perform the fixups, and do some commit cleanup?

I can do the final testing and review process on ffmpeg-devel. (I can do the rebase and cleanup as well, but perhaps you have some time and would like to do it yourself).

Ramiro

nyanmisaka, 2024-07-01:
The big elephant in the room is https://github.com/rockchip-linux/mpp/issues/530

I think upstream FFmpeg won't accept any more MPP related stuff until it is properly fixed.

hbiyik, 2024-07-02:
one suggestion would be to drop rga stuff completely when upstreaming, librga is also a wonky unmaintained dependency. However @nyanmisaka has several use cases depending rga based scalers.

PucklaJ, 2026-04-03:
> 2. The upstream prefers to use V4L2 Requests rather than adding another hwconext_rkmpp

Excuse me if this is off-topic, but I am very interested in this. What do you mean by this statement? Is the rkmpp hardware acceleration of upstream ffmpeg v4l2-requests based or do you mean that upstream prefers the v4l2-requests hardware acceleration over the rkmpp hardware acceleration, since the v4l2-requests is part of mainline linux, while rkmpp still requires a vendor kernel?

ramiropolla, 2026-04-05:
@nyanmisaka could you *please* rebase to master, perform the fixups, and do some commit cleanup? It's too hard to review this way. Some of the changes have already been done upstream.