nyanmisaka/ffmpeg-rockchip #85: h264_rkmpp decoder in ffmpeg: avcodec_receive_frame (code = -11)"

tu-nie, 2024-07-09:
您好，可以看一看什么问题吗，这是代码

#include<iostream>
#include<thread>
#include<string.h>
#include "decode_replay.h"

using namespace std;
QString Decode_Replay::replay_path=nullptr;

int useHardwareDecoding = 1; // 启用硬解码

int width = 0;
int height = 0;

static enum AVPixelFormat hw_pix_fmt = AV_PIX_FMT_DRM_PRIME; //硬件支持的像素格式
static enum AVPixelFormat get_hw_format(AVCodecContext *ctx,const enum AVPixelFormat *pix_fmts)
{
    const enum AVPixelFormat *p;
    for (p = pix_fmts; *p != -1; p++) {
        if (*p == hw_pix_fmt)
            return *p;
    }
    fprintf(stderr, "Failed to get HW surface format.\n");
    return AV_PIX_FMT_NONE;
}

void Decode_Replay::hardwave_init(AVCodecContext *codecCtx,const AVCodec * codec){
    //设置硬解码器
    AVHWDeviceType deviceType = AV_HWDEVICE_TYPE_DRM; // 根据架构设置硬解码类型
    AVBufferRef * bufferRef = av_hwdevice_ctx_alloc(deviceType);
    cout << "alloc hwdevice success address : " << bufferRef << endl;
    int ret = av_hwdevice_ctx_init(bufferRef);
    if(ret != 0){
        cout << "init hardwave device context failed!" << endl;
    }
    codecCtx->hw_device_ctx = av_buffer_ref(bufferRef);
}

Decode_Replay::Decode_Replay(QString path)
{
    this->videoIndex = -1; //记录视频流下标
    this->replay_path = path;
    this->play_speed = 60;
    this->status = 0;
    this->screenshots_mark = 0;
}

Decode_Replay::~Decode_Replay()
{
    if (packet) {
        av_packet_unref(packet);
        packet = NULL;
    }
    if (swsCtx) {
        sws_freeContext(swsCtx);
        swsCtx = 0;
    }
    if (frame) {
        av_frame_free(&frame);
        frame = 0;
    }
    if (frameHW) {
        av_frame_free(&frameHW);
        frameHW = 0;
    }
    if (codecCtx->hw_device_ctx) {
        av_buffer_unref(&codecCtx->hw_device_ctx);
    }
    if (codecCtx) {
[... 185 lines cut ...]
                    int ret = av_hwframe_transfer_data(frameHW, frame, 0);
                    if (ret < 0) {
                        fprintf(stderr, "Error: av_hwframe_transfer_data failed with error code %d\n", ret);
                        continue;
                    }
                    sws_scale(swsCtx, (const uint8_t* const*)frameHW->data, frameHW->linesize, 0, height, frameYUV->data, frameYUV->linesize);
                } else {
                    sws_scale(swsCtx, (const uint8_t* const*)frame->data, frame->linesize, 0, height, frameYUV->data, frameYUV->linesize);
                }
                QImage img(frameYUV->data[0], width, height, QImage::Format_Indexed8);
//                emit sendImg2(img);
                if (this->screenshots_mark == 1)
                {
                    QString image_name = this->getTime();
                    QString path = SetControl::getInstance()->getSetting_imagepath() + "/" + "shots_" + image_name + ".jpg";
                    img.save(path);
                    ImageControl::getInstance()->addImagePath(image_name, path);
                    this->screenshots_mark = 0;
                }
            }
        }
        av_packet_unref(packet);
    }
    end = clock();
}

nyanmisaka, 2024-07-09:
I can't review your code. But I guess you can't handle the default async API.

Check this. https://github.com/JeffyCN/FFmpeg/commit/aba28f2c1bb22d7929bb329e732f7cebdcd7bd71

closed 2024-07-16 by nyanmisaka

wzw88486969, 2024-10-14:
`    for (int i = 0; i < ifmt_ctx->nb_streams; i++) {
        if (ifmt_ctx->streams[i]->codecpar->codec_type == AVMEDIA_TYPE_VIDEO) {
            video_stream_index = i;
            is_annexb = strcmp(av_fourcc2str(ifmt_ctx->streams[i]->codecpar->codec_tag), "avc1") == 0 ? 0 : 1;
            break;
        }
    }

    if (video_stream_index == -1) {
        fprintf(stderr, "no video stream found.\n");
        goto end;
    }

    fprintf(stdout, "is_annexb=%d\n", is_annexb);  
    if (!is_annexb) {
        if (ret = open_bitstream_filter(ifmt_ctx->streams[video_stream_index], &bsf_ctx, "h264_mp4toannexb") < 0) {
           fprintf(stderr, "open_bitstream_filter failed, ret=%d\n", ret);  
           goto end;
        }
    }
    while (av_read_frame(ifmt_ctx, pkt) >= 0) {
        if (pkt->stream_index == video_stream_index) {
            if (is_annexb) {
                ret = write_output(of, pkt);
            } else {
                ret = filter_stream(bsf_ctx, pkt, of, 0);
            }
            if (ret < 0) goto end;
        } else {
            fprintf(stdout, "read a packet, not a video frame\n");
        }
        av_packet_unref(pkt);
    }
    if (!is_annexb) {//flush bistream filter
        filter_stream(bsf_ctx, NULL, of, 1);
    }`

use h264_mp4toannexb, then to do decode

wzw88486969, 2024-10-15:
`
AVCodecParameters *codecpar = p->frtCtx->streams[i]->codecpar; 
if(codecpar->extradata_size > 0){
      //store sps pps
}

use h264_mp4toannexb, then to do decode

sps + pps + I frame Feeding mpp
`