nyanmisaka/ffmpeg-rockchip #86: Not really able to play transcoded files

soetz, 2024-07-10:
Hey, I got a few files that contain a h264 video stream that I’d like to transcode to hevc purely to save space. I’m trying to use this fork to make use of my chip’s hardware acceleration. I’m using a RockPi 5C which is equipped with a RK3588S2.

I’ve built ffmpeg using the instructions from the wiki. I don’t think that I’ve made any mistake but it also wouldn’t be surprising considering that I’m really not familiar with building C, using `make`, etc.

I’ve tried different flavors of this command that is heavily inspired from the transcoding guide in the wiki, without success:

`ffmpeg -hwaccel rkmpp -hwaccel_output_format drm_prime -afbc rga -i input-file.mkv -c:a copy -c:s copy -strict -2 -c:v hevc_rkmpp -rc_mode VBR -b:v 6M -maxrate 6M -bufsize 12M -g:v 120 -map 0:0 -map 0:1 -map 0:2 -map 0:3 -y output-file.mkv`

I’m not getting any error or warning from the command line. Regardless of the original file, I always get a file with the first few frames working as expected, but then no more image at all for the rest of the file. The audio and subtitle streams still work, of course. The file looks like it’s approximately the right size. I use VLC on a completely different machine to check the result.

My best guess would be that I’m not using the right command options. I understand most of them but not all (for instance `-rc_mode`).

In any case, thanks for maintaining this fork!

nyanmisaka, 2024-07-11:
Please provide the ffmpeg output log.

nyanmisaka, 2024-07-11:
Which rockchip linux kernel version are you using? `uname -a`

soetz, 2024-07-11:
`Linux rock-5c 5.10.110-37-rockchip #27a257394 SMP Thu May 23 02:38:59 UTC 2024 aarch64 GNU/Linux`

nyanmisaka, 2024-07-11:
5.10.110 is from 2022, which is very old for RK3588, I have not verified ffmpeg-rockchip on it. The MPP kernel driver in it may contain many bugs, but they have been fixed in the latest kernel.

Can you try [Ubuntu-Rockchip](https://github.com/Joshua-Riek/ubuntu-rockchip/releases/tag/v2.2.1) or [Armbian](https://www.armbian.com/radxa-rock-5c/) images? They use the newer linux 6.1 kernel.

soetz, 2024-07-11:
I see. I’ll try Armbian. Thanks for your help!

soetz, 2024-07-12:
Hey, so as mentioned, I tried with Armbian, which has `Linux rock-5c 6.1.43-vendor-rk35xx #1 SMP Mon May 20 12:26:58 UTC 2024 aarch64 GNU/Linux`, but unfortunately I still get exactly the same issue.

I tried to convert various other h264 videos from my collection and in this case it did work. They probably don’t have exactly the same properties (profile, bitrate, pixel format, etc) though.

I also tried to convert the files that fail with `ffmpeg-rockchip` using regular ffmpeg on another machine, and it worked normally.

So there’s something about this collection of videos that makes `ffmpeg-rockchip` trip specifically. I really don’t see what that could be though.

I guess that I’ll close this issue, keep using `ffmpeg-rockchip` and if I get this problem with another set of videos and/or notice a pattern I’ll open another issue.

Thanks again!

closed 2024-07-12 by soetz

nyanmisaka, 2024-07-12:
@soetz If possible please share the video clip so I can analyze it. You can also try it with this [portable ffmpeg binary](https://github.com/MarcA711/Rockchip-FFmpeg-Builds/releases/tag/6.1-5).

soetz, 2024-07-12:
(note that I tried to transcode the HEVC version that I successfully obtained using another machine back to h264 and that this failed as well. I’m starting to think that the problem might be the decimal framerate?)

soetz, 2024-07-12:
> @soetz If possible please share the video clip so I can analyze it. You can also try it with this [portable ffmpeg binary](https://github.com/MarcA711/Rockchip-FFmpeg-Builds/releases/tag/6.1-5).

the result with the portable binary was the same, unfortunately

soetz, 2024-07-13:
FYI @nyanmisaka , I’ve tried using Ubuntu-Rockchip with 5.10 kernel and the `ffmpeg` package from the rockchip-multimedia repo, and it looks like it’s working as intended this time

nyanmisaka, 2024-07-13:
Amazingfate's rockchip-multimedia repo uses `ffmpeg-rockchip`. But they may have different MPP library versions.

It would be great if you could share a clip that reproduces the problem, that would help me locate the issue.

soetz, 2024-07-13:
> Amazingfate's rockchip-multimedia repo uses `ffmpeg-rockchip`. But they may have different MPP library versions.

Yeah I’m aware, that’s why I tried it. For the record, I was building MPP using Radxa’s tutorial. The test that they provide and mention in the tutorial was passing so I didn’t bother digging in that direction more.

> It would be great if you could share a clip that reproduces the problem, that would help me locate the issue.

Ah sorry, I got rid of all the faulty files. I probably have a SD card somewhere, I’ll flash it and redo the whole process.

soetz, 2024-07-20:
Hey @nyanmisaka, so I tried again with kernel 6.1 (specifically `Linux rock-5c 6.1.43-vendor-rk35xx #1 SMP Mon May 20 12:26:58 UTC 2024 aarch64 aarch64 aarch64 GNU/Linux`) using Armbian, this time with the package provided by `rockchip-multimedia`, and it failed.

Here are clips demonstrating the problem:

[Dix Pour Cent (2015) S04E02 tests.zip](https://github.com/user-attachments/files/16321543/Dix.Pour.Cent.2015.S04E02.tests.zip)

Hope they help!

# Commands used

## For `Dix Pour Cent (2015) S04E02.h264.original.mkv`

`ffmpeg -i "/media/store/test/Dix Pour Cent S04E02.mkv" -ss 0 -t 15 -map 0:0 -map 0:1 -map 0:2 -c copy -y "/tmp/Dix Pour Cent (2015) S04E02.h264.original.mkv"`

(note that this is not the "real" original and that a few frames seem to be missing at the beginning compared to the HEVC software version, I’m really curious why)

ffmpeg’s output:

```
ffmpeg version 6.1.1-3ubuntu5+git240504.09cd2a2~noble Copyright (c) 2000-2023 the FFmpeg developers
  built with gcc 13 (Ubuntu 13.2.0-23ubuntu4)
  configuration: --prefix=/usr --extra-version='3ubuntu5+git240504.09cd2a2~noble' --toolchain=hardened --libdir=/usr/lib/aarch64-linux-gnu --incdir=/usr/include/aarch64-linux-gnu --arch=arm64 --enable-gpl --disable-stripping --disable-omx --enable-gnutls --enable-libaom --enable-libass --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libdav1d --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libglslang --enable-libgme --enable-libgsm --enable-libharfbuzz --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzimg --enable-openal --enable-opencl --enable-opengl --enable-version3 --enable-rkmpp --enable-rkrga --disable-sndio --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-ladspa --enable-libbluray --enable-libjack --enable-libpulse --enable-librabbitmq --enable-librist --enable-libsrt --enable-libssh --enable-libsvtav1 --enable-libx264 --enable-libzmq --enable-libzvbi --enable-lv2 --enable-sdl2 --enable-libplacebo --enable-librav1e --enable-pocketsphinx --enable-librsvg --enable-libjxl --enable-shared
  libavutil      58. 29.100 / 58. 29.100
  libavcodec     60. 31.102 / 60. 31.102
  libavformat    60. 16.100 / 60. 16.100
  libavdevice    60.  3.100 / 60.  3.100
  libavfilter     9. 12.100 /  9. 12.100
  libswscale      7.  5.100 /  7.  5.100
  libswresample   4. 12.100 /  4. 12.100
  libpostproc    57.  3.100 / 57.  3.100
Input #0, matroska,webm, from '/media/store/test/Dix Pour Cent S04E02.mkv':
  Metadata:
    creation_time   : 2021-01-21T23:58:23.000000Z
    ENCODER         : Lavf58.29.100
  Duration: 00:54:26.34, start: 0.000000, bitrate: 3642 kb/s
  Stream #0:0: Video: h264 (High), yuv420p(tv, bt709, progressive), 1280x720 [SAR 1:1 DAR 16:9], 25 fps, 25 tbr, 1k tbn (default)
    Metadata:
      DURATION        : 00:54:25.445000000
  Stream #0:1(fre): Audio: eac3, 48000 Hz, 5.1(side), fltp, 640 kb/s (default)
    Metadata:
      title           : Surround
      DURATION        : 00:54:26.336000000
  Stream #0:2(eng): Subtitle: ass
    Metadata:
      DURATION        : 00:53:55.205000000
Output #0, matroska, to '/tmp/Dix Pour Cent (2015) S04E02.h264.original.mkv':
  Metadata:
    encoder         : Lavf60.16.100
  Stream #0:0: Video: h264 (High) (H264 / 0x34363248), yuv420p(tv, bt709, progressive), 1280x720 [SAR 1:1 DAR 16:9], q=2-31, 25 fps, 25 tbr, 1k tbn (default)
    Metadata:
      DURATION        : 00:54:25.445000000
  Stream #0:1(fre): Audio: eac3 ([0] [0][0] / 0x2000), 48000 Hz, 5.1(side), fltp, 640 kb/s (default)
    Metadata:
      title           : Surround
      DURATION        : 00:54:26.336000000
  Stream #0:2(eng): Subtitle: ass
    Metadata:
      DURATION        : 00:53:55.205000000
Stream mapping:
  Stream #0:0 -> #0:0 (copy)
  Stream #0:1 -> #0:1 (copy)
  Stream #0:2 -> #0:2 (copy)
Press [q] to stop, [?] for help
[out#0/matroska @ 0xaaaabad98650] video:4073kB audio:1172kB subtitle:0kB other streams:0kB global headers:1kB muxing overhead: 0.130530%
size=    5252kB time=00:00:14.97 bitrate=2873.0kbits/s speed= 222x
```

## For `Dix Pour Cent (2015) S04E02.hevc.sw.mkv`

`ffmpeg -hwaccel rkmpp -hwaccel_output_format drm_prime -afbc rga -i "/media/store/test/Dix Pour Cent S04E02.mkv" -ss 0 -t 15 -map 0:0 -map 0:1 -map 0:2 -c copy -strict -2 -filter:v scale_rkrga=format=nv12,hwmap=mode=read,format=nv12 -c:v hevc -rc_mode VBR -b:v "2.1M" -maxrate "2.1M" -bufsize 12M -g:v 120 -y "/tmp/Dix Pour Cent (2015) S04E02.hevc.sw.mkv"`
[... 116 lines cut ...]
      DURATION        : 00:53:55.205000000
Multiple -c, -codec, -acodec, -vcodec, -scodec or -dcodec options specified for stream 0, only the last option '-c:v hevc_rkmpp' will be used.
rga_api version 1.10.0_[8]
Stream mapping:
  Stream #0:0 -> #0:0 (h264 (h264_rkmpp) -> hevc (hevc_rkmpp))
  Stream #0:1 -> #0:1 (copy)
  Stream #0:2 -> #0:2 (copy)
Press [q] to stop, [?] for help
Output #0, matroska, to '/tmp/Dix Pour Cent (2015) S04E02.hevc.hw.mkv':
  Metadata:
    encoder         : Lavf60.16.100
[... cut ...]
  Stream #0:1(fre): Audio: eac3 ([0] [0][0] / 0x2000), 48000 Hz, 5.1(side), fltp, 640 kb/s (default)
    Metadata:
      title           : Surround
      DURATION        : 00:54:26.336000000
  Stream #0:2(eng): Subtitle: ass
    Metadata:
      DURATION        : 00:53:55.205000000
[out#0/matroska @ 0xaaab06eb9740] video:1467kB audio:1172kB subtitle:0kB other streams:0kB global headers:1kB muxing overhead: unknown
frame=  375 fps=0.0 q=-0.0 Lsize=    1477kB time=00:00:14.97 bitrate= 808.1kbits/s speed=31.1x
```

soetz, 2024-07-21:
I just tried with Ubuntu-Rockchip Noble (`Linux rock-5c 6.1.0-1020-rockchip #20-Ubuntu SMP Sat Jul 13 19:58:54 UTC 2024 aarch64 aarch64 aarch64 GNU/Linux`) and rockchip-multimedia’s ffmpeg-rockchip build and I get the problem there as well.

All in all, so far I’ve only been able to make HW acceleration work with Ubuntu-Rockchip Jammy, kernel 5.10 and the ffmpeg package from rockchip-multimedia. All other attempts have failed.

The problem is that I don’t really want to use this distro in the long run because after using it for a few hours, my machine doesn’t start anymore. Don’t know what’s wrong.

I wanted to check whether the issue was with the kernel by building a custom Armbian and forcing it to use 5.10, but either building the image failed or the machine wouldn’t boot when using it.

I’m kind of running out of options here. I guess I’ll try to make Jammy work longer, but with my proficiency that’s not a given.

I’m curious why I’m the only one reporting this kind of issues. Did I receive a faulty machine?

nyanmisaka, 2024-07-21:
Adding `-sn` to disable subtitle copying alleviates this problem. I need to investigate further.

soetz, 2024-07-21:
That might explain why I was able to transcode test videos (eg Big Buck Bunny) when I first reported the issue. Thanks for finding this! I might be able to get a suitable workflow after all, if I do a first pass to transcode the video, and then a second pass to integrate the other streams back into the final file.

soetz, 2024-07-21:
Works perfectly! Thanks again for finding this workaround. Hope you figure out what’s wrong with the subtitles!

nyanmisaka, 2024-11-18:
@soetz Just a FYI, the root cause has been fixed in some recent commits.