nyanmisaka/ffmpeg-rockchip #88: 是否支持H265编解码

xxj1213, 2024-07-19:
是否能将h265编解码加入进去

Tryanks, 2024-07-19:
https://github.com/nyanmisaka/ffmpeg-rockchip?tab=readme-ov-file#decodershwaccel

> ...
> V..... hevc_rkmpp           Rockchip MPP (Media Process Platform) HEVC decoder (codec hevc)
> ...

nyanmisaka, 2024-07-19:
h265 hw dec/enc has been supported since day one.

closed 2024-07-19 by nyanmisaka

xxj1213, 2024-07-19:
./ffmpeg -decoders | grep rmpp
V..... av1_rkmpp            Rockchip MPP (Media Process Platform) AV1 decoder (codec av1)
 V..... h263_rkmpp           Rockchip MPP (Media Process Platform) H263 decoder (codec h263)
 V..... h264_rkmpp           Rockchip MPP (Media Process Platform) H264 decoder (codec h264)
 V..... hevc_rkmpp           Rockchip MPP (Media Process Platform) HEVC decoder (codec hevc)
 V..... mpeg1_rkmpp          Rockchip MPP (Media Process Platform) MPEG1VIDEO decoder (codec mpeg1video)
 V..... mpeg2_rkmpp          Rockchip MPP (Media Process Platform) MPEG2VIDEO decoder (codec mpeg2video)
 V..... mpeg4_rkmpp          Rockchip MPP (Media Process Platform) MPEG4 decoder (codec mpeg4)
 V..... vp8_rkmpp            Rockchip MPP (Media Process Platform) VP8 decoder (codec vp8)
 V..... vp9_rkmpp            Rockchip MPP (Media Process Platform) VP9 decoder (codec vp9)

**not  h265_rkmpp**     , **why said "h265 hw dec/enc has been supported since day one."**

nyanmisaka, 2024-07-19:
hevc_rkmpp Rockchip MPP (Media Process Platform) HEVC decoder (codec hevc)

HEVC == H.265

https://en.wikipedia.org/wiki/High_Efficiency_Video_Coding

**High Efficiency Video Coding (HEVC), also known as H.265 and MPEG-H Part 2**

xxj1213, 2024-07-19:
ok, thanks