nyanmisaka/ffmpeg-rockchip #89: 关于推流的码率设置

ZH-1225, 2024-07-24:
hevc_rkmpp编码器是不是不支持自动调节bitrate,必须给-b:v一个初始值

nyanmisaka, 2024-07-24:
Use CQP rate control if you want to fix the quality factor instead of the bitrate.

https://github.com/nyanmisaka/ffmpeg-rockchip/wiki/Encoder#test-encoders

closed 2024-07-24 by nyanmisaka

ZH-1225, 2024-07-25:
> Use CQP rate control if you want to fix the quality factor instead of the bitrate.
> 
> https://github.com/nyanmisaka/ffmpeg-rockchip/wiki/Encoder#test-encoders

我如果想实现利用h265参数设置-qp_init 0 之后还能再设置-maxrate吗

nyanmisaka, 2024-07-25:
From what I can tell, it's not supported. For more info you can ask in [rockchip-linux/mpp](https://github.com/rockchip-linux/mpp).

ZH-1225, 2024-07-25:
感谢，请问一下我现在想保持画质最高用hevc_rkmpp推流h265，然后可变码率，大概的参数应该设置成什么样子，感谢

ZH-1225, 2024-07-26:
> From what I can tell, it's not supported. For more info you can ask in [rockchip-linux/mpp](https://github.com/rockchip-linux/mpp).

请问一下我用hevc_rkmpp推流的时候怎么可以减少马赛克
Could you tell me how I can reduce the Mosaic when I push the stream with hevc_rkmpp

nyanmisaka, 2024-07-26:
Macroblocking is usually caused by insufficient bitrate in the encoder. Increase the bitrate or decrease the QP value.