nyanmisaka/ffmpeg-rockchip #90: RV1103 support

14Og, 2024-07-26:
Hi there! I'm using luckfox pico mini b board, which runs on rv1103 SoC, that is not listed in rockchip mpp support list, but i have mpp headers and library in my buildroot images. So i have question: is there any chance to cross-compile your ffmpeg fork for this type of boards? I've tried to use default ffmpeg version in their buildroot sdk, but it doesn't have rkmpp support, and their h264_v4l2m2m encoder doesn't work too.  Assuming that there is mpp support for this SoC, how can i build this version of ffmpeg for target board on my host PC? FFmpeg has a great explanation on how to cross compile their library, but rockchip mpp has no documentation on how to cross-compile this library, and, as i can see, it is required to compile it on my host machine in order to build your ffmpeg version. Thanks in advance!

nyanmisaka, 2024-07-26:
> but rockchip mpp has no documentation on how to cross-compile this library,

Here is a [code snippet](https://github.com/jellyfin/jellyfin-ffmpeg/blob/02b8f47e4e063920403a52dbf196a83e5957b83a/docker-build.sh#L627-L638) showing how to compile mpp for armhf/arm64 on amd64. but i don't know what c library rv1103 uses, if it's not glibc you have to use the cross compiler provided by the vendor.

14Og, 2024-07-29:
Yep, thank you! Will try to build it for my platform

closed 2024-07-29 by 14Og