nyanmisaka/ffmpeg-rockchip #92: Can I use C++to call the MPP function of ffmpeg rockchip

jinjie412, 2024-07-29:
in FFmpeg CLI for Rockchip platforms  ，Can I use C++to call the MPP function of ffmpeg rockchip

nyanmisaka, 2024-07-29:
https://github.com/nyanmisaka/ffmpeg-rockchip/blob/master/doc/examples/qsv_decode.c
https://github.com/nyanmisaka/ffmpeg-rockchip/blob/master/doc/examples/qsv_transcode.c

Please refer to QSV examples. Usually you only need to replace the following content to adapt to RKMPP.

AV_PIX_FMT_QSV => AV_PIX_FMT_DRM_PRIME

AV_HWDEVICE_TYPE_QSV => AV_HWDEVICE_TYPE_RKMPP

h264_qsv => h264_rkmpp

closed 2024-07-30 by nyanmisaka

dwaithe, 2024-09-02:
I have setup your FFMPEG and can decode and encode video from the command line. I want to run directly  in my own code. When I run the qsv_decode.c with the above recommended changes I get the following error:

src/rkmpp_decode.c:160:47: error: ‘AV_HWDEVICE_TYPE_RKMPP’ undeclared (first use in this function); did you mean ‘AV_HWDEVICE_TYPE_DRM’?
  160 |     ret = av_hwdevice_ctx_create(&device_ref, AV_HWDEVICE_TYPE_RKMPP,
      |                                               ^~~~~~~~~~~~~~~~~~~~~~
      |                                               AV_HWDEVICE_TYPE_DRM
src/rkmpp_decode.c:160:47: note: each undeclared identifier is reported only once for each function it appears in
src/rkmpp_decode.c:161:34: error: expected expression before ‘auto’
  161 |                                  auto, NULL, 0);
      |                                  ^~~~
src/rkmpp_decode.c:160:11: error: too few arguments to function ‘av_hwdevice_ctx_create’
  160 |     ret = av_hwdevice_ctx_create(&device_ref, AV_HWDEVICE_TYPE_RKMPP,
      |           ^~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/aarch64-linux-gnu/libavcodec/avcodec.h:39,
                 from /usr/include/aarch64-linux-gnu/libavformat/avformat.h:312,
                 from src/rkmpp_decode.c:35:
/usr/include/aarch64-linux-gnu/libavutil/hwcontext.h:303:5: note: declared here
  303 | int av_hwdevice_ctx_create(AVBufferRef **device_ctx, enum AVHWDeviceType type,

Here is my ldd:
	linux-vdso.so.1 (0x0000007fbe4db000)
	libavformat.so.60 => /lib/libavformat.so.60 (0x0000007fbe203000)
	libavcodec.so.60 => /lib/libavcodec.so.60 (0x0000007fbd171000)
	libavutil.so.58 => /lib/libavutil.so.58 (0x0000007fbc095000)
	libc.so.6 => /lib/aarch64-linux-gnu/libc.so.6 (0x0000007fbbf22000)
	/lib/ld-linux-aarch64.so.1 (0x0000007fbe4ab000)
	libm.so.6 => /lib/aarch64-linux-gnu/libm.so.6 (0x0000007fbbe77000)
	libz.so.1 => /lib/aarch64-linux-gnu/libz.so.1 (0x0000007fbbe4d000)
	libpthread.so.0 => /lib/aarch64-linux-gnu/libpthread.so.0 (0x0000007fbbe1c000)
	libswresample.so.4 => /lib/libswresample.so.4 (0x0000007fbbdf5000)
	librga.so.2 => /lib/aarch64-linux-gnu/librga.so.2 (0x0000007fbbdb3000)
	liblzma.so.5 => /lib/aarch64-linux-gnu/liblzma.so.5 (0x0000007fbbd7f000)
	librockchip_mpp.so.1 => /usr/local/lib/librockchip_mpp.so.1 (0x0000007fbbb5e000)
	libdrm.so.2 => /lib/aarch64-linux-gnu/libdrm.so.2 (0x0000007fbbb3a000)
	libstdc++.so.6 => /lib/aarch64-linux-gnu/libstdc++.so.6 (0x0000007fbb955000)
	libgcc_s.so.1 => /lib/aarch64-linux-gnu/libgcc_s.so.1 (0x0000007fbb931000)

My compiler command: gcc -L/usr/local/lib -L/usr/lib -o rkmpp_decode src/rkmpp_decode.c -lrockchip_vpu -lavformat -lavcodec -lavutil -lm -lrockchip_mpp

Am I missing something?
Thanks for the great work!

nyanmisaka, 2024-09-02:
You are using the header files of vanilla ffmpeg, so of course there won't be RKMPP definitions there. `/usr/include/aarch64-linux-gnu/libavutil/hwcontext.h`

dwaithe, 2024-09-02:
Well spotted. I had the correct libraries but the wrong headers. Thanks so much! I have got it working with my code now.

Tryanks, 2024-11-18:
Hi @nyanmisaka, is it possible to obtain `AV_HWDEVICE_TYPE_RKMPP` using the method `av_hwdevice_find_type_by_name`? 

I am using [libav binding in another language](https://github.com/asticode/go-astiav/blob/master/hardware_device_type.go#L20), which doesn't provide the `HardwareDeviceTypeRKMPP` field as an available type. At the same time, there's no reason to request adapting it to a non-upstream version of FFmpeg. So my only option is to get this Type using `astiav.FindHardwareDeviceTypeByName`. However, I am not sure what the parameter Name should be, but I assume it is `rkmpp`?