nyanmisaka/ffmpeg-rockchip #94: Rock64 hw acceleration

fenouil-r, 2024-08-02:
Hello, thank you for the great work.

I'm trying to use hardware acceleration of a Rock64 board (rk3328) for Frigate.
The board is currently running OMV5 on debian from an old 'Ayufan' release, just updated the kernel from time to time.

I thought it would be easy with the '-rk' docker images provided by Frigate, but it looks like I need a BSP kernel, and this is where the adventure starts :)

I'm not familiar with linux kernels nor ffmpeg and hw acceleration... So total beginner here.
I also discover that there are 'many' kernel options for this board, and sometimes difficult to understand where they come from (BSP, Ayufan, mainline, ...).
I created an issue [there](https://github.com/blakeblackshear/frigate/discussions/12577) and got help from [NickM-27](https://github.com/NickM-27) and [MarcA711](https://github.com/MarcA711). Most information in this post is duplicated from there.

My current kernel is '6.0.0-1164-ayufan-ga2fd0c498'. I think (but not sure) it's not based on BSP so it can explains why I get an error message when testing ffmpeg from the container (copied from linked issue):
```
docker exec -it frigate bash
> ffmpeg -f lavfi -i testsrc2=s=1920x1080,format=nv12 -c:v hevc_rkmpp -qp_init 26 -profile:v main -level 4.1 -g:v 100 -vframes 5000 -y /tmp/tmp.mp4
```

I get:
```
# ffmpeg -f lavfi -i testsrc2=s=1920x1080,format=nv12 -c:v hevc_rkmpp -qp_init 26 -profile:v main -level 4.1 -g:v 100 -vframes 5000 -y /tmp/tmp.mp4     
ffmpeg version 9efe5bcff0-20240613 Copyright (c) 2000-2023 the FFmpeg developers
  built with gcc 13.2.0 (crosstool-NG 1.26.0.65_ecc5e41)
  configuration: --prefix=/ffbuild/prefix --pkg-config-flags=--static --pkg-config=pkg-config --cross-prefix=aarch64-ffbuild-linux-gnu- --arch=aarch64 --target-os=linux --enable-gpl --enable-version3 --disable-debug --enable-iconv --enable-libxml2 --enable-zlib --enable-libfreetype --enable-libfribidi --enable-gmp --enable-openssl --enable-fontconfig --enable-libharfbuzz --enable-libvorbis --enable-opencl --enable-libpulse --enable-libvmaf --enable-libxcb --enable-xlib --enable-amf --enable-libaom --enable-libaribb24 --enable-avisynth --enable-chromaprint --enable-libdav1d --disable-libdavs2 --disable-libfdk-aac --enable-ffnvcodec --enable-cuda-llvm --enable-frei0r --enable-libgme --enable-libkvazaar --enable-libaribcaption --enable-libass --enable-libbluray --enable-libjxl --enable-libmp3lame --enable-libopus --enable-librist --enable-libssh --enable-libtheora --disable-libvpx --enable-libwebp --enable-lv2 --disable-libvpl --enable-openal --enable-libopencore-amrnb --enable-libopencore-amrwb --enable-libopenh264 --enable-libopenjpeg --enable-libopenmpt --enable-librav1e --enable-rkmpp --enable-rkrga --enable-librubberband --disable-schannel --enable-sdl2 --enable-libsoxr --enable-libsrt --enable-libsvtav1 --enable-libtwolame --enable-libuavs3d --enable-libdrm
--disable-vaapi --enable-libvidstab --enable-vulkan --enable-libshaderc --enable-libplacebo --enable-libx264 --enable-libx265 --disable-libxavs2 --enable-libxvid --enable-libzimg --enable-libzvbi --extra-cflags=-DLIBTWOLAME_STATIC --extra-cxxflags= --extra-libs='-ldl -lstdc++ -lstdc++ -lgomp' --extra-ldflags=-pthread --extra-ldexeflags=-pie --cc=aarch64-ffbuild-linux-gnu-gcc --cxx=aarch64-ffbuild-linux-gnu-g++ --ar=aarch64-ffbuild-linux-gnu-gcc-ar --ranlib=aarch64-ffbuild-linux-gnu-gcc-ranlib --nm=aarch64-ffbuild-linux-gnu-gcc-nm --extra-version=20240613
  libavutil      58. 29.100 / 58. 29.100
  libavcodec     60. 31.102 / 60. 31.102
  libavformat    60. 16.100 / 60. 16.100
  libavdevice    60.  3.100 / 60.  3.100
  libavfilter     9. 12.100 /  9. 12.100
  libswscale      7.  5.100 /  7.  5.100
  libswresample   4. 12.100 /  4. 12.100
  libpostproc    57.  3.100 / 57.  3.100
Input #0, lavfi, from 'testsrc2=s=1920x1080,format=nv12':
  Duration: N/A, start: 0.000000, bitrate: N/A
  Stream #0:0: Video: wrapped_avframe, nv12, 1920x1080 [SAR 1:1 DAR 16:9], 25 fps, 25 tbr, 25 tbn                                                                                     
Stream mapping:
  Stream #0:0 -> #0:0 (wrapped_avframe (native) -> hevc (hevc_rkmpp))
Press [q] to stop, [?] for help
[hevc_rkmpp @ 0xaaaafa0775d0] Failed to init MPP context: -1
[vost#0:0/hevc_rkmpp @ 0xaaaafa077270] Error while opening encoder - maybe incorrect parameters such as bit_rate, rate, width or height.                                              
Error while filtering: Generic error in an external library
[out#0/mp4 @ 0xaaaafa075f00] Nothing was written into output file, because at least one of its streams received no packets.                                                           
frame=    0 fps=0.0 q=0.0 Lsize=       0kB time=N/A bitrate=N/A speed=N/A
Conversion failed!
```

I cannot install any of 5.10, 6.1, or 6.10 kernel (BSP versions provided by rockchip) from my package manager, so I'm trying to find it somewhere else.
I noticed LibreElec (LE) provides images using these exact versions, so I suspect they are based on BSP (but not sure).

I tried to load LE12 (kernel 6.1) and launch the static build of ffmpeg as suggested by [MarcA711](https://github.com/MarcA711) but got the same error message (exact log also in linked issue).

I also noticed [here](https://www.reddit.com/r/OrangePI/comments/170ayo5/rockchip_linux_61_bsp_roadmap/) that rockchip plans to support rk3328 only up to kernel 5.1.
However I could not test this version with LibreElec as their userspace is 32bit in this version.

Sorry for the long question, and mostly duplicated information from linked issue.
Any help on getting the HW acceleration working on my board would be appreciated.

Also opened a [thread on LE forum](https://forum.libreelec.tv/thread/28847-rock64-hardware-acceleration-and-kernel-versions/) to gather information from them too :)
Not sure if Ayufan is still active on rockchip linux, will try to contact him too and let you know.

If you prefer, feel free to close the issue and answer in Frigate [linked issue](https://github.com/blakeblackshear/frigate/discussions/12577).
Thank you !

nyanmisaka, 2024-08-20:
This ffmpeg fork has only been tested for the rk35xx series. I have no experience with the rk33xx, but anyway this fork will not work in the mainline kernel. You need the BSP kernel.

closed 2024-08-20 by nyanmisaka

fenouil-r, 2024-08-21:
Ok thank you for the information.

[MarcA711](https://github.com/MarcA711) Spotted a set of patch submitted for ffmpeg [here](https://patchwork.ffmpeg.org/project/ffmpeg/list/?series=12604).

I hope it'll make its way and allow hw acceleration to work for this chip in the (not too far) future.