nyanmisaka/ffmpeg-rockchip #95: Rockchip HW acceleration not working on Orange Pi 5 plus 16GB

abratchik, 2024-08-06:
Hello, 

I'm experimenting with the Orange PI5 Plus box and the frigate version with rknn but it doesnt seem to work on my setup at all. The inference seems to be very poor and also Capture FPS is close to zero. Hence all the detects are skipped:

[image]

Please kindly help me to understand what I'm missing. 

Here is my setup:

**OS/Kernel:** Linux orangepi5plus 6.1.43-rockchip-rk3588 #1.0.8 SMP Sat Feb  3 00:04:38 CST 2024 aarch64 aarch64 aarch64 GNU/Linux

**docker-compose.yml**

```
services:
    mqtt:
      container_name: mqtt
      image: eclipse-mosquitto:2
      restart: unless-stopped
      volumes:
        - /home/orangepi/config/mqtt:/mosquitto/config
      ports:
        - "1833:1833"
      networks:
        - br0
    frigate:
      container_name: frigate
      image: marca711/frigate:latest-rk 
#      image: ghcr.io/blakeblackshear/frigate:stable-rk
      privileged: true 
      restart: unless-stopped
      group_add:
        - "106" # render
        - "44"  # video
        - "46"  # plugdev
      security_opt:
        - apparmor=unconfined
      devices:
        - /dev/dri:/dev/dri
        - /dev/dri/renderD128:/dev/dri/renderD128
        - /dev/dma_heap:/dev/dma_heap
        - /dev/rga:/dev/rga
        - /dev/mpp_service:/dev/mpp_service
      volumes:
        - /home/orangepi/config/frigate/patches/usr/lib/btbn-ffmpeg/bin:/usr/lib/btbn-ffmpeg/bin # patch ffmpeg
        - /home/orangepi/config/frigate/patches/opt/frigate/frigate/ffmpeg_presets.py:/opt/frigate/frigate/ffmpeg_presets.py
        - /home/orangepi/config/frigate/patches/usr/local/go2rtc/create_config.py:/usr/local/go2rtc/create_config.py
        - /dev/bus/usb:/dev/bus/usb
        - /etc/localtime:/etc/localtime:ro
        - /home/orangepi/config/frigate/config.yml:/config/config.yml:rw
        - /home/orangepi/config/frigate/storage:/media/frigate:rw
        - type: tmpfs # 1GB of memory, reduces SSD/SD Card wear
          target: /tmp/cache
          tmpfs:
            size: 1000000000
      ports:
        - "5000:5000"
        - "1935:1935"
        - "8554:8554" # RTSP feeds
        - "8555:8555/tcp" # WebRTC over tcp
        - "8555:8555/udp" # WebRTC over udp
      networks:
        - br0
      device_cgroup_rules:
        - "c 189:* rmw"
      env_file:
        - ./.env.client

networks:
  br0:
    name: br0
```

**config.yml**

```
mqtt:
  enabled: true
  topic_prefix: frigate
  client_id: frigate
  host: mqtt
  user: '{FRIGATE_MQTT_USR}'
  password: '{FRIGATE_MQTT_PWD}'

logger:
  default: info

detectors:
  rknn:
    type: rknn
    core_mask: 0b111
model:
  path: default-yolov8n
  width: 320
  height: 320
  input_tensor: nhwc
  input_pixel_format: bgr

record:
  enabled: true
  retain:
    days: 7
    mode: motion
  events:
    retain:
      default: 7
      mode: motion

snapshots:
  enabled: true
  retain:
    default: 7

cameras:
  cam7:
    enabled: true
    ffmpeg:
      input_args: preset-rtsp-restream
      hwaccel_args: preset-rk-h265
      inputs:
      - path: rtsp://{FRIGATE_RTSP_USR}:{FRIGATE_RTSP_PWD2}@192.168.0.12:554/channel=1_stream=1&amp;onvif=0.sdp?real_st
        roles:
        - detect
    detect:
      enabled: true
      width: 800
      height: 488
      fps: 5
    objects:
      track:
        - person
```

The frigate starts normally with no obvious issues but then I see the log full of messages like:

```
2024-08-06 17:31:22.543948520  E RKNN: [17:31:22.543] failed to submit!, op id: 7, op name: Add:/model.2/m.0/Add, flags: 0x5, task start: 1390, task number: 6, run task counter: 0, int status: 0, please try updating to the latest version of the toolkit2 and runtime from: https://console.zbox.filez.com/l/I00fc3 (PWD: rknn)
```

is there an image where this is fixed already? Or I need to update the host?

nyanmisaka, 2024-08-06:
It looks like you opened the issue in the wrong repo? This is `ffmpeg-rockchip` not `frigate`.

abratchik, 2024-08-06:
Sorry, I'm a bit confused where to open this. There is a long thread about [this problem](https://github.com/blakeblackshear/frigate/issues/9201 ) in the frigate repo and @MarcA711 (seems like he is supporting the frigate fork for the rockchip) advised to open the issue in this repo. Should I submit it there as well?

nyanmisaka, 2024-08-06:
`
E RKNN: [17:31:22.543] failed to submit!, op id: 7, op name: Add:/model.2/m.0/Add, flags: 0x5, task start: 1390, task number: 6, run task counter: 0, int status: 0, please try updating to the latest version of the toolkit2 and runtime from: https://console.zbox.filez.com/l/I00fc3 (PWD: rknn)
`

It looks like your RKNN runtime library or driver is out of date, causing inference failure.

nyanmisaka, 2024-08-06:
https://github.com/Joshua-Riek/ubuntu-rockchip
Flash the latest `ubuntu-rockchip` image and try again.

abratchik, 2024-08-06:
Ok thanks, will try, many thanks!

MarcA711, 2024-08-06:
Hey,

Other users also reported issues with frigate version 13.2 and newer kernels. You could try to upgrade to v14 release candidate 2.

abratchik, 2024-08-06:
@MarcA711  Hi Marc, thanks, I will try that too and let know the results.

abratchik, 2024-08-06:
In the meantime I found this [thread](https://github.com/rockchip-linux/rknn-toolkit2/issues/320). According to @Link164735 

```
The reason for my issue was that the bit width of the model's output layer exceeded the maximum limit of the NPU, causing scheduling problems with the chip. It's important to note that the NPU's maximum bit width is 0x1fff, which is 8191. My original output layer was 2*9288. After modifying the dimensions of my output layer, the error disappeared. I hope this helps you.
```

If this is related to the model dimensions and NPU capacity, shouldn't the fix be on the model itself?

abratchik, 2024-08-07:
> @MarcA711 Hi Marc, thanks, I will try that too and let know the results.

Hi @MarcA711 , FYI - I tried the 14.0 rc2 on the same system I have (Linux orangepi5plus 6.1.43-rockchip-rk3588) and it happened to be unusable also. 

```
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service log-prepare: starting
s6-rc: info: service log-prepare successfully started
s6-rc: info: service nginx-log: starting
s6-rc: info: service go2rtc-log: starting
s6-rc: info: service frigate-log: starting
s6-rc: info: service certsync-log: starting
s6-rc: info: service frigate-log successfully started
s6-rc: info: service nginx-log successfully started
s6-rc: info: service certsync-log successfully started
s6-rc: info: service go2rtc-log successfully started
s6-rc: info: service go2rtc: starting
s6-rc: info: service go2rtc successfully started
s6-rc: info: service go2rtc-healthcheck: starting
s6-rc: info: service frigate: starting
s6-rc: info: service go2rtc-healthcheck successfully started
s6-rc: info: service frigate successfully started
s6-rc: info: service nginx: starting
2024-08-07 08:59:50.013134201  [INFO] Preparing Frigate...
2024-08-07 08:59:50.014692260  [INFO] Starting NGINX...
2024-08-07 08:59:50.033956541  [INFO] Preparing new go2rtc config...
2024-08-07 08:59:50.087776947  [INFO] Starting Frigate...
2024-08-07 08:59:50.648918479  [INFO] Starting go2rtc...
2024-08-07 08:59:50.780458957  08:59:50.780 INF go2rtc platform=linux/arm64 revision=b2399f3 version=1.9.2
2024-08-07 08:59:50.780497165  08:59:50.780 INF config path=/dev/shm/go2rtc.yaml
2024-08-07 08:59:50.781260737  08:59:50.781 INF [api] listen addr=:1984
2024-08-07 08:59:50.781274445  08:59:50.781 INF [rtsp] listen addr=:8554
2024-08-07 08:59:50.781545690  08:59:50.781 INF [webrtc] listen addr=:8555/tcp
s6-rc: info: service nginx successfully started
s6-rc: info: service certsync: starting
s6-rc: info: service certsync successfully started
s6-rc: info: service legacy-services: starting
2024-08-07 08:59:51.152682869  [INFO] Starting certsync...
s6-rc: info: service legacy-services successfully started
2024-08-07 08:59:51.318009193  127.0.0.1 - - [07/Aug/2024:08:59:51 +0300] "" 400 0 "-" "-" "-"
2024-08-07 08:59:52.328247056  [2024-08-07 08:59:52] frigate.app                    INFO    : Starting Frigate (0.14.0-da913d8)
2024-08-07 08:59:52.328648383  [2024-08-07 08:59:52] frigate.util.config            INFO    : Checking if frigate config needs migration...
2024-08-07 08:59:52.349969757  [2024-08-07 08:59:52] frigate.util.config            INFO    : frigate config does not need migration...
2024-08-07 08:59:52.580911176  [2024-08-07 08:59:52] frigate.util.services          WARNING : Did not detect hwaccel, using a GPU for accelerated video decoding is highly recommended
2024-08-07 08:59:52.608031626  [2024-08-07 08:59:52] peewee_migrate.logs            INFO    : Starting migrations
2024-08-07 08:59:52.609102026  [2024-08-07 08:59:52] peewee_migrate.logs            INFO    : There is nothing to migrate
2024-08-07 08:59:52.618451254  [2024-08-07 08:59:52] frigate.app                    INFO    : Recording process started: 422
2024-08-07 08:59:52.628202684  [2024-08-07 08:59:52] frigate.app                    INFO    : Recording process started: 423
2024-08-07 08:59:52.632184455  [2024-08-07 08:59:52] frigate.app                    INFO    : go2rtc process pid: 98
2024-08-07 08:59:52.679575754  [2024-08-07 08:59:52] detector.rknn                  INFO    : Starting detection process: 451
2024-08-07 08:59:52.680254451  Process detector:rknn:
2024-08-07 08:59:52.680537363  [2024-08-07 08:59:52] frigate.app                    INFO    : Output process started: 453
2024-08-07 08:59:52.682095422  Traceback (most recent call last):
2024-08-07 08:59:52.682102422    File "/usr/lib/python3.9/multiprocessing/process.py", line 315, in _bootstrap
2024-08-07 08:59:52.682104756      self.run()
2024-08-07 08:59:52.682130713    File "/usr/lib/python3.9/multiprocessing/process.py", line 108, in run
2024-08-07 08:59:52.682135672      self._target(*self._args, **self._kwargs)
2024-08-07 08:59:52.682139755    File "/opt/frigate/frigate/object_detection.py", line 102, in run_detector
2024-08-07 08:59:52.682189921      object_detector = LocalObjectDetector(detector_config=detector_config)
2024-08-07 08:59:52.682195462    File "/opt/frigate/frigate/object_detection.py", line 53, in __init__
2024-08-07 08:59:52.682198379      self.detect_api = create_detector(detector_config)
2024-08-07 08:59:52.682201587    File "/opt/frigate/frigate/detectors/__init__.py", line 18, in create_detector
2024-08-07 08:59:52.682265461      return api(detector_config)
2024-08-07 08:59:52.682269545    File "/opt/frigate/frigate/detectors/plugins/rknn.py", line 39, in __init__
2024-08-07 08:59:52.682273045      model_props = self.parse_model_input(config.model.path, soc)
2024-08-07 08:59:52.682276253    File "/opt/frigate/frigate/detectors/plugins/rknn.py", line 120, in parse_model_input
2024-08-07 08:59:52.682312127      raise Exception(
2024-08-07 08:59:52.682346543  Exception: Model default-yolov8n is unsupported. Provide your own model or choose one of the following: olona
2024-08-07 08:59:52.682350335  Exception ignored in: <function Rknn.__del__ at 0x7f8c48fb80>
2024-08-07 08:59:52.682352668  Traceback (most recent call last):
[... 13 lines cut ...]
s6-rc: info: service nginx: stopping
s6-rc: info: service certsync-log: stopping
[... cut ...]
2024-08-07 09:00:03.215645247  [2024-08-07 09:00:03] frigate.app                    INFO    : Waiting for process for cam7 to stop
2024-08-07 09:00:03.246003021  [2024-08-07 09:00:03] frigate.stats.emitter          INFO    : Exiting stats emitter...
2024-08-07 09:00:03.304925180  [2024-08-07 09:00:03] frigate.watchdog               INFO    : Exiting watchdog...
```

Seems like it is still under development. I will try to re-flash the host OS as @nyanmisaka suggested and will probably go back to your earlier image patched with the latest ffmpeg-rockchip from this repo. Will let you know the outcome.

MarcA711, 2024-08-07:
Hey,

you have to adjust two options in your config for v14. I made these changes bold:

<pre><code>detectors:
  rknn:
    type: rknn
    <b>num_cores: 3</b>
model:
  path: <b>deci-fp16-yolonas_m</b>
  width: 320
  height: 320
  input_tensor: nhwc
  input_pixel_format: bgr</code></pre>

You can also select one of the models here instead of yolonas_m: https://deploy-preview-11419--frigate-docs.netlify.app/configuration/object_detectors#choosing-a-model
There were license issues that forced us to drop support for yolov8. We use yolonas now. However, yolonas performs better than yolov8.

I will post this issue here later this day: https://github.com/blakeblackshear/frigate/discussions/11502
This issue is not related to ffmpeg at all as far as I can tell right now.

MarcA711, 2024-08-07:
I posted a comment about this potential error here: https://github.com/blakeblackshear/frigate/discussions/11502#discussioncomment-10263860

This issue is not related to ffmpeg, so it can be closed.

Moreover, I believe that this issue is just an incompatibility between the older yolov8 model and newer kernels. Since yolov8 is not supported anymore, I will not dig deeper into this problem. Please upgrade to Frigate v14 instead.

nyanmisaka, 2024-08-07:
> I posted a comment about this potential error here: [blakeblackshear/frigate#11502 (comment)](https://github.com/blakeblackshear/frigate/discussions/11502#discussioncomment-10263860)
> 
> This issue is not related to ffmpeg, so it can be closed.
> 
> Moreover, I believe that this issue is just an incompatibility between the older yolov8 model and newer kernels. Since yolov8 is not supported anymore, I will not dig deeper into this problem. Please upgrade to Frigate v14 instead.

Thanks for the confirmation.

closed 2024-08-07 by nyanmisaka

abratchik, 2024-08-07:
@MarcA711 Hello Marc, 

 I tested this on my old host setup,  did not re-flash the OS yet. The issue is still the same as before - very slow speed of rknn, and same error messages:

```
s6-rc: info: service s6rc-fdholder: starting
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service s6rc-fdholder successfully started
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service log-prepare: starting
s6-rc: info: service log-prepare successfully started
s6-rc: info: service nginx-log: starting
s6-rc: info: service go2rtc-log: starting
s6-rc: info: service frigate-log: starting
s6-rc: info: service certsync-log: starting
s6-rc: info: service go2rtc-log successfully started
s6-rc: info: service go2rtc: starting
s6-rc: info: service nginx-log successfully started
s6-rc: info: service certsync-log successfully started
s6-rc: info: service frigate-log successfully started
s6-rc: info: service go2rtc successfully started
s6-rc: info: service go2rtc-healthcheck: starting
s6-rc: info: service frigate: starting
s6-rc: info: service go2rtc-healthcheck successfully started
s6-rc: info: service frigate successfully started
s6-rc: info: service nginx: starting
2024-08-07 18:04:58.661538109  [INFO] Starting NGINX...
2024-08-07 18:04:58.661677523  [INFO] Preparing Frigate...
2024-08-07 18:04:58.680387373  [INFO] Preparing new go2rtc config...
2024-08-07 18:04:58.738599390  [INFO] Starting Frigate...
2024-08-07 18:04:59.350873949  [INFO] Starting go2rtc...
2024-08-07 18:04:59.454477672  18:04:59.454 INF go2rtc platform=linux/arm64 revision=b2399f3 version=1.9.2
2024-08-07 18:04:59.454510630  18:04:59.454 INF config path=/dev/shm/go2rtc.yaml
2024-08-07 18:04:59.455093955  18:04:59.455 INF [rtsp] listen addr=:8554
2024-08-07 18:04:59.455158121  18:04:59.455 INF [api] listen addr=:1984
2024-08-07 18:04:59.455302785  18:04:59.455 INF [webrtc] listen addr=:8555/tcp
s6-rc: info: service nginx successfully started
s6-rc: info: service certsync: starting
s6-rc: info: service certsync successfully started
s6-rc: info: service legacy-services: starting
2024-08-07 18:04:59.809889976  [INFO] Starting certsync...
s6-rc: info: service legacy-services successfully started
2024-08-07 18:04:59.964591196  127.0.0.1 - - [07/Aug/2024:18:04:59 +0300] "" 400 0 "-" "-" "-"
2024-08-07 18:05:01.001543372  [2024-08-07 18:05:00] frigate.app                    INFO    : Starting Frigate (0.14.0-da913d8)
2024-08-07 18:05:01.001832992  [2024-08-07 18:05:00] frigate.util.config            INFO    : Checking if frigate config needs migration...
2024-08-07 18:05:01.021506202  [2024-08-07 18:05:01] frigate.util.config            INFO    : frigate config does not need migration...
2024-08-07 18:05:01.263202724  [2024-08-07 18:05:01] frigate.util.services          WARNING : Did not detect hwaccel, using a GPU for accelerated video decoding is highly recommended
2024-08-07 18:05:01.290860691  [2024-08-07 18:05:01] peewee_migrate.logs            INFO    : Starting migrations
2024-08-07 18:05:01.291462099  [2024-08-07 18:05:01] peewee_migrate.logs            INFO    : There is nothing to migrate
2024-08-07 18:05:01.312438081  [2024-08-07 18:05:01] frigate.app                    INFO    : Recording process started: 421
2024-08-07 18:05:01.325222601  [2024-08-07 18:05:01] frigate.app                    INFO    : Recording process started: 422
2024-08-07 18:05:01.329100836  [2024-08-07 18:05:01] frigate.app                    INFO    : go2rtc process pid: 97
2024-08-07 18:05:01.380700575  [2024-08-07 18:05:01] frigate.app                    INFO    : Output process started: 452
2024-08-07 18:05:01.384746516  [2024-08-07 18:05:01] detector.rknn                  INFO    : Starting detection process: 450
2024-08-07 18:05:01.385143177  [2024-08-07 18:05:01] frigate.detectors.plugins.rknn INFO    : You are using yolo-nas with weights from DeciAI. These weights are subject to their license and can't be used commercially. For more information, see: https://docs.deci.ai/super-gradients/latest/LICENSE.YOLONAS.html
2024-08-07 18:05:01.408886535  [2024-08-07 18:05:01] frigate.app                    INFO    : Camera processor started for cam7: 466
2024-08-07 18:05:01.422128298  [2024-08-07 18:05:01] frigate.app                    INFO    : Capture process started for cam7: 467
2024-08-07 18:05:01.861423074  I RKNN: [18:05:01.861] RKNN Runtime Information, librknnrt version: 2.0.0b0 (35a6907d79@2024-03-24T10:31:14)
2024-08-07 18:05:01.861459532  I RKNN: [18:05:01.861] RKNN Driver Information, version: 0.9.3
2024-08-07 18:05:01.862178771  I RKNN: [18:05:01.862] RKNN Model Information, version: 6, toolkit version: 2.0.0b0+9bab5682(compiler version: 2.0.0b0 (35a6907d79@2024-03-24T02:34:11)), target: RKNPU v2, target platform: rk3588, framework name: ONNX, framework layout: NCHW, model inference type: static_shape
2024-08-07 18:05:08.653555614  [INFO] Starting go2rtc healthcheck service...
2024-08-07 18:05:13.254998227  E RKNN: [18:05:13.254] failed to submit!, op id: 1, op name: Conv:/backbone/stem/conv/rbr_reparam/Conv, flags: 0x5, task start: 1280, task number: 66, run task counter: 0, int status: 0, please try updating to the latest version of the toolkit2 and runtime from: https://console.zbox.filez.com/l/I00fc3 (PWD: rknn)
2024-08-07 18:05:19.654755583  E RKNN: [18:05:19.654] failed to submit!, op id: 1, op name: Conv:/backbone/stem/conv/rbr_reparam/Conv, flags: 0x5, task start: 1280, task number: 66, run task counter: 0, int status: 0, please try updating to the latest version of the toolkit2 and runtime from: https://console.zbox.filez.com/l/I00fc3 (PWD: rknn)
``` 

I also noticed that the models are proprietary, this is also not good. The performance did not change much:

[image]

MarcA711, 2024-08-07:
Hey,

Yes, you are using rknn driver 0.9.3. You should update your os.

yolonas is not proprietary. It is open source and available [here](https://github.com/Deci-AI/super-gradients).
They only have a special license for their weights. If you want to avoid this, you can train yolonas from scratch on your dataset.

abratchik, 2024-08-07:
Thank you, noted. Will re-flash tomorrow and let know the results. Many thanks again.