nyanmisaka/ffmpeg-rockchip #96: FFMPEG Causing color conversion causing kernel warning using HDMIRX

benhoff, 2024-08-08:
Hey trying to using HDMI-In on an orange pi 5 plus with a transition from BGR24 to RGBA is causing an error/kernel warning.
 Or at least, I think it is, if I'm not doing something wrong.

Command line invocation :

`ffmpeg -f video4linux2 -i /dev/video2 -pix_fmt bgra -f fbdev /dev/fb0`

result:
``` 
ffmpeg version 8164ff7 Copyright (c) 2000-2023 the FFmpeg developers
  built with gcc 13 (Ubuntu 13.2.0-23ubuntu4)
  configuration: --prefix=/usr --enable-gpl --enable-version3 --enable-libdrm --enable-rkmpp --enable-rkrga --disable-xlib --disable-libxcb --disable-libxcb-shm --disable-libxcb-xfixes --disable-libxcb-shape
  libavutil      58. 29.100 / 58. 29.100
  libavcodec     60. 31.102 / 60. 31.102
  libavformat    60. 16.100 / 60. 16.100
  libavdevice    60.  3.100 / 60.  3.100
  libavfilter     9. 12.100 /  9. 12.100
  libswscale      7.  5.100 /  7.  5.100
  libswresample   4. 12.100 /  4. 12.100
  libpostproc    57.  3.100 / 57.  3.100
[video4linux2,v4l2 @ 0xaaaaf948f090] ioctl(VIDIOC_REQBUFS): Invalid argument
[in#0 @ 0xaaaaf948ef70] Error opening input: Invalid argument
Error opening input file /dev/video1.
Error opening input files: Invalid argument

```
Journalctl log:
```
Aug 08 13:58:48 orangepi5-plus kernel: WARNING: CPU: 2 PID: 1790 at drivers/media/common/videobuf2/videobuf2-core.c:
944 vb2_core_reqbufs+0x188/0x4b4 [videobuf2_common]                                                                 
Aug 08 13:58:48 orangepi5-plus kernel: Modules linked in: sunrpc binfmt_misc snd_soc_simple_card crct10dif_ce gpio_i
r_recv snd_soc_simple_card_utils snd_soc_simple_amplifier leds_pwm pwm_fan rc_core btusb snd_soc_es8328_i2c btrtl sn
d_soc_rockchip_i2s_tdm snd_soc_rockchip_i2s snd_soc_es8328 btintel iwlmvm rk805_pwrkey snd_soc_core btbcm btmtk hant
ro_vpu snd_compress ac97_bus mac80211 bluetooth snd_pcm_dmaengine rockchip_vdec2(C) rockchip_rga v4l2_vp9 v4l2_h264 
snd_pcm libarc4 ecdh_generic synopsys_hdmirx rk_crypto2 snd_timer ecc snd v4l2_mem2mem v4l2_dv_timings sm3_generic c
rypto_engine videobuf2_dma_sg lz4hc videobuf2_dma_contig videobuf2_memops soundcore sm3 videobuf2_v4l2 lz4 iwlwifi v
ideodev panthor drm_gpuvm cfg80211 videobuf2_common mc gpu_sched drm_shmem_helper drm_exec rfkill_gpio rfkill zram d
m_mod nfnetlink ip_tables x_tables autofs4 rk808_regulator rockchipdrm rk8xx_spi dw_hdmi nvmem_rockchip_otp cec dw_h
dmi_qp fusb302 dw_mipi_dsi tcpm rk8xx_core drm_dma_helper phy_rockchip_snps_pcie3 analogix_dp drm_display_helper
Aug 08 13:58:48 orangepi5-plus kernel:  phy_rockchip_samsung_hdptx drm_kms_helper rockchip_rng r8169 drm adc_keys
Aug 08 13:58:48 orangepi5-plus kernel: CPU: 2 PID: 1790 Comm: ffmpeg Tainted: G        WC         6.10.0-edge-rockch
ip-rk3588 #4                                                                                                        
Aug 08 13:58:48 orangepi5-plus kernel: Hardware name: Xunlong Orange Pi 5 Plus (DT)
Aug 08 13:58:48 orangepi5-plus kernel: pstate: 60400009 (nZCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
Aug 08 13:58:48 orangepi5-plus kernel: pc : vb2_core_reqbufs+0x188/0x4b4 [videobuf2_common]
Aug 08 13:58:48 orangepi5-plus kernel: lr : vb2_core_reqbufs+0x154/0x4b4 [videobuf2_common]
Aug 08 13:58:48 orangepi5-plus kernel: sp : ffff800089443ac0
Aug 08 13:58:48 orangepi5-plus kernel: x29: ffff800089443ac0 x28: ffff80007a424e68 x27: ffff0001f6f3c6c0
Aug 08 13:58:48 orangepi5-plus kernel: x26: ffff0001f3964778 x25: ffff0001f3964590 x24: ffff0001f39645d0
Aug 08 13:58:48 orangepi5-plus kernel: x23: 0000000000000001 x22: ffff800089443d28 x21: 0000000000000000
Aug 08 13:58:48 orangepi5-plus kernel: x20: 0000000000000000 x19: ffff0001f3964528 x18: 0000000000000000
Aug 08 13:58:48 orangepi5-plus kernel: x17: 0000000000000000 x16: 0000000000000000 x15: 0000ffffdc8531e8
Aug 08 13:58:48 orangepi5-plus kernel: x14: 0000000000000000 x13: 0000000000000000 x12: 0000000000000000
Aug 08 13:58:48 orangepi5-plus kernel: x11: 0000000000000000 x10: 0000000000000000 x9 : 0000000000000000
Aug 08 13:58:48 orangepi5-plus kernel: x8 : ffff0001f0aea778 x7 : ffff0001f3964080 x6 : ffff80007a426564
Aug 08 13:58:48 orangepi5-plus kernel: x5 : 0000000000000001 x4 : 0000000000000001 x3 : 0000000000000001
Aug 08 13:58:48 orangepi5-plus kernel: x2 : ffff800089443b28 x1 : ffff800089443b2c x0 : 0000000000000000
Aug 08 13:58:48 orangepi5-plus kernel: Call trace:
Aug 08 13:58:48 orangepi5-plus kernel:  vb2_core_reqbufs+0x188/0x4b4 [videobuf2_common]
Aug 08 13:58:48 orangepi5-plus kernel:  vb2_ioctl_reqbufs+0xcc/0xf4 [videobuf2_v4l2]
Aug 08 13:58:48 orangepi5-plus kernel:  v4l_reqbufs+0x74/0x8c [videodev]
Aug 08 13:58:48 orangepi5-plus kernel:  __video_do_ioctl+0x188/0x3e8 [videodev]
Aug 08 13:58:48 orangepi5-plus kernel:  video_usercopy+0x104/0x76c [videodev]
Aug 08 13:58:48 orangepi5-plus kernel:  video_ioctl2+0x18/0x2c [videodev]
Aug 08 13:58:48 orangepi5-plus kernel:  v4l2_ioctl+0x40/0x60 [videodev]
Aug 08 13:58:48 orangepi5-plus kernel:  __arm64_sys_ioctl+0xa8/0xec
Aug 08 13:58:48 orangepi5-plus kernel:  invoke_syscall+0x48/0x118
Aug 08 13:58:48 orangepi5-plus kernel:  el0_svc_common.constprop.0+0x40/0xe8
Aug 08 13:58:48 orangepi5-plus kernel:  do_el0_svc+0x20/0x2c
Aug 08 13:58:48 orangepi5-plus kernel:  el0_svc+0x38/0x100
Aug 08 13:58:48 orangepi5-plus kernel:  el0t_64_sync_handler+0x13c/0x158
Aug 08 13:58:48 orangepi5-plus kernel:  el0t_64_sync+0x1a4/0x1a8
Aug 08 13:58:48 orangepi5-plus kernel: ---[ end trace 0000000000000000 ]---
```

The reason I think this is an issue with ffmpeg-rockchip is because this code works without crashing:
`ffmpeg -f video4linux2 -input_format bgr24 -i /dev/video1 -f rawvideo out.raw`

Any thoughts on how to fix/debug this? Happy to drop a patch if I can.

nyanmisaka, 2024-08-08:
> 6.10.0-edge-rockchip-rk3588

This ffmpeg fork is only targeted for use with the rockchip BSP kernel such as 5.10 or 6.1.
With mainline kernel you should use the vanilla ffmpeg instead.

benhoff, 2024-08-09:
@nyanmisaka , is it worth porting the rga device patches? I assumed maybe naively that the kernel wouldn't matter much.

I've thought about just doing the color space conversion in opengl as well

nyanmisaka, 2024-08-09:
I don't think it's worth it, the rga in mainline and bsp is a completely different thing, which uses the v4l2 interface and only supports rga2.

If you need full functionality, use the bsp kernel, the mainline is currently not suitable for daily use, especially media functions.

benhoff, 2024-08-09:
Thanks for the tips and the patience with me!

After compiling mainline/upstream, I found out that I actually need multiplanar support from ffmpeg, which only got added last month. With the current device you  always get the error/exception: "Not a video capture device"

See that commit [here](https://github.com/FFmpeg/FFmpeg/commit/7eb37c142a92b27b0d5f66f9d6ad5905085daa8e).

closed 2024-08-09 by benhoff