nyanmisaka/ffmpeg-rockchip #97: Does scale_rkrga support nv12 to bgr24?

wxd9199, 2024-08-14:
Does scale_rkrga support nv12 to bgr24?

nyanmisaka, 2024-08-14:
https://github.com/nyanmisaka/ffmpeg-rockchip/blob/8164ff7dadd655d78a28340ddc9fcef9af6999bb/libavfilter/rkrga_common.c#L45-L85

Yes.

closed 2024-08-14 by nyanmisaka

wxd9199, 2024-08-14:
if  AV_PIX_FMT_DRM_PRIME can  conversion to BGR24  ?

nyanmisaka, 2024-08-14:
`-vf scale_rkrga=format=bgr24,hwmap,format=bgr24`
`-vf scale_rkrga=format=bgr24,hwdownload,format=bgr24`

You need the hwmap or hwdownload filter to map an hardware frame to software frame.

wxd9199, 2024-08-14:
`bool VideoDecode::initRgaFilter(int width, int height, AVPixelFormat src_pix_fmt, AVPixelFormat dst_pix_fmt) {
    if (!filter_graph) {
        return false;
    }
     
    // Define the filter string
    const char *filter_desc = "scale_rkrga=format=bgr24,hwmap,format=bgr24";

    const AVFilter *buffersrc = avfilter_get_by_name("buffer");
    const AVFilter *buffersink = avfilter_get_by_name("buffersink");
    AVFilterInOut *outputs = avfilter_inout_alloc();
    AVFilterInOut *inputs = avfilter_inout_alloc();

    if (!outputs || !inputs || !buffersrc || !buffersink) {
        avfilter_inout_free(&outputs);
        avfilter_inout_free(&inputs);
        return false;
    }

    // Create and configure the buffer source and sink contexts
    char args[512];
    snprintf(args, sizeof(args),
             "video_size=%dx%d:pix_fmt=%d:time_base=%d/%d:pixel_aspect=%d/%d",
             width, height, src_pix_fmt,
             m_formatContext->streams[m_videoIndex]->time_base.num, m_formatContext->streams[m_videoIndex]->time_base.den,
             m_codecContext->sample_aspect_ratio.num, m_codecContext->sample_aspect_ratio.den);

    int err = avfilter_graph_create_filter(&buffersrc_ctx, buffersrc, "in", args, nullptr, filter_graph);
    if (err < 0) {
        std::cout << "avfilter_graph_create_filter--------------error" << std::endl;
        avfilter_inout_free(&outputs);
        avfilter_inout_free(&inputs);
        return false;
    }

    err = avfilter_graph_create_filter(&buffersink_ctx, buffersink, "out", nullptr, nullptr, filter_graph);
    if (err < 0) {
        std::cout << "avfilter_graph_create_filter--------------error" << std::endl;
        avfilter_inout_free(&outputs);
        avfilter_inout_free(&inputs);
        return false;
    }

    enum AVPixelFormat pix_fmts[] = { dst_pix_fmt, AV_PIX_FMT_NONE };
    err = av_opt_set_int_list(buffersink_ctx, "pix_fmts", pix_fmts, AV_PIX_FMT_NONE, AV_OPT_SEARCH_CHILDREN);
    if (err < 0) {
        avfilter_inout_free(&outputs);
        avfilter_inout_free(&inputs);
        return false;
    }

    outputs->name = av_strdup("in");
    outputs->filter_ctx = buffersrc_ctx;
    outputs->pad_idx = 0;
    outputs->next = nullptr;

    inputs->name = av_strdup("out");
    inputs->filter_ctx = buffersink_ctx;
    inputs->pad_idx = 0;
    inputs->next = nullptr;

    err = avfilter_graph_parse_ptr(filter_graph, filter_desc, &inputs, &outputs, nullptr);
    if (err < 0) {
        std::cout << "avfilter_graph_parse_ptr--------------error" << std::endl;
        avfilter_inout_free(&outputs);
        avfilter_inout_free(&inputs);
        return false;
    }
    
    err = avfilter_graph_config(filter_graph, nullptr);
    if (err < 0) {
        std::cout << "avfilter_graph_config--------------error" << std::endl;
        avfilter_inout_free(&outputs);
        avfilter_inout_free(&inputs);
        return false;
    }
    
    avfilter_inout_free(&outputs);
    avfilter_inout_free(&inputs);

    filtered_frame = av_frame_alloc();
    return filtered_frame != nullptr;
}
`initRgaFilter(m_size_w, m_size_h, AV_PIX_FMT_DRM_PRIME, AV_PIX_FMT_BGR24)

nyanmisaka, 2024-08-14:
Hardware filter requires a `hw_frames_ctx` which you did not provide.

https://github.com/nyanmisaka/ffmpeg-rockchip/blob/8164ff7dadd655d78a28340ddc9fcef9af6999bb/libavfilter/vf_vpp_rkrga.c#L355C18-L355C31

https://github.com/search?q=repo%3Avivictorg%2Fvivictpp%20hw_frames_ctx&type=code

wxd9199, 2024-08-15:
@nyanmisaka Hi  
int ret = av_hwframe_transfer_data(m_frameHW, m_frame, 0);    -------ok
int ret = av_hwframe_map(m_frameHW, m_frame, 0);                 --------error：Function not implemented  
Is it because rkmpp does not support it?