nyanmisaka/ffmpeg-rockchip #98: dolby vision tone mapping

baixin1228, 2024-08-15:
does it support dolby vision tone mapping?

nyanmisaka, 2024-08-15:
Nope.

closed 2024-08-15 by nyanmisaka

baixin1228, 2024-08-15:
> Nope.

Is there any diffculty involved here?

baixin1228, 2024-08-15:
> Nope.

I'am interested in adapting Dolby Vision tone mapping,but I'am not sure if there are any unsolvable diffculties involved.

nyanmisaka, 2024-08-15:
> > Nope.
> 
> I'am interested in adapting Dolby Vision tone mapping,but I'am not sure if there are any unsolvable diffculties involved.

The problem is not with ffmpeg, but with the incompatibility between the mpp parsed and exported dovi metadata and what ffmpeg requires. This needs to be fixed in mpp. I don't have time to look into it yet.
- https://github.com/search?q=repo%3Arockchip-linux%2Fmpp+dolby&type=code
- https://github.com/FFmpeg/FFmpeg/commit/a538df7eab19ecd0e2d59ef13a84e343de7f65b7

baixin1228, 2024-08-16:
> > > Nope.
> > 
> > 
> > I'am interested in adapting Dolby Vision tone mapping,but I'am not sure if there are any unsolvable diffculties involved.
> 
> The problem is not with ffmpeg, but with the incompatibility between the mpp parsed and exported dovi metadata and what ffmpeg requires. This needs to be fixed in mpp. I don't have time to look into it yet.
> 
> * https://github.com/search?q=repo%3Arockchip-linux%2Fmpp+dolby&type=code
> * [FFmpeg/FFmpeg@a538df7](https://github.com/FFmpeg/FFmpeg/commit/a538df7eab19ecd0e2d59ef13a84e343de7f65b7)

I'am look into it.