nyanmisaka/ffmpeg-rockchip #101: Failed to get key input frame from packet meta when encoding with h264_rkmpp encoder

lumiaJJ, 2024-08-21:
Hi,
I am transcoding two streams with h264_rkmpp encoder. But I got an error when my program called av_hwframe_transfer_data before encoded  the frame.  This is an occasional problem， I got some information from this blog :https://brainly.com/question/43026554. 
Can you help to analyze the issue from mpp extension's point of view.
```
[h264_rkmpp @ 0x7fa4015560] Failed to get key input frame from packet meta: -1
malloc(): unaligned tcache chunk detected
```

nyanmisaka, 2024-08-21:
The MPP library itself contains some bugs. Use [nyanmisaka/mpp](https://github.com/nyanmisaka/mpp/commits/jellyfin-mpp/) instead.

lumiaJJ, 2024-08-21:
> The MPP library itself contains some bugs. Use [nyanmisaka/mpp](https://github.com/nyanmisaka/mpp/commits/jellyfin-mpp/) instead.

[nyanmisaka/mpp](https://github.com/nyanmisaka/mpp/commits/jellyfin-mpp/) is already used to build MPP library, I will checked the version of in my local  tomorrow.

lumiaJJ, 2024-08-22:
@nyanmisaka Hi, checked version of [nyanmisaka/mpp](https://github.com/nyanmisaka/mpp/commits/jellyfin-mpp/) in my local env. The lastest commit is 4ded78b9510145fa79bcd51beb1591450c872277 (grafted, HEAD -> jellyfin-mpp, origin/jellyfin-mpp) is same as remote's

lumiaJJ, 2024-08-22:
> @nyanmisaka Hi, checked version of [nyanmisaka/mpp](https://github.com/nyanmisaka/mpp/commits/jellyfin-mpp/) in my local env. The lastest commit is 4ded78b9510145fa79bcd51beb1591450c872277 (grafted, HEAD -> jellyfin-mpp, origin/jellyfin-mpp) is same as remote's. Perhaps it might be a new issue

nyanmisaka, 2024-08-22:
I can't reproduce with the command described in the Wiki. https://github.com/nyanmisaka/ffmpeg-rockchip/wiki

Can you share a minimal code sample that can be compiled?

lumiaJJ, 2024-08-22:
> I can't reproduce with the command described in the Wiki. https://github.com/nyanmisaka/ffmpeg-rockchip/wiki
> 
> Can you share a minimal code sample that can be compiled?

Ok，I will provide it asap.

lumiaJJ, 2024-08-24:
https://github.com/nyanmisaka/ffmpeg-rockchip/pull/104  Hi, the issue might be related with this PR.  You can enable MJPEG decoding to produced. And qsv_decode.c can be a a minimal code sample to test.